MichaIng/DietPi #7402: Unable to locate package linux-image-rpi-v8

SOULV1CE, 2025-02-11:
#### Details:
- Date           | 2025-02-11 18:58:13
- Bug report     | a3fff781-9e74-4eb2-98a0-bca4147c85d2
- Program name   | DietPi-Set_software
- Command        | `apt-get -y --allow-change-held-packages install linux-image-rpi-v6 linux-image-rpi-v8 linux-image-rpi-v7 raspi-firmware raspi-utils-core tiny-initramfs`
- Exit code      | 100
- DietPi version | v9.9.0 (MichaIng/master)
- Distro version | bookworm (ID=7,RASPBIAN=1)
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Architecture   | `armhf`
- Hardware model | RPi Zero W (armv6l) (ID=1)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

1. Update DietPi from v9.9 to v9.10
2. Say yes to updating kernel version
3. Select option to download/keep additional images for other RPi hardware
#### Expected behaviour:

- Additional images should download
#### Actual behaviour:

- Unable to locate package linux-image-rpi-v8
#### Extra details:

- See additional logs below
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package linux-image-rpi-v8
```

MichaIng, 2025-02-12:
Fixed with: https://github.com/MichaIng/DietPi/commit/7279f87

MichaIng, 2025-02-12:
Live patch merged as well.

closed 2025-02-12 by MichaIng

Lycidias93, 2025-02-13:
[image]

Hello, I haven't found a solution for this.

Can anyone guide me?

Thanks

Lycidias93, 2025-02-13:
> I found the issue: When youvm are at the error handler, "Open shell" and run `apt update`. Else, do from console and rerun the `dietpi-update`.

This didn't help :(

Joulinar, 2025-02-13:
Issue changed, package is now found. But it seems trying to install a 64bit package while you are on 32bit. This of course will not work, 🤔

MichaIng, 2025-02-13:
Has `arm64` already been added as foreign architecture?
```sh
dpkg --print-foreign-architectures
```

MichaIng, 2025-02-13:
.... ohhh now I understand: It depends on the `arm64` version it any initramfs implementation. Hmm, I guess this causes a rats tail. Can you try this:
```sh
apt install tiny-initramfs
```
Hmm, nasty, this seems to have changed after the dependency has been removed in between and then re-added, because some people found it bad that no initramfs is pulled in automatically otherwise, when using `pi-gen` 🤔.

I'll run some tests, but either we need to remove the option to install a 64-bit kernel on 32-bit images, or we need to host the package ourselves with metadata changed to be `armhf`.

MichaIng, 2025-02-13:
Okay, works now on my RPi Zero W. Live patch merged.

Lycidias93, 2025-02-13:
Works now, thank you all for the quick fix!