MichaIng/DietPi #7404: Wifi driver doesn't work on Tinkerboard after fresh installation of Kodi

rgabbo, 2025-02-12:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.10
- Distro version | bookworm
- Kernel version | Linux DietPi 6.6.54-current-rockchip
- SBC model | ASUS Tinkerboard
- Power supply used | 5V 1A 
- SD card used | SanDisk 32gb

#### Additional Information (if applicable)
- Software title | Kodi
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Install dietpi and run first start
2. See that wifi is working as expected
3. Install kodi with dietpi-software

#### Expected behaviour

- Wifi shoud work after installation

#### Actual behaviour

- Wifi doesn't work and dietpi-config says that no adapter is available

rgabbo, 2025-02-15:
Indeed, downloading the image again fixed the issue. I had downloaded the old one last Monday, but the checksum was different, so I probably did it before the update.

closed 2025-02-15 by rgabbo

MichaIng, 2025-02-15:
Okay that is great. I thought the new kernel might be the issue, but then it was something different.