MichaIng/DietPi #7405: persistent reboot required after upgrade to DietPI 9.10, stays on old 5.10 kernel

schlomo, 2025-02-12:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.10`
- Distro version | `bookworm 1`
- Kernel version | `Linux XXXXXXXX 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | 5V 3A, never had an issue with it
- SD card used | SanDisk Ultra

#### Additional Information (if applicable)
- Bug report ID | `4f374398-466c-407b-9b61-d045605257ed`

#### Steps to reproduce
1. Use `dietpi-update` to upgrade. previous version was a little bit older (don't remember, sorry)
2. reboot

#### Expected behaviour
- reboot as indicated
- reboot hint should disappear
- kernel should be > `5.10`
- 

#### Actual behaviour
[image]
- repeated reboots don't resolve this
- kernel is `5.10.103-v8+`

#### Extra details
- Initially I also ran into the problem described in #7402, but then after a second attempt the `linux-image-rpi-v8` could be installed.
- `dietpi-config` → Advanced → RPi kernel choice fails with
   ```
   linux-image-6.6.74+rpt-rpi-v8:arm64 : Depends: initramfs-tools:arm64 (>= 0.120+deb8u2) or linux-initramfs-tool:arm64
   ```
- This is actually caused by `[FAILED] DietPi-Set_software | APT install linux-image-rpi-v8 raspi-firmware raspi-utils-core tiny-initramfs` and specifically the installation of `tiny-initramfs` which seems to have a conflict with `linux-image-6.6.74+rpt-rpi-v8` via the `initramfs-tools` package
- My APT sources:
    ```
    # grep '^deb' /etc/apt/sources.list /etc/apt/sources.list.d/*
    /etc/apt/sources.list:deb http://raspbian.raspberrypi.com/raspbian/ bookworm main contrib non-free rpi
    /etc/apt/sources.list.d/dietpi.list:deb https://dietpi.com/apt bookworm main rpi4
    /etc/apt/sources.list.d/dietpi.list:deb https://dietpi.com/apt all rpi
    /etc/apt/sources.list.d/raspi.list:deb https://archive.raspberrypi.com/debian/ bookworm main
    ```

#### Resolution / Workaround
- Manually installing `linux-image-rpi-v8 raspi-firmware raspi-utils-core` and rebooting seems to have fixed the problem, now the kernel version is `6.6.74+rpt-rpi-v8`

MichaIng, 2025-02-12:
Linux 5.10? Is this a BerryBoot image? Otherwise it should be Linux 6.1 since a long time 🤔.

Please show this:
```sh
ls -l /lib/modules /boot
lsblk
```

schlomo, 2025-02-12:
This system is quite old, I always kept updating it successfully so far (big thanks for long term working software BTW!!!)

```
# ls -l /lib/modules /boot ; lsblk
/boot:
total 36470
lrwxrwxrwx 1 root root      20 Feb 12 15:13 cmdline.txt -> firmware/cmdline.txt
-rw-r--r-- 1 root root  229751 Jan 27 18:19 config-6.6.74+rpt-rpi-v7l
-rw-r--r-- 1 root root  239262 Jan 27 18:19 config-6.6.74+rpt-rpi-v8
lrwxrwxrwx 1 root root      19 Feb 12 15:13 config.txt -> firmware/config.txt
drwxr-xr-x 4 root root    4096 Feb 12 15:29 dietpi
-rwxr-xr-x 1 root root    8833 Jul 12  2019 dietpi-README.md
-rw-r--r-- 1 root root   12310 Mar 25  2024 dietpi.txt
drwxr-xr-x 6 root root    5632 Jan  1  1970 firmware
-rw-r--r-- 1 root root 9756359 Feb 12 19:11 initrd.img-6.6.74+rpt-rpi-v7l
-rw-r--r-- 1 root root 9811543 Feb 12 19:11 initrd.img-6.6.74+rpt-rpi-v8
-rw-r--r-- 1 root root      83 Jan 27 18:19 System.map-6.6.74+rpt-rpi-v7l
-rw-r--r-- 1 root root      83 Jan 27 18:19 System.map-6.6.74+rpt-rpi-v8
-rw-r--r-- 1 root root 7887536 Jan 27 18:19 vmlinuz-6.6.74+rpt-rpi-v7l
-rw-r--r-- 1 root root 9362120 Jan 27 18:19 vmlinuz-6.6.74+rpt-rpi-v8

/lib/modules:
total 8
drwxr-xr-x 3 root root 4096 Feb 12 15:27 6.6.74+rpt-rpi-v7l
drwxr-xr-x 3 root root 4096 Feb 12 19:10 6.6.74+rpt-rpi-v8
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0 465.8G  0 disk
└─sda1        8:1    0 465.8G  0 part /mnt/TV
mmcblk0     179:0    0  29.7G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /boot/firmware
└─mmcblk0p2 179:2    0  29.5G  0 part /
```

This is after my *manual* fix of `apt install install linux-image-rpi-v8 raspi-firmware raspi-utils-core`

schlomo, 2025-02-12:
Maybe some previous upgrade already failed to upgrade the kernel from 5.10 to 6.1 and I didn't notice that

MichaIng, 2025-02-14:
@schlomo so the issue is solved?

So you did the kernel migration, right? Did you select an additional kernel to be installed? And was `linux-image-rpi-v8` installed before, or automatically selected? Because some things do not fit together the way it should be:

1. The original kernel version was `5.10.103-v8+`, in which case `linux-image-rpi-v8` should be selected automatically/forcefully to be installed, as direct compatible successor.
2. Since the kernel arch was `aarch64`, `arm_64bit=0` was obviously not passed via `config.txt`, in which case `linux-image-rpi-v7l` should not be offered as optional kernel. But it is installed. Did you do that manually before, or was it unintentionally offered with the menu?
3. I am still confused why you had Linux 5.10 running in the first place, and not Linux 6.1. Did you set the old kernel packages on hold or so?
    ```sh
    apt policy raspberrypi-kernel
    apt-merk showhold
    ```

The `tiny-initramfs` dependency issues has been solved in the meantime, also via live patch: https://github.com/MichaIng/DietPi/commit/d1b9e1d

MichaIng, 2025-06-02:
Closing this issue. Feel free to reopen if still relevant.

closed 2025-06-02 by MichaIng