MichaIng/DietPi #7412: install of Realtek r8152 borks system - will not boot after restart

bhouseski, 2025-02-15:
### Creating a bug report/issue
- I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.10.0
- Distro version | `bookworm`
- Kernel version | `Linux DietPi 6.1.84-vendor-rk35xx #1 SMP Tue Dec 24 13:36:04 UTC 2024 aarch64 GNU/Linux
- SBC model | Rock5B (aarch64)
- Power supply used | 5V 5A 
- SD card used | none - nvme

#### Additional Information (if applicable)
- Software title |
- Was the software title installed freshly or updated/migrated
- Can this issue be replicated on a fresh installation of DietPi yes
 no
- Bug report ID | 

#### Steps to reproduce

1. Attempt to install Realtek R8152 driver for a 5Gb USB Ethernet adapter.  
1a. instal kernel header - apt install linux-headers-vendor-rk35xx=25.02.0-trunk-dietpi1
1b. install dkms - apt install dkms
1c. git clone driver - git clone https://github.com/awesometic/realtek-r8152-dkms
1d. install driver - ./dkms-install.sh
2. restart system

#### Expected behaviour
system should restart and boot normally

#### Actual behaviour
system does not boot.  no console when display connected.  cannot ssh into system

#### Extra details
if I input the below commands before reboot the system will reboot and the USB drive will work, however other issues present itself with the hold command.

apt install linux-{image,dtb,headers}-vendor-rk35xx=24.11.0-trunk-dietpi1
apt-mark hold linux-{image,dtb,headers}-vendor-rk35xx

MichaIng, 2025-02-16:
Sadly it is a known issue. It has been reported at Armbian already, but so far it seems to not bother enough people. It is sadly beyond my knowledge, at a level deep in the kernel/module without even any real error message, other than that loading the module failed: https://forum.armbian.com/topic/49378-kernel-oops-after-zfs-module-and-6184-vendor-rk35xx/

I'll do a rebuild based on latest Armbian/Radxa kernel sources. It is btw not Armbian's fault, it is within the Radxa kernel sources they are based on, potentially even the Rockchip kernel sources themselves. Such kind of errors, caused by clear bugs in the kernel, will only go away once we switch to mainline Linux. While some hardware features may be limited, this is btw already possible now:
```sh
apt install linux-(image,dtb)-current-rockchip64
```
See also the related request: #7337

bhouseski, 2025-02-16:
when I try apt install linux-(image,dtb)-current-rockchip64 I get an error message, when I try apt install linux-{image,dtb}-current-rockchip64 the image and dtb install, but bricks the system on reboot.  I tried the process suggested in #7337, but this also bricks my system.  this is on a fresh install of DietPi for my Rock5B.

MichaIng, 2025-02-16:
Oh sorry, I see now the migration there did not went well either, but he tested new images generated with mainline kernel and bootloader installed OOTB. I guess the bootloader needs to be updated as well:
```sh
apt install linux-(image,dtb,headers)-current-rockchip64 linux-u-boot-rock-5b-current
/boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
```

bhouseski, 2025-02-20:
now working well, thanks again!

MichaIng, 2025-02-21:
@bhouseski 
Just to be sure, the migration to mainline Linux with flashing new U-Boot worked well? So far so good, let me know if you face any issues/limitations with this kernel. But I think we will move over and offer a migration with DietPi v9.12 then (DietPi v9.11 is for Pi-hole v6 support explicitly, released at latest tomorrow), for all RK3588 SBCs with mainline kernel support.

bhouseski, 2025-02-22:
Yes, the steps you posted above worked, and the Rock5B seems to be running well.  The 5Gb Ethernet adapter is working well too.  Everything t I have done with it since is working as I believe it should too (other than HWA for the GPU, but I haven’t had time to dig into that one)