MichaIng/DietPi #7413: DietPi-Software | Fails to install fail2ban

Stream5710, 2025-02-16:
#### Details:
- Date           | 2025-02-16 16:50:03
- Bug report     | ff78fac2-c052-4347-9627-06debb453942
- Program name   | DietPi-Software
- Command        | `curl -sSf https://raw.githubusercontent.com/fail2ban/fail2ban/master/config/filter.d/dropbear.conf /etc/fail2ban/filter.d/dropbear.local`
- Exit code      | 3
- DietPi version | v9.10.0 (MichaIng/master)
- Distro version | bookworm (ID=7,RASPBIAN=0)
- Kernel version | `Linux DietPi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Architecture   | `armhf`
- Hardware model | RPi 2 Model B (armv7l) (ID=2)
- Power supply   | Anker PowerPort Strip PD 6 (USB-A port - 5V 2.4A)
- SD card        | Transcend 16GB
#### Steps to reproduce:

1. Connect to the Raspberry Pi with SSH
2. `sudo dietpi-software`
3. Select fail2ban and install
#### Expected behaviour:

- Fail2ban installs with no errors
#### Actual behaviour:

- At `Step: Installing Fail2Ban: prevents brute-force attacks with ip ban`, after `mkdir -p /etc/fail2ban/fail2ban.d /etc/fail2ban/filter.d
`, curl outputs contents of the downloaded file and throws `curl: (3) URL using bad/illegal format or missing URL`, failing installation
#### Extra details:

- This is for Dropbear but it happens with OpenSSH too
#### Additional logs:
```
# Fail2Ban filter for dropbear
#
# NOTE: The regex below is ONLY intended to work with a patched
# version of Dropbear as described here:
# http://www.unchartedbackwaters.co.uk/pyblosxom/static/patches
#            ^%(__prefix_line)sexit before auth from <HOST>.*\s*$
#
# The standard Dropbear output doesn't provide enough information to
# ban all types of attack.  The Dropbear patch adds IP address
# information to the 'exit before auth' message which is always
# produced for any form of non-successful login. It is that message
# which this file matches.
#
# More information: http://bugs.debian.org/546913

[INCLUDES]

# Read common prefixes. If any customizations available -- read them from
# common.local
before = common.conf

[Definition]

_daemon = dropbear

prefregex = ^%(__prefix_line)s(?:\[\d+\] \w{2,3} [\d:\s]+)?<F-CONTENT>(?:[Ll]ogin|[Bb]ad|[Ee]xit).+</F-CONTENT>$

failregex = ^[Ll]ogin attempt for nonexistent user (?:'<F-USER>.*</F-USER>' )?from <HOST>:\d+$
            ^[Bb]ad (?:PAM )?password attempt for '<F-USER>.+</F-USER>' from <HOST>(?::\d+)?$
            ^[Ee]xit before auth from \<?<ADDR>:\d+\>?: (?:\([^\)]*\): )?Max auth tries reached - user '<F-USER>.+</F-USER>'\s*$
            ^[Ee]xit before auth \(user '.+', \d+ fails\): Max auth tries reached - user '<F-USER>.+</F-USER>' from <HOST>:\d+\s*$

ignoreregex = 

journalmatch = _SYSTEMD_UNIT=dropbear.service + _COMM=dropbear

# DEV Notes:
#
# The first two regexs here match the unmodified dropbear messages. It isn't
# possible to match the source of the 'exit before auth' messages from dropbear
# as they don't include the "from <HOST>" bit.
#
# The second last failregex line we need to match with the modified dropbear.
#
# For the second regex the following apply:
#
# http://www.netmite.com/android/mydroid/external/dropbear/svr-authpam.c
# http://svn.dd-wrt.com/changeset/16642#file64
#
# http://svn.dd-wrt.com/changeset/16642/src/router/dropbear/svr-authpasswd.c
#
# Author: Francis Russell
#         Zak B. Elep
curl: (3) URL using bad/illegal format or missing URL
```

MichaIng, 2025-02-16:
Many thanks for reporting!

Embarrassing, fix is here: https://github.com/MichaIng/DietPi/commit/05da889
I'll push a live patch as well.

MichaIng, 2025-02-16:
Live patch merged as well: #7414

closed 2025-02-16 by MichaIng

Stream5710, 2025-02-16:
It worked, thank you for quick response!