MichaIng/DietPi #7417: mali450 GPU not working / accelerated in chromium-kiosk (NanoPi-K2)

BananaAcid, 2025-02-19:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues, but I might not have been clever enough to figure out if there was a related issue

#### Required Information
- DietPi version
```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applied'
```
- Distro version `bookworm`
- Kernel version `Linux DietPi 6.6.65-current-meson64 #1 SMP PREEMPT Wed Dec 11 15:13:43 UTC 2024 aarch64 GNU/Linux`
- SBC model `NanoPi K2 (aarch64)`
- Power supply used `100W USAMS GAN 3 (Pluggify), 5V 20A USB-C to micro-USB-A`
- SD card used `Toshiba Exceria HC1 U3, 32GB`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
`DietPi was freshly instgalled, Chromium-Kiosk was selected (and chromium had to be installed from dietpi-software afterwards)`
- Can this issue be replicated on a fresh installation of DietPi?
`Yes`

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Do the regular DietPi SD-card flashing
2. configure to start chromium kiosk
3. edit the chormium-autolaunch.sh to not incldude the --kiosk param (since about:gpu does not seem to be a working url if set up for kiosk, but can be entered if not using the kiosk param)
4. see that no GPU support is available

#### Expected behaviour

- GPU Acceleration should be available using opengl

#### Actual behaviour

- no gpu found on the about:gpu page

#### Extra details

I have no idea if anything of this makes sense to anyone, but anyhow (LIMA seems to be the mali450 driver)
```
$ journalctl | grep gpu
Feb 16 04:17:02 DietPi kernel: lima d00c0000.gpu: gp - mali450 version major 0 minor 0
Feb 16 04:17:02 DietPi kernel: lima d00c0000.gpu: pp0 - mali450 version major 0 minor 0
Feb 16 04:17:02 DietPi kernel: lima d00c0000.gpu: pp1 - mali450 version major 0 minor 0
Feb 16 04:17:02 DietPi kernel: lima d00c0000.gpu: pp2 - mali450 version major 0 minor 0
Feb 16 04:17:03 DietPi kernel: lima d00c0000.gpu: l2 cache 8K, 4-way, 64byte cache line, 128bit external bus
Feb 16 04:17:03 DietPi kernel: lima d00c0000.gpu: l2 cache 64K, 4-way, 64byte cache line, 128bit external bus
Feb 16 04:17:03 DietPi kernel: lima d00c0000.gpu: bus rate = 166666667
Feb 16 04:17:03 DietPi kernel: lima d00c0000.gpu: mod rate = 24000000
Feb 16 04:17:03 DietPi kernel: lima d00c0000.gpu: error -ENODEV: _opp_set_regulators: no regulator (mali) found
Feb 16 04:17:03 DietPi kernel: [drm] Initialized lima 1.1.0 20191231 for d00c0000.gpu on minor 1
```

MichaIng, 2025-06-02:
Sorry for the late reply. Does this issue still persist with latest Linux 6.12?

Indeed, hardware acceleration for this GPU should be there. In case, does it help to pull newer Mesa from backports?
```sh
cat << '_EOF_' > /etc/apt/preferences.d/dietpi-mesa
Package: src:mesa
Pin: release a=bookworm-backports
Pin-Priority: 500
_EOF_
```

closed 2025-06-22 by MichaIng

MichaIng, 2025-08-16:
There should be package upgrades offered now:
```sh
apt update
apt upgrade
```