MichaIng/DietPi #7419: [DietPi-VPN] - ProtonVpn - Name in place of Domain

black00019, 2025-02-19:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
When choosing a server for Protonvpn, the "Domain" field on the Protonvpn API page is listed

#### Describe the solution you'd like:
It would be better to have "Name" instead of "Domain"

[image]

MichaIng, 2025-04-19:
I agree this would be helpful. We can add name and load additionally, to the right side of the menu. Those whiptail menus have two fields. The left one is returned as output and needs to be the domain so that we can apply it to the OVPN config. But the right field is currently empty and can be well used for additional info provided by VPN providers.

MichaIng, 2026-10-26:
Done in #8399: Server name and city are now shown in the server selection menu. Let me know if there is any other interesting info.

closed 2026-10-26 by MichaIng