MichaIng/DietPi #7420: HomeAssistant matter-server

Zer0x00, 2025-02-20:
The matter-server docker image is a communication bridge between HomeAssistant and the Matter protocol / Matter compatible devices.  
[Running the image](https://github.com/home-assistant-libs/python-matter-server/blob/main/docs/docker.md) works fine but trying to pair a device will result in the following error:

```
matter-server  | 2025-02-20 01:34:45.644 (MainThread) INFO [matter_server.server.server] Matter Server successfully initialized.
matter-server  | 2025-02-20 01:36:33.527 (MainThread) INFO [matter_server.server.device_controller] Starting Matter commissioning with code using Node ID 1.
matter-server  | 2025-02-20 01:36:43.564 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40004i with Node: <0000000000000000, 0> S:0 M:240430285] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:36:51.028 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:37:00.796 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40005i with Node: <0000000000000000, 0> S:0 M:240430286] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:37:03.534 (Dummy-2) CHIP_ERROR [chip.native.CTL] Discovery timed out
matter-server  | 2025-02-20 01:37:08.523 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:37:18.495 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40006i with Node: <0000000000000000, 0> S:0 M:240430287] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:37:26.019 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:37:35.871 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40007i with Node: <0000000000000000, 0> S:0 M:240430288] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:37:43.517 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:37:53.845 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40008i with Node: <0000000000000000, 0> S:0 M:240430289] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:38:01.013 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:38:11.108 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40009i with Node: <0000000000000000, 0> S:0 M:240430290] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:38:18.507 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:38:28.451 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40010i with Node: <0000000000000000, 0> S:0 M:240430291] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:38:36.004 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:38:46.210 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40011i with Node: <0000000000000000, 0> S:0 M:240430292] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:38:53.500 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:39:03.682 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40012i with Node: <0000000000000000, 0> S:0 M:240430293] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:39:10.996 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:39:21.397 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40013i with Node: <0000000000000000, 0> S:0 M:240430294] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:39:28.493 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:39:38.295 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40014i with Node: <0000000000000000, 0> S:0 M:240430295] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:39:45.990 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:39:56.293 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40015i with Node: <0000000000000000, 0> S:0 M:240430296] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:40:03.488 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
matter-server  | 2025-02-20 01:40:13.323 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:40016i with Node: <0000000000000000, 0> S:0 M:240430297] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
matter-server  | 2025-02-20 01:40:20.984 (Dummy-2) CHIP_ERROR [chip.native.SC] PASESession timed out while waiting for a response from the peer. Expected message type was 33
```

Depending on whether the device is connected via WiFi or Ethernet, the following settings should be made:

```bash
sysctl -w net.ipv6.conf.wlan0.accept_ra=1
sysctl -w net.ipv6.conf.wlan0.accept_ra_rt_info_max_plen=64
```

or

```bash
sysctl -w net.ipv6.conf.eth0.accept_ra=1
sysctl -w net.ipv6.conf.eth0.accept_ra_rt_info_max_plen=64
```

Should we apply those sysctl variables during the installation of HomeAssistant, only document them or neither of both?

MichaIng, 2025-04-19:
Hmm, are you sure these both are really always required?
- `accept_ra=1` is the default. It accepts RAs in that case, as long as `ip_forwarding` is not enabled. If it is, it guesses the device is a router and hence does not accept RAs from other routers.
- `accept_ra=2` is required if you configure a (WiFi) hotspot, shared internet connection via VPN or similar, but without the device being the router. `accept_ra=1` would lead to RAs being ignored and hence the global IPv6 prefix/gateway is not updated. This also makes it impossible for us to apply `accept_ra=1` forcefully with all HA installs, to not break IPv6 connectivity in such cases only because a particular HA app seems to require it (for unknown reasons so far).
- `accept_ra_rt_info_max_plen` defines the maximum prefix length in RAs. I am not sure why this would need to be adjusted? If a router in your network provides a larger prefix than what you configure, this would break IPv6 connectivity as well.

It somehow seems that you need to apply those settings as of your particular home network setup, at least I cannot imagine why a specific IPv6 route would affect the HA container but not the host system itself 🤔. Let me have a closer look what matter really is.

MichaIng, 2025-04-19:
Okay I understood now that Matter is a smart home device communication standard which makes use of IPv6 and RAs: https://github.com/home-assistant-libs/python-matter-server/blob/main/docs/os_requirements.md

As said, `accept_ra=1` is the default anyway, and when installing a WiFi hotspot or VPN with shared internet via `dietpi-software`, `accept_ra=2` is set automatically, required for Matter as well im this case. So that setting must not be touched.

I am not sure about the `accept_ra_rt_info_max_plen`. If `accept_ra` is enabled, it seems to default to 0, otherwise to -1. However, routes in RAs are accepted by default. So either 0 means accept all routes and -1 none, or I misunderstand the effect of the setting on the default route/gateway provided by the router.

What are the default value of both settings in your case?

Zer0x00, 2025-04-19:
I can't say for sure what was the default before since I've already set the above values. I only know that I've struggled multiple hours because of those configs not set. After setting those the docker container ran just fine.

MichaIng, 2025-04-19:
Maybe when you need to do a reboot anyway, temporarily take out of the way the script/config which adjusts those, to check back the defaults.

Zer0x00, 2025-04-19:
Default values:
`net.ipv6.conf.eth0.accept_ra` is set to 1
`net.ipv6.conf.eth0.accept_ra_rt_info_max_plen` is set to 0

As you said `accept_ra` has indeed a default of 1. Therefore setting `accept_ra_rt_info_max_plen` to 64 is sufficient.

MichaIng, 2025-04-21:
Can you replicate that leaving/setting back `accept_ra_rt_info_max_plen` to 0 causes the issue?

Zer0x00, 2025-04-21:
Yes, I can replicate that leaving `accept_ra_rt_info_max_plen` to 0 causes the issue.
Here's the log with `accept_ra_rt_info_max_plen` set to 0:
```
2025-04-21 16:28:21.036 (MainThread) INFO [matter_server.server.stack] Initializing CHIP/Matter Logging...
2025-04-21 16:28:21.037 (MainThread) INFO [matter_server.server.stack] Initializing CHIP/Matter Controller Stack...
[1745245701.361293][1:1] CHIP:CTL: Setting attestation nonce to random value
[1745245701.373649][1:1] CHIP:CTL: Setting CSR nonce to random value
[1745245701.416840][1:1] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1745245701.419423][1:1] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /data/chip_factory.ini
[1745245701.431655][1:1] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /data/chip_config.ini
[1745245701.432512][1:1] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /data/chip_counters.ini
2025-04-21 16:28:21.469 (MainThread) INFO [chip.storage] Initializing persistent storage from file: /data/chip.json
2025-04-21 16:28:21.469 (MainThread) INFO [chip.storage] Loading configuration from /data/chip.json...
[1745245701.467455][1:1] CHIP:DL: Wrote settings to /data/chip_counters.ini
[1745245701.467519][1:1] CHIP:DL: NVS set: chip-counters/reboot-count = 18 (0x12)
[1745245701.468076][1:1] CHIP:DL: Got Ethernet interface: eth0
[1745245701.468311][1:1] CHIP:DL: Found the primary Ethernet interface:eth0
[1745245701.468611][1:1] CHIP:DL: Failed to get WiFi interface
[1745245701.468630][1:1] CHIP:DL: Failed to reset WiFi statistic counts
2025-04-21 16:28:21.744 (MainThread) INFO [chip.CertificateAuthority] Loading certificate authorities from storage...
2025-04-21 16:28:21.744 (MainThread) INFO [chip.CertificateAuthority] New CertificateAuthority at index 1
2025-04-21 16:28:21.749 (MainThread) INFO [chip.CertificateAuthority] Loading fabric admins from storage...
2025-04-21 16:28:21.749 (MainThread) INFO [chip.FabricAdmin] New FabricAdmin: FabricId: 0x0000000000000001, VendorId = 0xFFF1
2025-04-21 16:28:21.750 (MainThread) INFO [matter_server.server.stack] CHIP Controller Stack initialized.
2025-04-21 16:28:21.751 (MainThread) INFO [matter_server.server.server] Starting the Matter Server...
2025-04-21 16:28:21.762 (MainThread) INFO [matter_server.server.helpers.paa_certificates] Fetching the latest PAA root certificates from DCL.
2025-04-21 16:28:24.968 (MainThread) INFO [matter_server.server.helpers.paa_certificates] Fetched 66 PAA root certificates from DCL.
2025-04-21 16:28:24.969 (MainThread) INFO [matter_server.server.helpers.paa_certificates] Fetching the latest PAA root certificates from Git.
2025-04-21 16:28:25.399 (MainThread) INFO [matter_server.server.helpers.paa_certificates] Fetched 2 PAA root certificates from Git.
2025-04-21 16:28:25.402 (MainThread) INFO [chip.FabricAdmin] Allocating new controller with CaIndex: 1, FabricId: 0x0000000000000001, NodeId: 0x000000000001B669, CatTags: []
2025-04-21 16:28:25.574 (Dummy-2) CHIP_ERROR [chip.native.DIS] Failed to advertise records: src/inet/UDPEndPointImplSockets.cpp:421: OS Error 0x02000065: Network is unreachable
2025-04-21 16:28:25.578 (Dummy-2) CHIP_ERROR [chip.native.DIS] Failed to advertise records: src/inet/UDPEndPointImplSockets.cpp:421: OS Error 0x02000065: Network is unreachable
2025-04-21 16:28:25.582 (Dummy-2) CHIP_ERROR [chip.native.DL] Long dispatch time: 178 ms, for event type 2
2025-04-21 16:28:25.644 (MainThread) INFO [matter_server.server.vendor_info] Loading vendor info from storage.
2025-04-21 16:28:25.655 (MainThread) INFO [matter_server.server.vendor_info] Loaded 308 vendors from storage.
2025-04-21 16:28:25.656 (MainThread) INFO [matter_server.server.vendor_info] Fetching the latest vendor info from DCL.
2025-04-21 16:28:26.073 (MainThread) INFO [matter_server.server.vendor_info] Fetched 307 vendors from DCL.
2025-04-21 16:28:26.074 (MainThread) INFO [matter_server.server.vendor_info] Saving vendor info to storage.
2025-04-21 16:28:26.111 (MainThread) INFO [matter_server.server.device_controller] Loaded 1 nodes from stored configuration
2025-04-21 16:28:26.147 (MainThread) INFO [matter_server.server.server] Matter Server successfully initialized.
2025-04-21 16:28:26.740 (MainThread) INFO [matter_server.server.device_controller.mdns] <Node:4> Discovered on mDNS
2025-04-21 16:28:26.741 (MainThread) INFO [matter_server.server.device_controller] <Node:4> Setting-up node...
2025-04-21 16:28:27.901 (MainThread) INFO [matter_server.server.device_controller.mdns] <Node:4> Discovered on mDNS
2025-04-21 16:28:37.112 (Dummy-2) CHIP_ERROR [chip.native.EM] <<5 [E:49694i with Node: <0000000000000000, 0> S:0 M:89159706] (U) Msg Retransmission to 0:0000000000000000 failure (max retries:4)
```

vs. `accept_ra_rt_info_max_plen` set to 64:
```
2025-04-21 16:30:06.817 (MainThread) INFO [matter_server.server.stack] Initializing CHIP/Matter Logging...
2025-04-21 16:30:06.818 (MainThread) INFO [matter_server.server.stack] Initializing CHIP/Matter Controller Stack...
[1745245807.151272][1:1] CHIP:CTL: Setting attestation nonce to random value
[1745245807.161265][1:1] CHIP:CTL: Setting CSR nonce to random value
[1745245807.199228][1:1] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1745245807.201839][1:1] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /data/chip_factory.ini
[1745245807.214438][1:1] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /data/chip_config.ini
[... cut ...]
2025-04-21 16:30:08.919 (MainThread) INFO [matter_server.server.device_controller] <Node:4> Setting-up node...
2025-04-21 16:30:09.613 (MainThread) INFO [matter_server.server.device_controller.mdns] <Node:4> Discovered on mDNS
2025-04-21 16:30:10.154 (MainThread) INFO [matter_server.server.device_controller] <Node:4> Setting up attributes and events subscription.
2025-04-21 16:30:19.476 (MainThread) INFO [matter_server.server.device_controller] <Node:4> Subscription succeeded with report interval [1, 60]
```

MichaIng, 2025-04-21:
Yeah, it makes sense now, but as said nothing that can be done by us. If it was not prominent enough for you in the Matter server docs, probably suggest it being mentioned there among install instructions and/or FAQ/troubleshooting sections which people usually will see.

closed 2025-04-21 by MichaIng