MichaIng/DietPi #7422: Pihole fresh install

denis-g, 2025-02-21:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bookworm 0`
- Kernel version | `Linux DietPi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linu`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | Pihole
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. fresh DietPi `v9.10`
2. `dietpi-software install 93`

#### Expected behaviour

- install without errors

#### Actual behaviour

- errors
```
[FAILED] File does not exist or cannot be written to by current user                                                 │
│                                                                                                                      │
│ Please verify the existence of the file $3                                                                           │
│         /etc/pihole/pihole-FTL.conf                                                                                  │
│                                                                                                                      │
│ Retry with proper permissions or apply the setting manually:                                                         │
│         BLOCKINGMODE=NULL
```

```
  - Command: a2enmod headers                                                                                          │
│  - Exit code: 127                                                                                                    │
│  - DietPi version: v9.10.0 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 7                                  │
│  - Error log:                                                                                                        │
│ /boot/dietpi/func/dietpi-globals: line 904: a2enmod: command not found
```

#### Extra details

- ...

Joulinar, 2025-02-21:
This is expected as PiHole v6 has been released some days ago but our scripts did not take this into account actually. Something we are heavily working on.

In meantime you could creating the file manually or just skip the step.

MichaIng, 2025-02-21:
I have #7421 somewhat ready, but testing a few things in parallel. Would be great if someone could give it a try, the more the better:
```sh
G_DEV_BRANCH pihole-v6
```
Fresh install of Pi-hole v6. But the update will also migrate existing Pi-hole v6 instances, v5 was previously installed via `dietpi-software` and upgraded via `pihole -up`.

Some more tests of this would be good. I then aim to release DietPi v9.11 as hotfix release.

Joulinar, 2025-02-21:
@MichaIng just for my understanding. What about people who are still on v5? Are we going to migrate them as well to v6?

MichaIng, 2025-02-21:
Nope, instead we need to move forward this migration patch until we either think that Pi-hole v6 is stable enough to enforce the upgrade or until we think everyone did the upgrade already. But right now, there are quite some issue reports, and I myself fixed two bugs in Pi-hole just yesterday. So I would recommend production systems to wait at least for a week or two, before considering an update, to give some more time for hotfixes.

closed 2025-02-23 by MichaIng