MichaIng/DietPi #7423: [Dietpi-Config] Unable to connect to wifi with Static IP.

Maggus101, 2025-02-22:
### Creating a bug report/issue
- [Y] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9`
- Distro version | `bullseye 0`
- Kernel version | `Linux --- 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- SBC model | `RPi 3 Model B (armv7l)`
- Power supply used | `Official Rasp PSU`
- SD card used | `Sandisk`

#### Additional Information (if applicable)
- Software title | `dietpi-config`
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `8a3c9c74-cc9d-45e9-8fe5-f20d84f77235`

#### Steps to reproduce

1. `sudo dietpi-config` > `Network Options: Adapters` > Wifi > Static > Fillout correct network info > Apply

```
Feb 22 13:02:55 Helgen systemd[1]: Starting ifup for wlan0...
Feb 22 13:02:55 Helgen wpa_supplicant[16815]: Successfully initialized wpa_supplicant
Feb 22 13:02:55 Helgen ifup[16823]: RTNETLINK answers: File exists
Feb 22 13:02:55 Helgen ifup[16801]: ifup: failed to bring up wlan0
Feb 22 13:02:55 Helgen systemd[1]: ifup@wlan0.service: Main process exited, code=exited, status=1/FAILURE
Feb 22 13:02:55 Helgen wpa_supplicant[16820]: wlan0: CTRL-EVENT-SCAN-FAILED ret=-512 retry=1
Feb 22 13:02:55 Helgen wpa_supplicant[16820]: nl80211: deinit ifname=p2p-dev-wlan0 disabled_11b_rates=0
Feb 22 13:02:55 Helgen wpa_supplicant[16820]: p2p-dev-wlan0: CTRL-EVENT-TERMINATING
Feb 22 13:02:55 Helgen wpa_supplicant[16820]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Job for ifup@wlan0.service failed because the control process exited with error code.
See "systemctl status ifup@wlan0.service" and "journalctl -xe" for details.
[FAILED] DietPi-Config | systemctl restart ifup@wlan0

```

#### Expected behaviour
- Connect to wifi

#### Actual behaviour
- Not connecting

#### Extra details

It will connect via DHCP 

```
-- Journal begins at Sat 2025-02-22 12:24:58 GMT. --
Feb 22 13:12:14 Helgen systemd[1]: Stopping ifup for eth0...
[  OK  ] DietPi-Config | systemctl restart ifup@eth0
[ INFO ] DietPi-Config | systemctl restart ifup@wlan0, please wait...
-- Journal begins at Sat 2025-02-22 12:24:58 GMT. --
Feb 22 13:12:14 Helgen systemd[1]: Starting ifup for wlan0...
Feb 22 13:12:14 Helgen wpa_supplicant[19907]: Successfully initialized wpa_supplicant
Feb 22 13:12:14 Helgen dhclient[19915]: Internet Systems Consortium DHCP Client 4.4.1
Feb 22 13:12:14 Helgen ifup[19915]: Internet Systems Consortium DHCP Client 4.4.1
Feb 22 13:12:14 Helgen ifup[19915]: Copyright 2004-2018 Internet Systems Consortium.
Feb 22 13:12:14 Helgen ifup[19915]: All rights reserved.
Feb 22 13:12:14 Helgen ifup[19915]: For info, please visit https://www.isc.org/software/dhcp/
Feb 22 13:12:14 Helgen dhclient[19915]: Copyright 2004-2018 Internet Systems Consortium.
Feb 22 13:12:14 Helgen dhclient[19915]: All rights reserved.
Feb 22 13:12:14 Helgen dhclient[19915]: For info, please visit https://www.isc.org/software/dhcp/
Feb 22 13:12:14 Helgen dhclient[19915]:
Feb 22 13:12:14 Helgen dhclient[19915]: Listening on LPF/wlan0/b8:27:eb:45:e0:46
Feb 22 13:12:14 Helgen ifup[19915]: Listening on LPF/wlan0/b8:27:eb:45:e0:46
Feb 22 13:12:14 Helgen ifup[19915]: Sending on   LPF/wlan0/b8:27:eb:45:e0:46
Feb 22 13:12:14 Helgen ifup[19915]: Sending on   Socket/fallback
Feb 22 13:12:14 Helgen ifup[19915]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
Feb 22 13:12:14 Helgen dhclient[19915]: Sending on   LPF/wlan0/b8:27:eb:45:e0:46
Feb 22 13:12:14 Helgen dhclient[19915]: Sending on   Socket/fallback
Feb 22 13:12:14 Helgen dhclient[19915]: DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 7
Feb 22 13:12:15 Helgen wpa_supplicant[19910]: wlan0: Trying to associate with SSID 'A Link Between Worlds_iot'
Feb 22 13:12:15 Helgen wpa_supplicant[19910]: wlan0: Associated with fa:a7:31:ca:9c:ad
Feb 22 13:12:15 Helgen wpa_supplicant[19910]: wlan0: CTRL-EVENT-CONNECTED - Connection to fa:a7:31:ca:9c:ad completed [id=0 id_str=]
Feb 22 13:12:15 Helgen wpa_supplicant[19910]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Feb 22 13:12:15 Helgen wpa_supplicant[19910]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=DE
[  OK  ] DietPi-Config | systemctl restart ifup@wlan0

```

```
sudo ifup wlan0
RTNETLINK answers: File exists
ifup: failed to bring up wlan0
```

```
cat /etc/network/interfaces
# WiFi
allow-hotplug wlan0
iface wlan0 inet static
address 192.168.1.160
netmask 255.255.255.0
gateway 192.168.1.254
#dns-nameservers 192.168.1.254
pre-up iw dev wlan0 set power_save off
post-down iw dev wlan0 set power_save on
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

```

```
sudo journalctl -u ifup@wlan0
Feb 22 13:02:55  systemd[1]: Starting ifup for wlan0...
Feb 22 13:02:55  wpa_supplicant[16815]: Successfully initialized wpa_supplicant
Feb 22 13:02:55  ifup[16823]: RTNETLINK answers: File exists
Feb 22 13:02:55  ifup[16801]: ifup: failed to bring up wlan0
Feb 22 13:02:55  systemd[1]: ifup@wlan0.service: Main process exited, code=exited, status=1/FAILURE
Feb 22 13:02:55  wpa_supplicant[16820]: wlan0: CTRL-EVENT-SCAN-FAILED ret=-512 retry=1
Feb 22 13:02:55  wpa_supplicant[16820]: nl80211: deinit ifname=p2p-dev-wlan0 disabled_11b_rates=0
Feb 22 13:02:55  wpa_supplicant[16820]: p2p-dev-wlan0: CTRL-EVENT-TERMINATING
Feb 22 13:02:55  wpa_supplicant[16820]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Feb 22 13:02:56  wpa_supplicant[16820]: wlan0: CTRL-EVENT-TERMINATING
Feb 22 13:02:56  systemd[1]: ifup@wlan0.service: Failed with result 'exit-code'.
Feb 22 13:02:56  systemd[1]: Failed to start ifup for wlan0.
```

```
dpkg -l | grep dhcp
ii  isc-dhcp-client              4.4.1-2.3+deb11u2                armhf        DHCP client for automatically obtaining an IP address

```
Thank you for your help!

mdr92, 2025-02-28:
I just upgraded my Pi 5 from 9.9 to 9.11 and have lost network, I can't even open Network Options to switch it back as it errors 
`/boot/dietpi/dietpi-config: line 2094: ! : syntax error: operant expected (error token is "! ")`

My Pi Zero 2 and Pi 400 upgraded fine though but they were on 9.10 first. Did you upgrade and can you switch back to DHCP? Trying to work out if we have the same issue.

MichaIng, 2025-02-28:
Sorry for the missing reply on this issue. Let's check the latest error first.

The error is weird, as there should be `! 0` or `! 1` in any case, based on a `grep -cm1` call. So either `grep` is damaged, or `/etc/network/interfaces` is. Can you please run this manually and paste the output`?
```sh
grep -Ecm1 "^[[:blank:]]*(allow-hotplug|auto)[[:blank:]]+eth0" /etc/network/interfaces
```

mdr92, 2025-02-28:
Thanks for the quick reply, if that's for me I won't be able to sorry, I was burning the image to test with a clean install and overwrote the wrong card, luckily my Pi 5 was just a new test system but appreciate it doesn't help you debug. If there's a way I can grab a 9.9 image I can test again, mine was on wifi rather than ethernet.

MichaIng, 2025-02-28:
Oh I see just now the different accounts. No need to replicate, must have been some data corruption.

@Maggus101 your issue is still present? I think the problem is, with Ethernet and WiFi enabled both, that a default route is assigned for Ethernet, and cannot be assigned anymore for WiFi afterwards. We need to find time to finish the planned `dietpi-network` script to configure each network adapter individually. `dietpi-config` does not really support having both enabled with a static gateway defined.

Do you want to switch to WiFi, or really use both in some way? What should work is disabling Ethernet first, then switching WiFi static IP. Otherwise you'd currently need to edit `/etc/network/interfaces` manually. The `gateway` line with `static` IP can only exist once. Use it in for the adapter you use to access the Internet, and skip it for the other one, if it is LAN facing or such.

Maggus101, 2025-03-02:
Hi All, thanks for the responses.

A bit of context. My router (TP-Link Deco X20 Series) doesn't have a way to split the SSIDs into 5ghz/2.4ghz. The only way to do this is to turn on it's guest network, which I have done, and added my IOT devices to. I then realised that those devices couldn't access my pi-hole instance on dietpi, because the guest network isolates those devices. So my only option was to set up the dietpi on both wired for my main network, and wireless for my Guest network under sa different IP address, and had to set DNS1 as my main IP, DNS2 as my wireless.

I was able to get a workaround in place by keeping the Wifi using DHCP and using Address Reservation in my router settings. 

It's a pretty fringe case, so happy for you to close out unless you want to keep it open as a potential thing for `dietpi-network`

MichaIng, 2025-03-02:
Okay, I guess then the Ethernet adapter should be the one your Pi-hole gets Internet access through, right? In this case, change your `/etc/network/interfaces` like so:
```sh
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
allow-hotplug eth0
iface eth0 inet static
address 192.168.0.160
netmask 255.255.255.0
gateway 192.168.0.254
#dns-nameservers 192.168.1.254

# WiFi
allow-hotplug wlan0
iface wlan0 inet static
address 192.168.1.160
netmask 255.255.255.0
#dns-nameservers 192.168.1.254
pre-up iw dev wlan0 set power_save off
post-down iw dev wlan0 set power_save on
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```
as example if`192.168.0.0/24` to be the main network on Ethernet and `192.168.0.254` the router IP on that network. Basically, remove `gateway` from the WiFi block.

MichaIng, 2025-06-02:
Closing this issue. Feel free to reopen if still relevant.

closed 2025-06-02 by MichaIng