MichaIng/DietPi #7424: usb serial adaptor can not be accessed

hallabalooza, 2025-02-22:
Hi.

On a
```
root@DietPi:~# cat /sys/firmware/devicetree/base/model;echo
Raspberry Pi Model B Rev 2
root@DietPi:~# grep Revision /proc/cpuinfo
Revision        : 000d
```
with `DietPi ARMv6 Bookworm` I want to use the 2 serial usb adaptors listed in the following output:
```
root@DietPi:~# lsusb
Bus 001 Device 007: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 006: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
...
```
To be able to access them always with the same name I created a udev rule 
```
root@DietPi:~# head /usr/lib/udev/rules.d/99-usb-serial.rules
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="XXXXXXX", MODE="0666", SYMLINK+="usb_XXXXXXX"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="YYYYYYY", MODE="0666", SYMLINK+="usb_YYYYYYY"
```

therefore.
But unfortunately I can not connect to any of those adaptors.

- 
```
root@DietPi:~# picocom -b 9600 /dev/usb_XXXXXXX picocom v3.1

port is        : /dev/usb_XXXXXXX
flowcontrol    : none
baudrate is    : 9600
parity is      : none
databits are   : 8
stopbits are   : 1
...

FATAL: failed to add port: Filedes is not a tty
```
-
```
root@DietPi:~# runuser -u dietpi python3 /home/dietpi/test.py Traceback (most recent call last):
   File
"/home/dietpi/.local/lib/python3.11/site-packages/serial/serialposix.py",
line 398, in _reconfigure_port
     orig_attr = termios.tcgetattr(self.fd)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
termios.error: (22, 'Invalid argument')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
   File "/home/dietpi/test.py", line 5, in <module>
     with serial.Serial(port="/dev/usb_XXXXXXX", baudrate=9600, bytesize=serial.EIGHTBITS, parity=serial.PARITY_NONE,
stopbits=serial.STOPBITS_ONE) as vSerial:
...
serial.serialutil.SerialException: Could not configure port: (22, 'Invalid argument')
```
The pyserial/serialposix.py exception is commented with "# if a port is nonexistent but has a /dev file, it'll fail here".

In a hurry, I tested it with Fedora 41 Workstation without errors.
Another, more modern RasPi I first have to get and set it up.

But what could be the reason for a not existent port but an existing /dev file?

hallabalooza, 2025-02-25:
Hi.

I compared DietPi and Fedora.

In that process, I saw that both systems add/enable /dev/ttyUSB0 and ttyUSB1 files if I connect my 2 serial usb adaptors.
Additionally, on both systems, the serials for /dev/ttyUSB0 and ttyUSB1 include the expected “ATTR(S){serial}”’s "XXXXXXX" or "YYYYYYY" (“udevadm info -a /dev/ttyUSB?”, SUBSYSTEM(S)=”usb”).

But while on Fedora there is no SUBSYSTEM “tty” for the IR heads and the following UDEV rule works
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="XXXXXXX", MODE="0666", SYMLINK+="usb_XXXXXXX"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="YYYYYYY", MODE="0666", SYMLINK+="usb_YYYYYYY"
```

DietPi has this SUBSYSTEM and the rule has to be changed into
```
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="XXXXXXX", MODE="0666", SYMLINK+="usb_XXXXXXX"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="YYYYYYY", MODE="0666", SYMLINK+="usb_YYYYYYY"
```

(see the difference of the singular and plural forms of UDEV criterias (KERNEL. SUBSYSTEM, …) in [https://reactivated.net/writing_udev_rules.html#basic](https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Freactivated.net%2Fwriting_udev_rules.html%23basic&data=05%7C02%7C%7C1459f7b48b0c4aba48bd08dd5570e146%7Ce764c36b012e4216910d8fd16283182d%7C0%7C0%7C638760665479471039%7CUnknown%7CTWFpbGZsb3d8eyJFbXB0eU1hcGkiOnRydWUsIlYiOiIwLjAuMDAwMCIsIlAiOiJXaW4zMiIsIkFOIjoiTWFpbCIsIldUIjoyfQ%3D%3D%7C0%7C%7C%7C&sdata=pTmeo2G8XNDOXbX%2B6s0RfJYAJr2WLwmeNeQMoP9of0c%3D&reserved=0) and [https://reactivated.net/writing_udev_rules.html#hierarchy](https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Freactivated.net%2Fwriting_udev_rules.html%23hierarchy&data=05%7C02%7C%7C1459f7b48b0c4aba48bd08dd5570e146%7Ce764c36b012e4216910d8fd16283182d%7C0%7C0%7C638760665479493626%7CUnknown%7CTWFpbGZsb3d8eyJFbXB0eU1hcGkiOnRydWUsIlYiOiIwLjAuMDAwMCIsIlAiOiJXaW4zMiIsIkFOIjoiTWFpbCIsIldUIjoyfQ%3D%3D%7C0%7C%7C%7C&sdata=TtieiHQiZV4KWzK0cCufEON2PqyMBMnErhkxGV2Pr2Y%3D&reserved=0))

closed 2025-02-25 by hallabalooza

MichaIng, 2025-02-25:
Right, USB serial devices appear as `/dev/ttyUSB*`. Weird that in one case, the correct sysfs entries are in the USB subsystem, while in the other it is TTY subsystem. I wonder whether this is coming from different kernel configurations or from different kernel versions. I guess Fedora is Linux v6.12 at least?