MichaIng/DietPi #7428: Update to 9.10 broke GPIO

barrelltitor, 2025-02-23:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`s

```
cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='applied'
```

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bookworm 0
```
- Kernel version | `uname -a`
```
Linux octopi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
RPi Zero 2 W (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower)
5V 3A usb adapter
- SD card used | (EG: SanDisk ultra)
Samsung evo plus a2 v30
#### Additional Information (if applicable)

I'm using Octoprint in docker and I'm trying to install a filament sensor plugin, and I got this error

```

2025-02-23 19:27:20,812 - octoprint.server.api.settings - ERROR - Could not save settings for plugin Filament Sensor Reloaded (1.4.3)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/octoprint/server/api/settings.py", line 1246, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "/usr/local/lib/python3.10/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/octoprint/plugins/lib/python3.10/site-packages/octoprint_filamentreload/__init__.py", line 169, in on_settings_save
    self._setup_sensor()
  File "/octoprint/plugins/lib/python3.10/site-packages/octoprint_filamentreload/__init__.py", line 140, in _setup_sensor
    GPIO.add_event_detect(
RuntimeError: Failed to add edge detection

```

#### Steps to reproduce

1. Install octoprint
2. Install any filament sensor plugin
3. Set the GPIO

#### Expected behaviour

It should work

#### Actual behaviour

It doesn't work

#### Extra details

barrelltitor, 2025-02-24:
This is an issue from the kernel or firmware, but as this is a massive breaking change that will affect a lot of software using rpi gpio maybe a warning could be made for this, should be pretty easy to do

It's my fault for not taking a backup(though it's also dietpi's fault for trying to backup my nas that I mounted in /nas ??), but it could help others if a warning could be added.

# Fix for people with the same issue
The solution is to use lgpio instead, but this won't build without some work in containers at least, haven't tried native. For containers this needs to be done inside the container/image https://abyz.me.uk/lg/download.html for lgpio to actually build. Also remember to uninstall rpi gpio

No code changes need done, it seems to be an in place replacement

barrelltitor, 2025-02-26:
> Use the include/exclude filters to exclude mounts. There is not really a way to guess well which mount is intended to be backed up with the system backup, and which not, hence the filters.

Thank you! That's my fault for not reading the documentation. Was it on the backup screen and I just blindly sped through it? Might be a good idea to add the information there if it's not

> So as Python library, GPIO Zero is what RPi engineers suggests: https://www.raspberrypi.com/documentation/computers/os.html#use-gpio-from-python

That might be a good idea, but I was using an octoprint plugin that needed a drop in replacement so I don't have to change the code just to detect if a pin is high or low.  gpiozero seems to require a rewrite

I did not expect you to reply so fast. Dietpi is incredible and the work you're doing is great, thank you again!

MichaIng, 2025-02-28:
We even have an RPi.GPIO software option. I will check whether `rpi-lgpio`/`python3-rpi-lgpio` package is available on Bullseye already, and in case replace it. Although, it is really only the APT package, and only WebIOPi depends on it, which needs to be tested anyway. Pretty much a cleanup needed across all GPIO libraries, CLI tools and web interfaces, either migrating to modern backends, or removing them as obsolete.

JanFellner, 2025-04-04:
Just opend a ticket in MQTT-IO which also suffers from this issue.
As there has not been an update on this issue for a month.
Any idea when this will get fixed or how to temporarily circumvent it?

MichaIng, 2025-04-19:
To check back, the `python3-rpi-lgpio` alone without compiling `lgpio` or anuthing else manually works as well?