MichaIng/DietPi #7434: Upgrade from v9.10 to 9.11.2 fails | dpkg was interrupted

SOULV1CE, 2025-02-26:
#### Details:
- Date           | 2025-02-26 15:15:12
- Bug report     | a3fff781-9e74-4eb2-98a0-bca4147c85d2
- Program name   | DietPi-Update
- Command        | `apt-get -y --with-new-pkgs upgrade`
- Exit code      | 100
- DietPi version | v9.10.0 (MichaIng/master)
- Distro version | bookworm (ID=7,RASPBIAN=1)
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Architecture   | `armhf`
- Hardware model | RPi Zero W (armv6l) (ID=1)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

1. dietpi-update   : Run now to update DietPi from v9.10.0 to v9.11.2
#### Expected behaviour:

- Uprgrade should proceed without error
#### Actual behaviour:

- Upgrade fails due to error "dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem."
#### Extra details:

- N/A
#### Additional logs:
```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.

MichaIng, 2025-02-26:
Please run the suggested command and in case of failure post its output:
```sh
sudo dpkg --configure -a
```

SOULV1CE, 2025-02-26:
It appears that it was failing when trying to update Tailscale. During my initial attempt I was doing the update while connected via Tailscale - I tried to run the suggested command while connected via Tailscale but it still failed to continue with the update.

Then I connected locally without Tailscale and ran the suggested command which seems to have successfully updated Tailscale. I then ran the dietpi-update command again which now completed successfully as well.

`[ INFO ] DietPi-Update | APT upgrade, please wait...
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem. 
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -y --with-new-pkgs upgrade
[FAILED] DietPi-Update | Unable to continue, DietPi-Update will now terminate.

root@DietPi:~# dpkg --configure -a
Setting up tailscale (1.80.2) ...
root@DietPi:~# dietpi-update
`

MichaIng, 2025-02-26:
Interesting, now that you say it, I remember there was such a report in the past already, also with Tailscale already. But I was never able to replicate it. Let me check ...

SOULV1CE, 2025-02-26:
Huh...that's interesting.

For what it's worth I also have the following set regarding Tailscale:

- Key expiry disabled
- Tailscale SSH server running
- Advertising subnet routes

Maybe any of those things could be effecting it somehow?

SOULV1CE, 2025-02-27:
Yeah, I do use the embedded SSH server. It is not enabled by default.

MichaIng, 2025-02-28:
Can you test whether the same happens, when you disable the embedded SSH and use dedicated SSH server instead?

SOULV1CE, 2025-03-01:
I dont have another device to test with and the latest Tailscale update I got to complete successfully after running the suggested command. However, I can definitely do this test for the next Tailscale update if the issue is still open.

jalsco, 2025-03-02:
I have same problem going from 9.10.0 to 9.11.2
```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 9.11.2
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v9.10.0
[ INFO ] DietPi-Update | Latest version  : v9.11.2
```
When it failed I sent the reports automatically via the menus. 

I tried again after a week or so, and still get same problems.

jalsco, 2025-03-08:
Yesterday I tried the 9.10.0 to 9.11.2 upgrade on a DIFFERENT system ( Device model : NanoPi 6 series (aarch64) )  and it also failed on first attempt. Then the first RETRY and it succeeded. 

So with renewed optimism 😎  I went back to the system that failed the 9.10.0 to 9.11.2 upgrade last week (see previous post). The failed system is  Device model : ROCK 4 (aarch64) . , and I  tried again. Still fails today. Doing retries on that system doesn't make it succeed.  After a few retries I uploaded the bug report.  

[image]

Joulinar, 2025-03-08:
This issue is completely unrelated to the original one. It basically a duplicate to https://github.com/MichaIng/DietPi/issues/7445

As you can see on the error messages,  a key depreciated and would need to be renewed 

```
eval curl -sSfL 'https://download.opensuse.org/repositories/home:/jcorporation/Debian_11/Release.key' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-mympd.gpg --yes
apt update
```

jalsco, 2025-03-09:
I'm sorry - yes, that was my mistake putting it on this one.
I missed the message about "key deprecated" until you pointed it out. 

YES. after running:

```eval curl -sSfL 'https://download.opensuse.org/repositories/home:/jcorporation/Debian_11/Release.key' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-mympd.gpg --yes
apt update
```

the dietpi-update from 9.10.0 to 9.11.2 ran successfully. 

Thanks for replying 🙏🏻

closed 2025-05-01 by MichaIng