MichaIng/DietPi #7435: PIVPN does not work anymore

Andaloup666, 2025-02-26:
I made a complete uninstall and install.
The debug-output (pivpn -d) says

::: Generating Debug Output
::::		PiVPN debug		 ::::
=============================================
::::		Latest commit		 ::::
Branch: master
Commit: cc909fd82e6eaf888598072dad69d667646da5ad
Author: Orazio
Date: Thu Feb 20 08:58:34 2025 +0100
Summary: feat(install): support configuring Pi-hole v6
=============================================
::::	    Installation settings    	 ::::
PLAT=Debian
OSCN=bullseye
USING_UFW=0
pivpnforceipv6route=1
IPv4dev=eth0
install_user=dietpi
install_home=/home/dietpi
VPN=wireguard
pivpnPORT=51820
pivpnDNS1=10.199.163.1
pivpnDNS2=
pivpnHOST=REDACTED
INPUT_CHAIN_EDITED=1
FORWARD_CHAIN_EDITED=1
INPUT_CHAIN_EDITEDv6=
FORWARD_CHAIN_EDITEDv6=
pivpnPROTO=udp
pivpnMTU=1420
pivpnDEV=wg0
pivpnNET=10.199.163.0
subnetClass=24
pivpnenableipv6=0
ALLOWED_IPS="0.0.0.0/0, ::0/0"
INSTALLED_PACKAGES=(wireguard-tools)
=============================================
::::  Server configuration shown below   ::::
[Interface]
PrivateKey = server_priv
Address = 10.199.163.1/24
MTU = 1420
ListenPort = 51820
=============================================
::::  Client configuration shown below   ::::
::: There are no clients yet
=============================================
:::: 	Recursive list of files in	 ::::
::::	/etc/wireguard shown below	 ::::
/etc/wireguard:
configs
keys
wg0.conf

/etc/wireguard/configs:
clients.txt

/etc/wireguard/keys:
server_priv
server_pub
=============================================
::::		Self check		 ::::
:: [OK] IP forwarding is enabled
:: [OK] Iptables MASQUERADE rule set
:: [OK] Iptables INPUT rule set
:: [OK] Iptables FORWARD rule set
:: [ERR] WireGuard is not running, try to start now? [Y/n] 
**Job for wg-quick@wg0.service failed because the control process exited with error code.**
See "systemctl status wg-quick@wg0.service" and "journalctl -xe" for details.
Done
:: [OK] WireGuard is enabled 
(it will automatically start on reboot)
:: [OK] WireGuard is listening on port 51820/udp
[INFO] Run pivpn -d again to see if we detect issues

systemctl status says:
● wg-quick@wg0.service - WireGuard via wg-quick(8) for wg0
     Loaded: loaded (/lib/systemd/system/wg-quick@.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2025-02-26 19:00:17 CET; 7min ago
       Docs: man:wg-quick(8)
             man:wg(8)
             https://www.wireguard.com/
             https://www.wireguard.com/quickstart/
             https://git.zx2c4.com/wireguard-tools/about/src/man/wg-quick.8
             https://git.zx2c4.com/wireguard-tools/about/src/man/wg.8
    Process: 5131 ExecStart=/usr/bin/wg-quick up wg0 (code=exited, status=1/FAILURE)
   Main PID: 5131 (code=exited, status=1/FAILURE)
        CPU: 44ms

Feb 26 19:00:17 DietPi systemd[1]: Starting WireGuard via wg-quick(8) for wg0...
Feb 26 19:00:17 DietPi wg-quick[5131]: wg-quick: `wg0' already exists
Feb 26 19:00:17 DietPi systemd[1]: wg-quick@wg0.service: Main process exited, code=exited, status=1/FAILURE
Feb 26 19:00:17 DietPi systemd[1]: wg-quick@wg0.service: Failed with result 'exit-code'.
Feb 26 19:00:17 DietPi systemd[1]: Failed to start WireGuard via wg-quick(8) for wg0.

What should I do?

Joulinar, 2025-02-26:
There seems to be an additional VPN running. Did you tried to reboot?

> wg-quick: `wg0' already exists

Or did you setup something in addition?

MichaIng, 2025-02-26:
So PiVPN is controlling `wg-quick@wg0`, but the `wg0` interface already exists 🤔. Can you try this:
```sh
systemctl stop wg-quick@wg0
ip l l wg0
```

Andaloup666, 2025-02-26:
this is the result:

3: wg0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/none

MichaIng, 2025-02-26:
So the VPN interface is/has been brought up by something else 🤔. It disappears when you bring it down manually, right?
```sh
wg-quick down wg0
ip l l wg0
```

Andaloup666, 2025-02-27:
Yes, it does.

wg-quick down wg0
[#] ip link delete dev wg0

 ip l l wg0
Device "wg0" does not exist.

Andaloup666, 2025-02-27:
pivpn ist working now. thanks a lot.

Andaloup666, 2025-02-27:
yes. i rebooted 3 times, kept running

Joulinar, 2025-02-27:
ok I'm going to close the topic for now, feel free to reopen if needed.

closed 2025-02-27 by Joulinar