MichaIng/DietPi #7439: DietPi-Backup resets the back-up location to default settings.

Krouwndouwn, 2025-03-01:
### Creating a bug report/issue
- [ x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bookworm 0

- Kernel version | `uname -a`

Linux DietPi 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 5 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)

Raspberry Pi 27W USB-C Power Supply

- SD card used | (EG: SanDisk ultra)

Pffttt... SD Cards... :P
Pironman 5 NVMe M.2 SSD PCIe Mini PC Case + Samsung 980 Pro (with heatsink) 1TB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)

DietPi-Backup

- Was the software title installed freshly or updated/migrated?

System software.

- Can this issue be replicated on a fresh installation of DietPi?

Don't know and unable to test this on the only Raspberry Pi 5B I have (running with software required for infrastructure).

- Bug report ID | `echo $G_HW_UUID`

...

#### Steps to reproduce

1. Setup NFS network drive (DietPi-Drive_Manager, my case a "Synology NAS > USB device" with auto standby settings for hdd's)
2. Change back-up location to the NFS network drive (DietPi-Backup).
3. Possibly do a back-up (optional?, also seems to be working).
4. Close ssh connection (optional?).
5. Possibly wait some time (optional?).
6. Open ssh connection (optional?).
7. Open the DietPi-Backup software again and the location is changed to the default settings (Added location (Step 1) can still be selected within the Location > List settings).

Result after step 7:
[image]

#### Expected behaviour

- Changed DietPi-Backup location settings will still be applied when leaving the settings or after reconnecting to the device.

#### Actual behaviour

- DietPi-Backup location settings resets to default.

#### Extra details

- Please ask if needed.

MichaIng, 2025-03-02:
Many thanks for your report.

After changing the backup location and exiting `dietpi-backup` can you show the output of this command:
```sh
cat /boot/dietpi/.dietpi-backup_settings
```

Krouwndouwn, 2025-03-02:
@MichaIng no problem at all. Happy to help!!

The output after editing location and closing dietpi-backup is:

```
FP_TARGET=/mnt/BACK-UPS/Raspberry Pi 5B/dietpi-backup
DAILY_BACKUP=0
AMOUNT=2
SPACE_CHECK=1
```
(This is the correct and changed location.)

I tested it some more and found out that the location resets after opening dietpi-backup again:

```
FP_TARGET=/mnt/dietpi-backup
DAILY_BACKUP=0
AMOUNT=2
SPACE_CHECK=1
```

Extra info: I always open `dietpi-launcher` and from there I open `DietPi Backup`.

Krouwndouwn, 2025-03-04:
@MichaIng I was testing it again today and found out that spaces are not allowed so I used `_` and the settings seem te be saved correctly now. (not resetting after opening DietPi Backup again.)

But if I am right, I was able to create a back-up before with the spaces before leaving and reopening DietPi Backup so possibly still an issue arround supporting spaces. If not it will be great if it's documented ;)

Small question if u have the time: Is there any way to do weekly or monthly back-ups? Daily is to overkill for my setup and I would prefer a monthly back-up.

I will keep u up-to-date if I find some other issues with the `_` that seem to fix the issues at the moment.

MichaIng, 2025-04-19:
Ohh, a simple fix is possible quoting the given path 😅.

Weekly/monthly backups are requested here: #6869