MichaIng/DietPi #7442: Install on "Other" hardware issue.

chrisjwhite30, 2025-03-05:
Have a Seagate Personal Cloud 1 bay NAS, did get Debian installed on it.
Here are the specs followed by the install log below.

_lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

uname -r
6.1.0-31-armmp_

From Dietpi failed install log

#### Details:
- Date           | 2025-03-04 21:27:53
- Program name   | DietPi-Installer
- Command        | `apt-get -y --allow-change-held-packages install initramfs-tools firmware-linux-free linux-image-6.1.0-18-armmp linux-image-6.1.0-20-armmp linux-image-6.1.0-27-armmp linux-image-6.1.0-31-armmp linux-image-armmp u-boot-tools`
- Exit code      | 100
- DietPi version | v9.11.2 (MichaIng/master)
- Distro version | bookworm (ID=7,RASPBIAN=0)
- Kernel version | `Linux seagate-nas 6.1.0-31-armmp #1 SMP Debian 6.1.128-1 (2025-02-07) armv7l GNU/Linux`
- Architecture   | `armhf`
- Hardware model |  (ID=22)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
**Package linux-image-6.1.0-18-armmp is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package linux-image-6.1.0-20-armmp is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'linux-image-6.1.0-18-armmp' has no installation candidate
E: Package 'linux-image-6.1.0-20-armmp' has no installation candidate**
```

MichaIng, 2025-03-13:
Many thanks for reporting. Hmm, this error is unexpected since the installer checks for installed kernel packages, which then do not need an installation candidate when using `apt install`, because the installed package is one. Can you show the output of this please?
```sh
dpkg-query -Wf '${Package}\n' | grep -E '^linux-(image|dtb)|u-boot|bootloader|firmware'
dpkg-query -S /{lib/modules,boot}/* 2> /dev/null
```

chrisjwhite30, 2025-03-14:
Here is the requested info
_chris@seagate-nas:~$ dpkg-query -Wf '${Package}\n' | grep -E '^linux-(image|dtb)|u-boot|bootloader|firmware'_
firmware-linux-free
linux-image-6.1.0-18-armmp
linux-image-6.1.0-20-armmp
linux-image-6.1.0-27-armmp
linux-image-6.1.0-31-armmp
linux-image-armmp
u-boot-tools
_chris@seagate-nas:~$ dpkg-query -S /{lib/modules,boot}/* 2> /dev/null_
linux-image-6.1.0-18-armmp: /lib/modules/6.1.0-18-armmp
linux-image-6.1.0-20-armmp: /lib/modules/6.1.0-20-armmp
linux-image-6.1.0-27-armmp: /lib/modules/6.1.0-27-armmp
linux-image-6.1.0-31-armmp: /lib/modules/6.1.0-31-armmp
linux-image-6.1.0-31-armmp: /boot/config-6.1.0-31-armmp
linux-image-6.1.0-31-armmp: /boot/System.map-6.1.0-31-armmp
linux-image-6.1.0-31-armmp: /boot/vmlinuz-6.1.0-31-armmp
_chris@seagate-nas:~$_

MichaIng, 2025-03-20:
It seems they are only partially installed, or only config files left. We might need to adjust the installer in this case indeed, to only consider fully installed packages. Can you show this:
```sh
dpkg -l | grep linux-image-
```

chrisjwhite30, 2025-03-20:
Before purge
chris@seagate-nas:~$ dpkg -l | grep linux-image-
rc  linux-image-6.1.0-18-armmp     6.1.76-1                            armhf        Linux 6.1 for ARMv7 multiplatform compatible SoCs
rc  linux-image-6.1.0-20-armmp     6.1.85-1                            armhf        Linux 6.1 for ARMv7 multiplatform compatible SoCs
rc  linux-image-6.1.0-27-armmp     6.1.115-1                           armhf        Linux 6.1 for ARMv7 multiplatform compatible SoCs
ii  linux-image-6.1.0-31-armmp     6.1.128-1                           armhf        Linux 6.1 for ARMv7 multiplatform compatible SoCs
rc  linux-image-armmp              6.1.128-1                           armhf        Linux for ARMv7 multiplatform compatible SoCs (meta-package)
chris@seagate-nas:~$
After purge
chris@seagate-nas:~$ dpkg -l | grep linux-image-
ii  linux-image-6.1.0-31-armmp     6.1.128-1                           armhf        Linux 6.1 for ARMv7 multiplatform compatible SoCs
rc  linux-image-armmp              6.1.128-1                           armhf        Linux for ARMv7 multiplatform compatible SoCs (meta-package)
chris@seagate-nas:~$

MichaIng, 2025-03-21:
Thanks, so indeed the old kernel packages are in the list but with remaining config files. I think we should purge all such leftovers within the installer, to have a clean package list.

MichaIng, 2025-05-01:
Solved with: https://github.com/MichaIng/DietPi/commit/8793589

closed 2025-05-25 by MichaIng