MichaIng/DietPi #7443: I have lost internet access if turn on wifi and reboot

TooManyLard, 2025-03-05:
### Creating a bug report/issue
- [ v] I have searched the existing open and closed issues

Notice: I got some infomation and send bug report from image backup from 6 hours ago because i couldn't access terminal with headless.
#### Required Information
- DietPi version | `DietPi_RPi234-ARMv8-Bookworm`
- Distro version | `bookworm 0`
- Kernel version | `Linux DietPi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux`
- SBC model | Raspberry Pi Zero 2 W
- Power supply used | Mobile Battery with 5v2.4A output
- SD card used | i cant remember product info but it has 32GB 

- Bug report ID | ``bd164dbd-1966-4d50-b546-4b7a072aa083

#### Steps to reproduce

1. Set up Dietpi with usb lan adaper
2. Turn on wifi with dietpi-config 
3. Just Exit and reboot (I was required to setting wifi)
4. After rebooting, i have lost access with ssh

#### Expected behaviour

1. The USB LAN adapter is link up even wifi is enabled.
2. i  can access SBC with ssh and configure wifi with dietpi-config

#### Actual behaviour

1. After rebooting, i have lost access SBC with ssh
2. That is happening even I replug USB LAN.

#### Extra details

1. It seems not bug. but i think it isn't work as intented.
2. If typed "ip link set usb0 up" on console by anyway (even Dietpi-Autostart script), it couldn't happening. i have done it before.

TooManyLard, 2025-03-05:
Additional Infomation
It's not happening with this step
1 Enable "OnBoard Wifi" (Third line of Network Options : Adapters)
2 Reboot
3 Enable and Setting WiFi (Second Line of Network Options : Adapters)
but doing third step first, I encount this problem again.

MichaIng, 2025-03-06:
There is indeed a limitation with `dietpi-config` when enabling WiFi and Ethernet concurrently. It tries to attack a default route for both, which cannot work.

If you enable WiFi, then disable Ethernet, does that work?

Regarding your second comment, so you mean it works with onboard WiFi but not with a USB WiFi adapter?

MichaIng, 2025-06-02:
Closing this issue. Feel free to reopen if still relevant.

closed 2025-06-02 by MichaIng