MichaIng/DietPi #7445: MyMPD's APT repository has an expired key

cbrenberg, 2025-03-07:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
All dietpi, any kernel, any hardware.

#### Additional Information (if applicable)
- Software title | dietpi-update
- Bug report ID | 2477b93b-e40c-4610-98da-0005844afec8

#### Steps to reproduce

1. Use any device running dietpi with myMPD installed
2. Run dietpi-update or apt update

#### Expected behaviour

- apt should receive a successful response from the jcorporation repository that hosts the myMPD package

#### Actual behaviour

- apt is returning an error (below), which is preventing dietpi-update from completing.
```
Err:8 https://download.opensuse.org/repositories/home:/jcorporation/Debian_12  InRelease
  The following signatures were invalid: EXPKEYSIG AA09B8CCE895BD7D home:jcorporation OBS Project <home:jcorporation@build.opensuse.org>
Fetched 1532 B in 5s (330 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://download.opensuse.org/repositories/home:/jcorporation/Debian_12  InRelease: The following signatures were invalid: EXPKEYSIG AA09B8CCE895BD7D home:jcorporation OBS Project <home:jcorporation@build.opensuse.org>
E: Failed to fetch https://download.opensuse.org/repositories/home:/jcorporation/Debian_12/InRelease  The following signatures were invalid: EXPKEYSIG AA09B8CCE895BD7D home:jcorporation OBS Project <home:jcorporation@build.opensuse.org>
```

#### Extra details
- The maintainer of myMPD is aware and has already rotated the expired signing key (https://github.com/jcorporation/myMPD/issues/1411)
- Dietpi probably needs to fetch the new key and somehow propagate it out to those of us with myMPD installed. Maybe you already have mechanisms for doing this, but maybe some version of these instructions from the myMPD docs could be integrated into the dietpi-update scripts: https://jcorporation.github.io/myMPD/010-installation/prebuild-packages-debian/

Joulinar, 2025-03-07:
We already know

https://dietpi.com/forum/t/squeezelite-issues-on-rpi-3b/22750/4?u=joulinar

MichaIng, 2025-03-12:
I added a pre-patch to update the key with next DietPi release. It also switches to `.asc` key format, so no `gpg` required anymore.

closed 2025-03-12 by MichaIng