MichaIng/DietPi #7448: NFS4 mounts are not automatically mounted at system start

maartenlangeveld, 2025-03-08:
NFS4 mounts are not auto mounted after reboot

Using dietpi-drive_manager mount an NFS (nfs4) network drive.

result:
```
2025-03-08 12:39:43 root@M-DNS:~# cat /etc/fstab
# You can use "dietpi-drive_manager" to setup mounts.
# NB: It overwrites and re-creates physical drive mount entries on use.
#----------------------------------------------------------------
# NETWORK
#----------------------------------------------------------------
192.x.x.x:/mnt/storage/music /mnt/dietpi_userdata/Music nfs noauto,x-systemd.automount
```
Drive-manager adds these parameters: noauto,x-systemd.automount
Does the 'noauto' parameter prohibit auto mount at start?

Would be nice if the NSF mounts are automatically mounted when system starts.

Joulinar, 2025-03-08:
> x-systemd.automount

did you tried to access the share? usually, it should auto mount as soon as you try to access folder `/mnt/dietpi_userdata/Music`
Mounts are not performed directly during booting

maartenlangeveld, 2025-03-08:
I did, but that does not work

MichaIng, 2025-03-12:
> Is there a reason dietpi does not mount these drives at start?

This assures that mounting drives does not fail and hold boot in case it is no attached, the device/controller takes longer to initialize (USB drives) and in this case especially if network connection is not up yet. For network drive, there is a `netdev` mount option, to wait for systemd network targets, but those are not reliable, e.g. WPA authentication and DHCP client are not assured to delay network targets until network is really there. The systemd automount feature is hence somewhat default on Debian.

Does it cause a particular issue in your case?

maartenlangeveld, 2025-04-03:
No really, thanks for your explanation. I'll close this issue,