MichaIng/DietPi #7449: unable to update Home Assistant due to PyRIC

frentemorao, 2025-03-08:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SD card used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2025-03-12:
When you call `/home/homeassistant/homeassistant-update.sh`, can you show the whole output? The relevant lines of the error output are missing. The deprecation warning is not an error, would not cancel the build, so the error should be shown below that.

frentemorao, 2025-03-14:
Hi @MichaIng 

How can I have the whole output?

Is there any file where I can find the update log?

Or do I need to call /home/homeassistant/homeassistant-update.sh **again** to capture the error?

Joulinar, 2025-03-14:
You should be able to scroll up on your SSH terminal window. There you see the whole output

frentemorao, 2025-03-16:
The other day, the first thing I did was to update DietPi (from v9.2.1 to v9.10 or 9.11), and after that, I tried to reinstall Home Assistant (dietpi-software reinstall 157).
In the proccess, Python 3.13 was installed.
All these steps resulted in the error I previously mentioned.

Today, I called /home/homeassistant/homeassistant-update.sh **without** updating DietPi first.
And I have been able to update Home Assistant with Python 3.12 !!

**However** my Home Assistant Core version now is 2025.1.4 (instead of 2025.3 that I think it is the latest one).
And moreover, my Home Assistant Front End version is still 2024.9 ...
[image]

Also, I would like to point out that I received a notification regarding the incompatibility of Python 3.12 with Home Assistant 2025.2
[image]

Given all this information:
- should I update DietPi first and then run /home/homeassistant/homeassistant-update.sh?
- or should I reinstall Home Assistant without updating?

Joulinar, 2025-03-16:
You would need to update DietPi first as we set a specific python version during update or install. The python version is hard coded within the `dietpi-sofrware` script. Therefore an update is required first.

frentemorao, 2025-03-16:
@Joulinar a Home Assistant update better than a reinstall??

Thanks in advance

Joulinar, 2025-03-16:
A reinstall is needed to update python version

frentemorao, 2025-03-16:
That's exactly what I did the other day when I opened the issue... update DietPi first and then reinstall Home Assistant... 

I will try to do it again in order to get the full log

frentemorao, 2025-03-23:
The first thing I did is to update Dietpi to its latest version v.9.11.2.

Afterward, I have tried to reinstall Home Assistant **without success** (again).

These are the errors I have gotten:
>   ERROR: Failed building wheel for **dbus-fast**
>   Building wheel for **PyRIC** (setup.py): started
>   Building wheel for PyRIC (setup.py): finished with status 'error'

This is the full log:
`Using username "root".
root@192.168.1.111's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v9.11.2 : 13:41 - Sat 03/22/25
 ─────────────────────────────────────────────────────
 - Device model : RPi B (armv6l)
 - CPU temp : 42 °C / 107 °F : Optimal temperature
 - LAN IP : 192.168.1.111 (eth0)
curl: (28) Connection timed out after 4838 milliseconds
 ─────────────────────────────────────────────────────

 DietPi Team     : https://github.com/MichaIng/DietPi#the-dietpi-project-team
 Patreon Legends : Chris Gelatt, ADSB.im
 Website         : https://dietpi.com/ | https://x.com/DietPi_ | Bsky: @dietpi.c                                                                                                                     om
 Contribute      : https://dietpi.com/contribute.html
 Web Hosting by  : https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place
 dietpi-config   : Feature rich configuration tool for your device
 dietpi-software : Select optimised software for installation
 htop            : Resource monitor
 cpu             : Shows CPU information and stats

root@DietPi:~# htop
root@DietPi:~# dietpi-software reinstall 157
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated reinstall

[  OK  ] DietPi-Software | Reinstalling Home Assistant: Open source home automation platform
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for conflicts and missing inputs

 DietPi-Software
─────────────────────────────────────────────────────
 Step: Checking for prerequisite software

[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt /usr/local/bin
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 0775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : home-assistant
[  OK  ] DietPi-Services | stop : proftpd
[ INFO ] DietPi-Software | APT update, please wait...
Get:1 http://raspbian.raspberrypi.com/raspbian bookworm InRelease [15.0 kB]
Hit:2 https://archive.raspberrypi.com/debian bookworm InRelease
[... 936 lines cut ...]
      creating build/lib/pyric
      copying pyric/__init__.py -> build/lib/pyric
      error: [Errno 28] No space left on device
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for PyRIC
  Running setup.py clean for PyRIC
Successfully built aiohttp audioop-lts bcrypt ciso8601 cryptography fnv-hash-fast lru-dict orjson Pillow PyYAML ulid-transform zeroconf cffi fnvhash habluetooth MarkupSafe pycares psutil bluetooth-data-tools
Failed to build dbus-fast PyRIC
ERROR: Failed to build installable wheels for some pyproject.toml based projects (dbus-fast, PyRIC)
[FAILED] DietPi-Software | Installing Python with Home Assistant module into pyenv
 - Command: runuser -u homeassistant -- dash -c
. /home/homeassistant/pyenv-activate.sh
pyenv install 3.13.1
pyenv local 3.13.1
pip3 install -U pip setuptools wheel
[ -z '' ] || pip3 install
exec pip3 install homeassistant
`
And the screenshot:
[image]

Could you please check the log in order to find a solution???
Thanks in advance

Joulinar, 2025-03-23:
> [Errno 28] No space left on device

> chitee: /tmp/G_EXEC_LOG: No space left on device

I gues `/tmp` is running out o disk space. Theoretically, this could be extended, but HA is already a relatively heavy application for a RPiB

frentemorao, 2025-03-23:
@Joulinar I am aware Home Assistant is a heavy application BUT it is working flawlessly at this moment with version 2024.09...

Is there any way of increasing the space of /tmp ??
Or is it possible to re-install HA into a non-temporary directory??

As you can see, I have enough space in the SD:
> root@DietPi:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  5.8G  8.0G  43% /
devtmpfs        207M     0  207M   0% /dev
tmpfs           239M     0  239M   0% /dev/shm
tmpfs            96M  4.7M   91M   5% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           1.0G  896K  1.0G   1% /tmp
tmpfs            50M   16K   50M   1% /var/log
/dev/mmcblk0p1  127M   52M   76M  41% /boot

Joulinar, 2025-03-23:
you can comment out the corresponding line in the `/etc/fstab` file

frentemorao, 2025-03-25:
Moved /tmp from tmpfs to /dev/mmcblk0p2 (in order to avoid the space limitation), and **successfully** reinstalled HA !!!

```
sudo mkdir /mnt/tmpdisk
sudo mount /dev/mmcblk0p2 /mnt/tmpdisk
sudo mount --bind /mnt/tmpdisk /tmp

sudo nano /etc/fstab 
/dev/mmcblk0p2 /mnt/tmpdisk ext4 defaults 0 0
/mnt/tmpdisk /tmp none bind 0 0

sudo reboot
```

Once everything is working, I will move /tmp back to tmpfs (RAM):
```
sudo umount /tmp
sudo umount /mnt/tmpdisk

sudo mount -t tmpfs -o size=1G tmpfs /tmp

sudo nano /etc/fstab
#busca y elimina cualquier línea relacionada con /tmp que apunta al disco 

sudo reboot
```

@Joulinar - thank you so much for your help!!

MichaIng, 2025-04-19:
Hmm, I will check whether we can reduce the `/tmp` disk usage during the install process, or whether it is really only all the wheel builds themselves eating that space alone. Probably we need to work around this in a generic way. Let me reopen this issue.

frentemorao, 2025-04-19:
@MichaIng @Joulinar 

To be honest, I was about to reopen the thread because I haven't been able to reinstall HA **totally** successful.

After resolving the space issue in /tmp, there is a process (av=13.1.0) that attempts to install itself unsuccessfully when Home Assistant starts. 

I have tried to manually force the installation of av=13.1.0, and here are the logs:
<details>
```
root@DietPi:~# /home/homeassistant/.pyenv/versions/3.13.1/bin/pip install av==13.1.0 --constraint /home/homeassistant/.pyenv/versions/3.13.1/lib/python3.13/site-packages/homeassistant/package_constraints.txt --target /mnt/dietpi_userdata/homeassistant/deps --force-reinstall --no-cache-dir
Collecting av==13.1.0
  Downloading av-13.1.0.tar.gz (4.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.0/4.0 MB 2.8 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: av
  Building wheel for av (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for av (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [587 lines of output]
      /tmp/pip-build-env-9k6xgh29/overlay/lib/python3.13/site-packages/setuptools/dist.py:759: SetuptoolsDeprecationWarning: License classifiers are deprecated.
      !!

              ********************************************************************************
              Please consider removing the following classifiers in favor of a SPDX license expression:

              License :: OSI Approved :: BSD License

              See https://packaging.python.org/en/latest/guides/writing-pyproject-toml/#license for details.
              ********************************************************************************

      !!
        self._finalize_license_expression()
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-armv6l-cpython-313/av
      copying av/datasets.py -> build/lib.linux-armv6l-cpython-313/av
      copying av/about.py -> build/lib.linux-armv6l-cpython-313/av
      copying av/__main__.py -> build/lib.linux-armv6l-cpython-313/av
      copying av/__init__.py -> build/lib.linux-armv6l-cpython-313/av
      creating build/lib.linux-armv6l-cpython-313/av/video
      copying av/video/__init__.py -> build/lib.linux-armv6l-cpython-313/av/video
      creating build/lib.linux-armv6l-cpython-313/av/subtitles
      copying av/subtitles/__init__.py -> build/lib.linux-armv6l-cpython-313/av/subtitles
      creating build/lib.linux-armv6l-cpython-313/av/sidedata
      copying av/sidedata/__init__.py -> build/lib.linux-armv6l-cpython-313/av/sidedata
      creating build/lib.linux-armv6l-cpython-313/av/filter
      copying av/filter/__init__.py -> build/lib.linux-armv6l-cpython-313/av/filter
      creating build/lib.linux-armv6l-cpython-313/av/data
      copying av/data/__init__.py -> build/lib.linux-armv6l-cpython-313/av/data
      creating build/lib.linux-armv6l-cpython-313/av/container
      copying av/container/__init__.py -> build/lib.linux-armv6l-cpython-313/av/container
      creating build/lib.linux-armv6l-cpython-313/av/codec
      copying av/codec/__init__.py -> build/lib.linux-armv6l-cpython-313/av/codec
      creating build/lib.linux-armv6l-cpython-313/av/audio
      copying av/audio/__init__.py -> build/lib.linux-armv6l-cpython-313/av/audio
      creating build/lib.linux-armv6l-cpython-313/av/attachments
      copying av/attachments/__init__.py -> build/lib.linux-armv6l-cpython-313/av/attachments
      copying av/stream.pyi -> build/lib.linux-armv6l-cpython-313/av
      copying av/plane.pyi -> build/lib.linux-armv6l-cpython-313/av
      copying av/packet.pyi -> build/lib.linux-armv6l-cpython-313/av
      copying av/option.pyi -> build/lib.linux-armv6l-cpython-313/av
      copying av/logging.pyi -> build/lib.linux-armv6l-cpython-313/av
      copying av/frame.pyi -> build/lib.linux-armv6l-cpython-313/av
      copying av/format.pyi -> build/lib.linux-armv6l-cpython-313/av
      copying av/error.pyi -> build/lib.linux-armv6l-cpython-313/av
[... 527 lines cut ...]
      15806 |   __pyx_t_2 = __Pyx_PyInt_From___pyx_anon_enum(AV_CODEC_FLAG_RECON_FRAME); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 71, __pyx_L1_error)
            |                                                ^~~~~~~~~~~~~~~~~~~~~~~~~
            |                                                AV_CODEC_FLAG_QSCALE
      src/av/codec/context.c:15806:48: note: each undeclared identifier is reported only once for each function it appears in
      src/av/codec/context.c:15827:48: error: ‘AV_CODEC_FLAG_COPY_OPAQUE’ undeclared (first use in this function); did you mean ‘AV_CODEC_FLAG2_NO_OUTPUT’?
      15827 |   __pyx_t_2 = __Pyx_PyInt_From___pyx_anon_enum(AV_CODEC_FLAG_COPY_OPAQUE); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 72, __pyx_L1_error)
            |                                                ^~~~~~~~~~~~~~~~~~~~~~~~~
            |                                                AV_CODEC_FLAG2_NO_OUTPUT
      src/av/codec/context.c:15848:48: error: ‘AV_CODEC_FLAG_FRAME_DURATION’ undeclared (first use in this function); did you mean ‘AV_CODEC_CAP_FRAME_THREADS’?
      15848 |   __pyx_t_2 = __Pyx_PyInt_From___pyx_anon_enum(AV_CODEC_FLAG_FRAME_DURATION); if (unlikely(!__pyx_t_2)) __PYX_ERR(0, 76, __pyx_L1_error)
            |                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for av
Failed to build av
ERROR: Failed to build installable wheels for some pyproject.toml based projects (av)
```
</details>

Could you please help me (again) ??

I am not pretty sure but it could be related to this thread (https://github.com/home-assistant/core/issues/131853) although in this case, av failed to load on armv7l (I'm using armv6)

MichaIng, 2025-04-21:
`error: ‘AV_CODEC_FLAG_RECON_FRAME’ undeclared` I think reason is too old FFmpeg. As long as you do not use any actual audio/video, this should not cause issues. HA core brings this dependency for no good reason.

The issue you linked is a different one, falsely built armv7 `av` wheels, while in your case they are compiled (fail to do so) on your machine.

frentemorao, 2025-04-22:
> `error: ‘AV_CODEC_FLAG_RECON_FRAME’ undeclared` I think reason is too old FFmpeg. As long as you do not use any actual audio/video, this should not cause issues. HA core brings this dependency for no good reason.

The problem is that the av installation consumes almost 100% of the available CPU. 
Since the rpb1 is very limited, this process completely prevents the use of Home Assistant, as it runs out of resources.

Is there anything that can be done to avoid this wheel?
I'd like to be able to use the latest version of Home Assistant

frentemorao, 2025-04-22:
The thing is, this didn't happen with version 2024.9.
So something must have changed from that version until now.

MichaIng, 2025-04-22:
Yes until later versions, HA used an own fork of `av`: [`ha-av`](https://pypi.org/project/ha-av/), which contained some needed new changes, and did not require such new FFmpeg. At some point, the changes reached the `av` upstream, so HA switched back, which caused this incompatibility with older FFmpeg.

frentemorao, 2025-08-31:
@MichaIng - any update ? I would like to install the latest versions of Home Assistant

Thanks in advance ;)

MichaIng, 2026-03-09:
I monitored RAM and `/tmp` usage during a HA install on a `riscv64` system, where almost every Python module with non-Python code parts needs to be compiled. And it finally failed using up the full 2 GiB `/tmp` size. Reason is the following:
- Every downloaded wheel and sdist is stored to a `/tmp/pip-unpack-XXX` dir until the installation of all modules completed.
- A dedicated `/tmp/pip-metadata-XXXX` or `/tmp/pip-modern-metadata-XXXX` dir is created for each module until the installation of all modules completed.
- A build isolation environment dir `/tmp/pip-build-env-XXXX` is created for every module that needs to be compiled, which contains every of its build dependencies (which is in case first compiled) in an extracted form, and all of them are kept until the installation of all modules completed.
- A dedicated `/tmp/tmpXXXX` dir and/or `/tmp/tmpXXXX.whl` file is also created. The dir remained after the `isal` finished, which is installed first in a dedicated call (along with custom dependencies if defined).
- There is a `/tmp/pip-install-XXXX` dir which contains the extracted sources of each package which needs to be compiled, and each subdir is filled with a build dir as well, which contains the compiled module, in case further ephemeral sources etc. Just an example for the `pillow` wheel, at a point where its wheel was compiled already:
  ```console
  root@Star64:/tmp# du -sm pip-install-0fea0mug/pillow_4cb0e96de9eb4e308c1e09aa4f52e14c/build
  19      pip-install-0fea0mug/pillow_4cb0e96de9eb4e308c1e09aa4f52e14c/build
  root@Star64:/tmp# du -sm pip-install-0fea0mug/pillow_4cb0e96de9eb4e308c1e09aa4f52e14c/
  93      pip-install-0fea0mug/pillow_4cb0e96de9eb4e308c1e09aa4f52e14c/
  root@Star64:/tmp# du -sm pip-unpack-s1a47teh/pillow-12.1.1.tar.gz
  45      pip-unpack-s1a47teh/pillow-12.1.1.tar.gz
  ```
  - 19 MiB obsolete build data
  - 93 MiB of obsolete sources + build data
  - 45 MiB of an obsolete tarball
  - 157 MiB of data in `/tmp` wir are not used anymore, and won't be used either after a failed attempt/retry.
  - For only a single compiled wheel. There are 20 overall in this case which need to be compiled.

I am currently testing the effect of `--no-cache` that we use. All above is true as well without `--no-cache`. But compiled wheels, including those which are build dependencies only, are stored in the (disk) cache. Without cache, they are probably stored to `/tmp` as well. However, as wheel they consume only a very minor part of the space, and sadly the cache is not used for downloaded wheels, only for those compiled.

We could try to install the dependencies in chunks. There is no native `pip` feature for this. Only ways I could think of (to assure dependency versions are correct):
1. `pip --dry-run --report install`, then parsing the report, to install each listed dependency individually. However, I don't think they are sorted from back to front in the dependency tree, so one package could pull a large number of dependencies, introducing the very same issue, while other calls would be noops since they were installed as dependency before. `--no-deps` should solve that part.
2. `pip download` everything first, then install the wheels with `--no-deps` first, remove them, then install the tarballs individually.

However, I first want to switch to `uv`, which is planned anyway: https://github.com/MichaIng/DietPi/issues/7644#issuecomment-3142021355
And we just added the standalone `uv` as dedicated software option. That already takes away compiling Python on each HA reinstall, maybe it reduces `/tmp` occupation as well.

Interestingly, overall memory usage for the HA build remains mostly below 2 GiB. It only raises significantly for compiling `grpcio`: one C++ compiler process for each core, consuming ~500 MiB each. They fill up the 4 GiB RAM of my Star64 regularly, but seem to have some internal OOM protection: shortly before the 4 GiB are all used up, CPU usage drops from 400% to ~10%, but it continues, just at a much slower pace. That build eventually goes through, but it stores a lot in `/tmp` as well so that one of the next build fails with no space left in `/tmp`.