MichaIng/DietPi #7451: Dietpi-VPN Killswitch

black00019, 2025-03-08:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applied'

- Distro version | bookworm 0
- Kernel version | Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | RPi 4 Model B (aarch64)
- Power supply used | 5V 1A RAVpower
- SD card used | Western Digital Purple

#### Additional Information (if applicable)
- Software title | dietpi-vpn

#### Steps to reproduce

1. ... Configure ProtonVPN on dietpi-vpn
2. ... Up scirpt: iptables -t nat -A POSTROUTING -s 192.168.10.0/24 -o tun0 -m comment --comment lan-nat-rule -j MASQUERADE
iptables -t nat -A POSTROUTING -s 10.6.0.0/24 -o tun0 -m comment --comment wireguard-nat-rule-2 -j MASQUERADE
ip route add to default via 192.168.10.1 table isp
ip rule add iif lo sport 51820 to default lookup isp prio 15000
3. Down script:
iptables -t nat -D POSTROUTING -s 192.168.10.0/24 -o tun0 -m comment --comment lan-nat-rule -j MASQUERADE
iptables -t nat -D POSTROUTING -s 10.6.0.0/24 -o tun0 -m comment --comment wireguard-nat-rule-2 -j MASQUERADE
ip route del to default via 192.168.10.1 table isp
ip rule del iif lo sport 51820 to default lookup isp prio 15000
4. Enable Autostart and Killswitch

#### Expected behaviour
when i enable killswitch i would like that if the vpn disconnects, both the dietpi and the clients on my network are disconnected from the internet

#### Actual behaviour
When i disconnect vpn, dietpi connect to internet with my isp and my client on network not work, when i reconnect vpn dietpi connect with vpn ip, but my client not work
when i restart dietpi, with killswitch and autostart, my dietpi and client not work on internet

#### Extra details
i have pihole with unbound, pihole is a dhcp server, and the gatway on lan is dietpi
i have installed pivpn with wireguard and the client use pihole and vpn proton

MichaIng, 2025-03-12:
I try to understand the up/down scripts. You you want VPN clients and LAN clients to use the upstream VPN for Internet access, and add the NAT rules via `iptables` for this. To forward packages between interfaces, there are `sysctl` rules needed as well, do you add them somewhere?
```sh
echo -e 'net.ipv4.conf.tun0.forwarding=1\nnet.ipv4.conf.wg0.forwarding=1\nnet.ipv4.conf.eth0.forwarding=1' > "/etc/sysctl.d/nat.conf"
sysctl net.ipv4.conf.tun0.forwarding=1 net.ipv4.conf.wg0.forwarding=1 net.ipv4.conf.eth0.forwarding=1
```
This should enable IP forwarding from and to the upstream VPN (`tun0`), the PiVPN (`wg0`) and the physical network adapter, assuming Ethernet `eth0` here. It skips this for IPv6, as IPv6 is disabled when the VPN comes up, to prevent IPv6 leaks.

Now, you add a new routing table which uses the router (`192.168.10.1`?) as default gateway directly, hence bypassing the VPN, and apply it for the VPN host, when it uses the `51820` as source port. If I understand it correctly, your aim is that DietPi communicates with the VPN clients through the router, instead of through the upstream VPN? I don't think there is a rule/route needed for this, as the `10.6.0.0/24` route should override the upstream VPN route.

I think what is missing is that, due to the killswitch clients cannot establish a connection with the WireGuard server. It allows only inbound SSH connections. If WireGuard clients and hosts from the `192.168.10.0/24` network shall be able to establish connections with the DietPi server, these `iptables` rules should work:
```
-A INPUT -s 10.6.0.0/24 -j ACCEPT
-A INPUT -s 192.168.10.0/24 -j ACCEPT
```
And as the killswitch explicitly drops forwarding as well, for the NAT to work:
```
-A FORWARD -s 10.6.0.0/24 -j ACCEPT
-A FORWARD -s 192.168.10.0/24 -j ACCEPT
```
For the WireGuard clients, `-i wg0` should work as filter as well, but above, narrowing down to the expected IP address range, is a little stricter.

black00019, 2025-03-13:
Hi, i have insert this in my configuration:
`/proc/sys/net/ipv4/ip_forward 1`

and this is my configuration with vpn enable:
`root@DietPi:~# ip -4 addr; ip -4 ro list table all; ip -4 ru; iptables-save -c
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    inet 192.168.10.5/24 brd 192.168.10.255 scope global eth0
       valid_lft forever preferred_lft forever
3: wg0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN group default qlen 1000
    inet 10.6.0.1/24 scope global wg0
       valid_lft forever preferred_lft forever
4: docker0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default 
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
6: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 500
    inet 10.16.0.7/16 scope global tun0
       valid_lft forever preferred_lft forever
default via 192.168.10.1 dev eth0 table isp 
0.0.0.0/1 via 10.16.0.1 dev tun0 
default via 192.168.10.1 dev eth0 onlink 
10.6.0.0/24 dev wg0 proto kernel scope link src 10.6.0.1 
10.16.0.0/16 dev tun0 proto kernel scope link src 10.16.0.7 
128.0.0.0/1 via 10.16.0.1 dev tun0 
149.102.237.129 via 192.168.10.1 dev eth0 
172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 
192.168.10.0/24 dev eth0 proto kernel scope link src 192.168.10.5 
local 10.6.0.1 dev wg0 table local proto kernel scope host src 10.6.0.1 
broadcast 10.6.0.255 dev wg0 table local proto kernel scope link src 10.6.0.1 
local 10.16.0.7 dev tun0 table local proto kernel scope host src 10.16.0.7 
broadcast 10.16.255.255 dev tun0 table local proto kernel scope link src 10.16.0.7 
local 127.0.0.0/8 dev lo table local proto kernel scope host src 127.0.0.1 
local 127.0.0.1 dev lo table local proto kernel scope host src 127.0.0.1 
broadcast 127.255.255.255 dev lo table local proto kernel scope link src 127.0.0.1 
local 172.17.0.1 dev docker0 table local proto kernel scope host src 172.17.0.1 
broadcast 172.17.255.255 dev docker0 table local proto kernel scope link src 172.17.0.1 
local 192.168.10.5 dev eth0 table local proto kernel scope host src 192.168.10.5 
broadcast 192.168.10.255 dev eth0 table local proto kernel scope link src 192.168.10.5 
0:	from all lookup local
15000:	from all iif lo sport 51820 lookup isp
32766:	from all lookup main
32767:	from all lookup default
# Generated by iptables-save v1.8.9 (nf_tables) on Thu Mar 13 22:01:18 2025
*raw
:PREROUTING ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
[0:0] -A PREROUTING -d 172.17.0.2/32 ! -i docker0 -p tcp -m tcp --dport 2525 -j DROP
COMMIT
# Completed on Thu Mar 13 22:01:18 2025
# Generated by iptables-save v1.8.9 (nf_tables) on Thu Mar 13 22:01:18 2025
*filter
:INPUT ACCEPT [1714987:1928948976]
:FORWARD ACCEPT [1771212:1690878684]
:OUTPUT ACCEPT [1223262:680317445]
:DOCKER - [0:0]
:DOCKER-BRIDGE - [0:0]
:DOCKER-CT - [0:0]
:DOCKER-FORWARD - [0:0]
:DOCKER-ISOLATION-STAGE-1 - [0:0]
:DOCKER-ISOLATION-STAGE-2 - [0:0]
:DOCKER-USER - [0:0]
[1771212:1690878684] -A FORWARD -j DOCKER-USER
[1771212:1690878684] -A FORWARD -j DOCKER-FORWARD
[0:0] -A DOCKER -d 172.17.0.2/32 ! -i docker0 -o docker0 -p tcp -m tcp --dport 2525 -j ACCEPT
[0:0] -A DOCKER ! -i docker0 -o docker0 -j DROP
[0:0] -A DOCKER-BRIDGE -o docker0 -j DOCKER
[0:0] -A DOCKER-CT -o docker0 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
[1771212:1690878684] -A DOCKER-FORWARD -j DOCKER-CT
[1771212:1690878684] -A DOCKER-FORWARD -j DOCKER-ISOLATION-STAGE-1
[1771212:1690878684] -A DOCKER-FORWARD -j DOCKER-BRIDGE
[0:0] -A DOCKER-FORWARD -i docker0 -j ACCEPT
[0:0] -A DOCKER-ISOLATION-STAGE-1 -i docker0 ! -o docker0 -j DOCKER-ISOLATION-STAGE-2
[0:0] -A DOCKER-ISOLATION-STAGE-2 -o docker0 -j DROP
[1771212:1690878684] -A DOCKER-USER -j RETURN
COMMIT
# Completed on Thu Mar 13 22:01:18 2025
# Generated by iptables-save v1.8.9 (nf_tables) on Thu Mar 13 22:01:18 2025
*nat
:PREROUTING ACCEPT [18499:2211293]
:INPUT ACCEPT [16472:1228277]
:OUTPUT ACCEPT [78011:6051730]
:POSTROUTING ACCEPT [78011:6051730]
:DOCKER - [0:0]
[15852:1132883] -A PREROUTING -m addrtype --dst-type LOCAL -j DOCKER
[0:0] -A OUTPUT ! -d 127.0.0.0/8 -m addrtype --dst-type LOCAL -j DOCKER
[0:0] -A POSTROUTING -s 172.17.0.0/16 ! -o docker0 -j MASQUERADE
[18:9706] -A POSTROUTING -s 10.6.0.0/24 -o eth0 -m comment --comment wireguard-nat-rule -j MASQUERADE
[427:154912] -A POSTROUTING -s 192.168.10.0/24 -o tun0 -m comment --comment lan-nat-rule -j MASQUERADE
[1582:818398] -A POSTROUTING -s 10.6.0.0/24 -o tun0 -m comment --comment wireguard-nat-rule-2 -j MASQUERADE
[0:0] -A DOCKER -i docker0 -j RETURN
[0:0] -A DOCKER ! -i docker0 -p tcp -m tcp --dport 25 -j DNAT --to-destination 172.17.0.2:2525
COMMIT
# Completed on Thu Mar 13 22:01:18 2025`

I would like the dietpi to use protonvpn for all its traffic and to provide the service to clients in the network (as a gateway), it must also be able to do pihole/unbound services (dns),
and when I connect with wg0 (pivpn) it must be as if I were in a local network (so vpn proton, pihole/unbound)

I followed this post that I had opened and it seems to work at the moment except for the killswitch (which at this point is important)
[](https://dietpi.com/forum/t/dietpi-vpn-gateway-with-pivpn-server-and-pihole/16620/8)

Sorry I'm not an expert on iptables but I wanted to optimize this configuration (also because I think it is of common interest).
Thanks

MichaIng, 2025-04-19:
Can you try to apply only the 4 iptables rules I mentioned with custom up script in `dietpi-vpn` and skip the custom routing tables/iptables rules you applied? To my understanding ot should be sufficient to allow connecting to WireGuard, with the forwarding through the ProtonVPN for incoming traffic being done/enforced automatically by OpenVPN routes and killswitch.