MichaIng/DietPi #7457: orange pi 5 max - unable to see CPU freq / set cpu governor

mrtnjcl, 2025-03-12:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | bookworm
 
- Kernel version | Linux DietPi 6.1.84-vendor-rk35xx #1 SMP Sun Jan  5 09:09:31 UTC 2025 aarch64 GNU/Linux
- SBC model | Orange Pi 5 Max (aarch64)
- Power supply used | 5V 20W
- SD card used | SanDisk ultra

#### Steps to reproduce
(clean install)
dietpi-config -> Performance -> CPU Governor shows [N/A]

#### Expected behaviour
I should be able to set governor / see cpu freq

#### Actual behaviour
dietpi-config -> Performance -> CPU Governor shows [N/A]

dmesg output:
Mar 12 14:37:56 DietPi systemd[1]: Starting dietpi-preboot.service - DietPi-PreBoot...
Mar 12 14:37:56 DietPi DietPi-PreBoot[609]: DietPi-CPU_set | CPU governors are not available on this system. This is probably a virtual machine. Aborting...
Mar 12 14:37:56 DietPi systemd[1]: Finished dietpi-preboot.service - DietPi-PreBoot.

dietpi-preboot is running, no extra software installed

MichaIng, 2025-03-12:
Thanks for reporting. I'll try to replicate.

MichaIng, 2025-03-13:
Hmm it works well here, that is strange:
```console
root@DietPi:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     aarch64
 Temperature  |     32 °C / 89 °F : Cool runnings
 Governor     |     schedutil

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1200 MHz      1008 MHz    1800 MHz
 CPU1         |      1200 MHz      1008 MHz    1800 MHz
 CPU2         |      1200 MHz      1008 MHz    1800 MHz
 CPU3         |      1200 MHz      1008 MHz    1800 MHz
 CPU4         |      1200 MHz      408 MHz    2256 MHz
 CPU5         |      1200 MHz      408 MHz    2256 MHz
 CPU6         |      408 MHz      408 MHz    2304 MHz
 CPU7         |      408 MHz      408 MHz    2304 MHz

[ INFO ] DietPi-CPU_info | The current CPU frequency may be affected by processing this script itself.

root@DietPi:~# systemctl status dietpi-preboot.service
● dietpi-preboot.service - DietPi-PreBoot
     Loaded: loaded (/etc/systemd/system/dietpi-preboot.service; enabled; preset: enabled)
     Active: active (exited) since Mon 2025-02-10 08:12:54 CET; 1 month 1 day ago
   Main PID: 511 (code=exited, status=0/SUCCESS)
        CPU: 190ms

Feb 10 08:12:54 DietPi systemd[1]: Starting dietpi-preboot.service - DietPi-PreBoot...
Feb 10 08:12:54 DietPi DietPi-PreBoot[517]:  DietPi-CPU_set
Feb 10 08:12:54 DietPi DietPi-PreBoot[517]: ─────────────────────────────────────────────────────
Feb 10 08:12:54 DietPi DietPi-PreBoot[517]:  Mode: Applying CPU governor settings: schedutil
Feb 10 08:12:54 DietPi DietPi-PreBoot[517]: [ INFO ] DietPi-CPU_set | Setting CPU frequency limits : Max = Disabled MHz | Min = Disabled MHz
Feb 10 08:12:54 DietPi DietPi-PreBoot[517]: [  OK  ] DietPi-CPU_set | Applied CPU governor settings: schedutil
Feb 10 08:12:54 DietPi systemd[1]: Finished dietpi-preboot.service - DietPi-PreBoot.
root@DietPi:~# uname -a
Linux DietPi 6.1.84-vendor-rk35xx #1 SMP Sun Jan  5 09:09:31 UTC 2025 aarch64 GNU/Linux
```
Does this directory exist in your case?
```sh
ls -l /sys/devices/system/cpu/
```
And can you show loaded modules with `lsmod`, to compare with the ones I have laoded?
```sh
root@DietPi:~# lsmod
Module                  Size  Used by
btsdio                 16384  0
pwm_fan                20480  0
bcmdhd               1318912  0
goodix_ts              28672  0
rk805_pwrkey           16384  0
nvmem_rockchip_otp     20480  0
panfrost               61440  0
drm_shmem_helper       20480  1 panfrost
gpu_sched              32768  1 panfrost
rockchip_cpuinfo       16384  1 nvmem_rockchip_otp
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fuse                  110592  1
dm_mod                106496  0
ip_tables              28672  0
ipv6                  454656  26
r8169                  77824  0
adc_keys               16384  0
pwm_bl                 16384  0
```

mrtnjcl, 2025-03-16:
**root@DietPi:~# cpu**

```
 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     aarch64
 Temperature  |     52 °C / 125 °F : Running warm, but safe
 Governor     |     N/A
```

**systemctl status dietpi-preboot.service**

```
 dietpi-preboot.service - DietPi-PreBoot
     Loaded: loaded (/etc/systemd/system/dietpi-preboot.service; enabled; preset: enabled)
     Active: active (exited) since Wed 2025-03-12 14:37:56 GMT; 3 days ago
   Main PID: 603 (code=exited, status=0/SUCCESS)
        CPU: 52ms
```

**uname -a**
`Linux DietPi 6.1.84-vendor-rk35xx #1 SMP Sun Jan  5 09:09:31 UTC 2025 aarch64 GNU/Linux`

**root@DietPi:~# ls -l /sys/devices/system/cpu/**

```
total 0
drwxr-xr-x 8 root root    0 Mar 12 14:37 cpu0
drwxr-xr-x 8 root root    0 Mar 12 14:37 cpu1
drwxr-xr-x 8 root root    0 Mar 12 14:37 cpu2
drwxr-xr-x 8 root root    0 Mar 12 14:37 cpu3
drwxr-xr-x 8 root root    0 Mar 12 14:37 cpu4
drwxr-xr-x 8 root root    0 Mar 12 14:37 cpu5
drwxr-xr-x 8 root root    0 Mar 12 14:37 cpu6
drwxr-xr-x 8 root root    0 Mar 12 14:37 cpu7
drwxr-xr-x 2 root root    0 Mar 12 14:42 cpufreq
drwxr-xr-x 2 root root    0 Mar 16 10:24 cpuidle
drwxr-xr-x 2 root root    0 Mar 16 10:24 hotplug
-r--r--r-- 1 root root 4096 Mar 16 10:24 isolated
-r--r--r-- 1 root root 4096 Mar 12 14:42 kernel_max
-r--r--r-- 1 root root 4096 Mar 16 10:24 modalias
-r--r--r-- 1 root root 4096 Mar 16 10:24 offline
-r--r--r-- 1 root root 4096 Mar 12 14:42 online
-r--r--r-- 1 root root 4096 Mar 12 14:37 possible
drwxr-xr-x 2 root root    0 Mar 16 10:24 power
-r--r--r-- 1 root root 4096 Mar 12 14:42 present
drwxr-xr-x 2 root root    0 Mar 16 10:24 smt
-rw-r--r-- 1 root root 4096 Mar 12 14:37 uevent
drwxr-xr-x 2 root root    0 Mar 12 14:42 vulnerabilities
```

**lsmod**
```
root@DietPi:~# lsmod
Module                  Size  Used by
btsdio                 16384  0
pwm_fan                20480  0
bcmdhd               1318912  0
goodix_ts              28672  0
rk805_pwrkey           16384  0
nvmem_rockchip_otp     20480  0
panfrost               61440  0
drm_shmem_helper       20480  1 panfrost
gpu_sched              32768  1 panfrost
rockchip_cpuinfo       16384  1 nvmem_rockchip_otp
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
dm_mod                106496  0
fuse                  110592  1
ip_tables              28672  0
ipv6                  454656  26
r8169                  77824  0
adc_keys               16384  0
pwm_bl                 16384  0
```

rockchip_cpuinfo seems to be loaded...

mrtnjcl, 2025-03-16:
dmesg
```
   13.473238] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[   13.473269] Linux version 6.1.84-vendor-rk35xx (build@armbian) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP Sun Jan  5 09:09:31 UTC 2025
[   13.481137] Machine model: RK3588 OPi 5 Max
[   13.481244] efi: UEFI not found.
[   13.484622] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@0': base 0x0000000000000000, size 0 MiB
[   13.484641] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@0': base 0x0000000000000000, size 0 MiB
[   13.484723] Reserved memory: created CMA memory pool at 0x00000004ff800000, size 8 MiB
[   13.484729] OF: reserved mem: initialized node cma, compatible id shared-dma-pool
[   13.773465] Zone ranges:
[   13.773472]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[   13.773484]   DMA32    empty
[   13.773488]   Normal   [mem 0x0000000100000000-0x00000004ffffffff]
[   13.773495] Movable zone start for each node
[   13.773497] Early memory node ranges
[   13.773499]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[   13.773507]   node   0: [mem 0x0000000100000000-0x00000003fbffffff]
[   13.773516]   node   0: [mem 0x00000003fc500000-0x00000003ffefffff]
[   13.773521]   node   0: [mem 0x00000004f0000000-0x00000004ffffffff]
[   13.773527] Initmem setup node 0 [mem 0x0000000000200000-0x00000004ffffffff]
[   13.793841] On node 0, zone DMA: 512 pages in unavailable ranges
[   13.858779] On node 0, zone Normal: 1280 pages in unavailable ranges
[   13.860208] On node 0, zone Normal: 256 pages in unavailable ranges
[   13.860419] psci: probing for conduit method from DT.
[   13.860428] psci: PSCIv1.1 detected in firmware.
[   13.860433] psci: Using standard PSCI v0.2 function IDs
[   13.860439] psci: MIGRATE_INFO_TYPE not supported.
[   13.860489] psci: SMC Calling Convention v1.2
[   13.860877] percpu: Embedded 30 pages/cpu s82728 r8192 d31960 u122880
[   13.860929] pcpu-alloc: s82728 r8192 d31960 u122880 alloc=30*4096
[   13.860942] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7
[   13.861083] Detected VIPT I-cache on CPU0
[   13.861139] CPU features: detected: GIC system register CPU interface
[   13.861143] CPU features: detected: Virtualization Host Extensions
[   13.861159] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[   13.861164] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[   13.861170] alternatives: applying boot alternatives
[   13.861595] Built 1 zonelists, mobility grouping on.  Total pages: 4126752
[   13.861608] Kernel command line: root=UUID=ad01d802-1473-430e-91c9-1cd44e60ebeb rootfstype=ext4 rootwait console=tty1 consoleblank=0 coherent_pool=2M usb-storage.quirks= net.ifnames=0 androidboot.fwver=ddr-v1.16-9fffbe1e78,bl31-v1.45,uboot-rmbian-201-10/14/2024
[   13.863466] Dentry cache hash table entries: 2097152 (order: 12, 16777216 bytes, linear)
[   13.864242] Inode-cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[   13.864251] mem auto-init: stack:off, heap alloc:off, heap free:off
[   13.864258] software IO TLB: area num 8.
[   13.876816] software IO TLB: mapped [mem 0x00000000ec000000-0x00000000f0000000] (64MB)
[   14.008063] Memory: 16316192K/16769024K available (18688K kernel code, 3958K rwdata, 16004K rodata, 8192K init, 703K bss, 444640K reserved, 8192K cma-reserved)
[   14.008233] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
[   14.008275] ftrace: allocating 71507 entries in 280 pages
[   14.136732] ftrace: allocated 280 pages with 3 groups
[   14.136858] trace event string verifier disabled
[   14.137093] rcu: Hierarchical RCU implementation.
[   14.137098] rcu: 	RCU event tracing is enabled.
[   14.137102] 	Rude variant of Tasks RCU enabled.
[   14.137104] 	Tracing variant of Tasks RCU enabled.
[   14.137107] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[   14.155322] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[   14.158668] GICv3: GIC: Using split EOI/Deactivate mode
[   14.158673] GICv3: 480 SPIs implemented
[   14.158677] GICv3: 0 Extended SPIs implemented
[   14.158711] Root IRQ handler: gic_handle_irq
[   14.158726] GICv3: GICv3 features: 16 PPIs
[   14.158770] GICv3: CPU0: found redistributor 0 region 0:0x00000000fe680000
[   14.159177] ITS [mem 0xfe640000-0xfe65ffff]
[   14.159234] ITS@0x00000000fe640000: allocated 8192 Devices @100170000 (indirect, esz 8, psz 64K, shr 0)
[   14.159266] ITS@0x00000000fe640000: allocated 32768 Interrupt Collections @100180000 (flat, esz 2, psz 64K, shr 0)
[   14.159283] ITS: using cache flushing for cmd queue
[   14.159319] ITS [mem 0xfe660000-0xfe67ffff]
[   14.159366] ITS@0x00000000fe660000: allocated 8192 Devices @1001a0000 (indirect, esz 8, psz 64K, shr 0)
[   14.159396] ITS@0x00000000fe660000: allocated 32768 Interrupt Collections @1001b0000 (flat, esz 2, psz 64K, shr 0)
[   14.159411] ITS: using cache flushing for cmd queue
[... 1304 lines cut ...]
[   20.861921] [dhd] dhd_txglom_enable: enable 0
[   20.861931] [dhd] dhd_bus_devreset: making DHD_BUS_DOWN
[   20.861970] [dhd] wifi_platform_set_power = 0, delay: 10 msec
[   20.861983] [dhd] ======== PULL WL_REG_ON(-1) LOW! ========
[   20.861992] [WLAN_RFKILL]: rockchip_wifi_power: 0
[   20.862004] [WLAN_RFKILL]: rockchip_wifi_power: toggle = false
[   20.862013] wifi power off
[... cut ...]
[   21.581579] RTL8226B_RTL8221B 2.5Gbps PHY r8169-3-3100:00: attached PHY driver (mii_bus:phy_addr=r8169-3-3100:00, irq=MAC)
[   21.678257] rk-pcie fe190000.pcie: failed to initialize host
[   21.678305] rk-pcie fe170000.pcie: failed to initialize host
[   21.771758] r8169 0003:31:00.0 eth0: Link is Down
[   24.744893] r8169 0003:31:00.0 eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   24.744947] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   31.412487] platform mtd_vendor_storage: deferred probe pending
[   41.224785] [GTP-ERR][__do_register_ext_module:79] Module [goodix-fwu] timeout
```

MichaIng, 2025-04-19:
Weird, exactly the same kernel. Should not make a difference, but can you try to flash the bootloader image:
```sh
sudo /boot/dietpi/func/dietpi-set_hardware flash-u-boot-mmc
```

MichaIng, 2025-06-02:
Is the issue still present with latest kernel and bootloader (flashed with above command)?

mrtnjcl, 2025-06-06:
Bootloader update didn't help. I ended up returning the board, it seemed to be hardware issue. With new board it worked flawlessly straight out of the box.

 I found one user who reported same problem on their Radxa Rock 5B board, which shares the same chipset - i guess it's manufacturing defect.

closed 2025-06-06 by mrtnjcl

MichaIng, 2025-06-06:
Seems so. Though strange hardware defect when the system boots and works, but only CPUFreq does not.

mrtnjcl, 2025-06-06:
this is the link to radxa forum, https://forum.radxa.com/t/rock5b-dmesg-pvtm/13859

my understanding is that the pvtm part of the cpu was not working properly, and as a result the CPU ran normally, but underclocked (?) B/c my board had the exact same issue with low geekbench score