MichaIng/DietPi #7463: Wi-Fi not working after update

holdit, 2025-03-15:
#### Details:
- Date           | 2025-03-15 16:33:28
- Bug report     | d726a1b7-2b54-4680-a784-d9ac21834b79
- Program name   | DietPi-Set_hardware
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- DietPi version | v9.11.2 (MichaIng/master)
- Distro version | bookworm (ID=7,RASPBIAN=0)
- Kernel version | `Linux hostname 6.6.74+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.74-1+rpt1 (2025-01-27) armv7l GNU/Linux`
- Architecture	 | `armhf`
- Hardware model | RPi 2 Model B (armv7l) (ID=2)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

I updated my Pi and afterwards, WiFi wasn't working anymore.

Using DietPi-Config > Network options: adapters, shows WiFi "on" but "not found".

Disabling/disabling WiFi ends up with this error.

│  - Command: systemctl start hostapd isc-dhcp-server                                                                  │ 
│  - Exit code: 1                                                                                                      │ 
│  - DietPi version: v9.11.2 (MichaIng/master) | HW_MODEL: 2 | HW_ARCH: 2 | DISTRO: 7                                  │ 
│  - Error log:                                                                                                        │ 
│ Failed to start hostapd.service: Unit hostapd.service is masked.                                                     │ 
│ Failed to start isc-dhcp-server.service: Unit isc-dhcp-server.service is masked.                                     │ 

Bug report generated by the tool: d726a1b7-2b54-4680-a784-d9ac21834b79

I'm a bit of a noob, so suggestions/commands would be appreciated.

Joulinar, 2025-03-15:
Did you setup some kind of hotspot? At least the 2 services failing are indicating this.

If not, you might remove them.

holdit, 2025-03-15:
No hotspots. I connected to my WiFi when I flashed DietPi and that was it. It then stopped connecting to the WiFi after updating the Pi.

holdit, 2025-03-15:
Thank you. I removed them both and the error went away (no idea why it was there in the first place as I never used this as a hotspot).

Anyway, even after a reboot, WiFi is still "not found":

[image]

After using dmesg, I noticed this:

[ 1026.697735] mt76x0u 1-1.2:1.0: ASIC revision: 76100002 MAC revision: 76502000
[ 1026.700197] mt76x0u 1-1.2:1.0: mediatek/mt7610e.bin not found, switching to mediatek/mt7610u.bin
[ 1026.700278] mt76x0u 1-1.2:1.0: Direct firmware load for mediatek/mt7610u.bin failed with error -2
[ 1026.702906] mt76x0u: probe of 1-1.2:1.0 failed with error -2

By following the steps on the first post here, it started working again: https://github.com/raspberrypi/firmware/issues/1563

[image]

No idea why this happened, but it's working now.

Thanks for taking some time to help me with this.

closed 2025-03-15 by holdit