MichaIng/DietPi #7465: Plexmediaserver package missing after update to v9.11.2

ZiggyPi, 2025-03-18:
After running dietpi-update on my rpi 4b i noticed plex wasnt loading my movies and series.

Running systemctl etc led me to the forums and running the command below.

root@DietPi-HTPC:~# apt install --reinstall plexmediaserver
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package plexmediaserver

No matter what i tried (apt-get update && apt-get upgrade or apt update && apt upgrade) the package is still missing.

Any tips on how to fix this?

Joulinar, 2025-03-18:
Did you already installed flex from `dietpi-software` tool? Because it has a 3rd party apt repository that would need to be available on your system.

Can you check if it is available?

```
for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
```

ZiggyPi, 2025-03-18:
Yep it has been working for years without any problem. And it is selected in the software list (nr. 42).

`root@DietPi-HTPC:~# for i in /etc/apt/sources.list{,.d/*.list}; do echo "$i:"; cat "$i"; done
/etc/apt/sources.list:
deb https://mirror.netcologne.de/raspbian/raspbian/ bullseye main contrib non-free
/etc/apt/sources.list.d/buster-backports.list:
deb http://deb.debian.org/debian bullseye-backports main
/etc/apt/sources.list.d/deluge.list:
deb http://ppa.launchpad.net/deluge-team/ppa/ubuntu xenial main
/etc/apt/sources.list.d/dietpi.list:
deb https://dietpi.com/apt bullseye main rpi4
deb https://dietpi.com/apt all rpi
/etc/apt/sources.list.d/dietpi-mono.list:
deb https://download.mono-project.com/repo/debian/ raspbianbuster main
/etc/apt/sources.list.d/docker.list:
deb [arch=armhf] https://download.docker.com/linux/raspbian bullseye stable/etc/apt/sources.list.d/php.list:
deb https://packages.sury.org/php/ bullseye main
/etc/apt/sources.list.d/raspi.list:
deb https://archive.raspberrypi.com/debian/ bullseye main
/etc/apt/sources.list.d/sonarr.list:
deb https://apt.sonarr.tv/debian/ buster main
/etc/apt/sources.list.d/webmin.list:
deb https://download.webmin.com/download/newkey/repository stable contrib
/etc/apt/sources.list.d/wsdd.list:
deb https://pkg.ltec.ch/public/ bullseye main`

It all started with : `sub-process /usr/bin/dpkg returned an error code (1)` 
And
`dpkg: error processing package plexmediaserver (--configure):`

Joulinar, 2025-03-18:
There is no Plex source list anymore. You might need to reinstall Plex 

```
dietpi-software reinstall 42
```

Joulinar, 2025-03-19:
> I tried sudo `apt install --reinstall plexmediaserver` earlier but clearly that didnt work.

Yes, that couldn't work because the source list was missing

> Is there any change i can restore my plex setup so i dont have to start all over again?

I don't use Plex myself and can't answer that.

MichaIng, 2025-03-19:
Is there a chance that you uninstalled Plex somewhat before? There is no other way how the data and the APT list could be removed, along with the installed package. Also the DietPi updates do not touch any of this. It might be possible that due to unexpected APT package conflicts, the `plexmediaserver` package gets removed while other packages get upgraded, but that would cause a failure in any of the DietPi scripts, as APT requires interactive confirmation to remove any package for upgrading another one. And then the APT list and key would be still in place, and also `/var/lib/plexmediaserver` as data directory. Moreless impossible that all 3 things + the package get all somewhat purged, without actively uninstalling it via `dietpi-software`.

A backup would be a way to recover the data.

ZiggyPi, 2025-03-19:
@MichaIng no never uninstalled. The only commands i run are dietpi-update and apt upgrade if dietpi tells me to do so.

Maybe one of those commands got interrupted while running?

I tried restoring it from the backup but that didnt work, so i just started with a fresh plex and changed all the settinga i could remember. For now it works, so that's good enough for now.

closed 2025-03-20 by ZiggyPi

MichaIng, 2025-03-20:
Weird, I have no idea either then. Can you check whether `/var/lib/dpkg/info/plexmediaserver.postrm` contains a command to remove `/var/lib/plexmediaserver` on purge?

In our uninstall code I still have a comment that it is left in place on package purge.