MichaIng/DietPi #7468: [Proxmox .iso] The tool to update initramfs was NOT found in function do_run_update_initrd_from

TheLinuxGuy, 2025-03-23:
### Creating a bug report/issue
- [X ] I have searched the existing open and closed issues

#### Required Information
Not applicable, dietpi was not successfully installed.

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce
Followed "Proxmox" .iso install instructions.
1. Downloaded https://dietpi.com/downloads/images/DietPi_VM-x86_64-Bookworm_Installer.iso
2. Booted off the .iso
3. Selected install dietpi on boot menu
4. Selected disk to install dietpi on
5. Software dumped the images onto /dev/vda (selected disk)
6. On the final screen right after "Test if we can chroot the restored OS partition /dev/vda1" it shows an irrecoverable error "Failed to create initrd in the restored". A tool binary is missing from the .iso
7. Installer will reboot the VM and the VM will be unbootable/unusable.

[image]

#### Expected behaviour
Install should succeed without an error, the initrd/boot loader should have been installed properly.

#### Actual behaviour
See screenshot, install crashes on the final step after restoring the .img onto vdisk.

Been able to reproduce this on several different attempts on the same .iso source downloaded fresh from dietpi.com today.

```
% md5sum DietPi_VM-x86_64-Bookworm_Installer.iso
cf31c154a101fa46525e33c7a53591f7  DietPi_VM-x86_64-Bookworm_Installer.iso
```

pulpe, 2025-03-23:
I can not reproduce it. Tried now and got same "error", but dietpi is normally bootable and working.

TheLinuxGuy, 2025-03-25:
> I can not reproduce it. Tried now and got same "error", but dietpi is normally bootable and working.

Here is a better screenshot, it actually hangs at initramfs looking for disk. Similar to this issue [2yrs ago](https://dietpi.com/forum/t/new-bullseye-install-on-esxi-can-t-boot/5897/5) on bullseye 

[image]

pulpe, 2025-03-25:
Strange indeed. For me it's working 

[image]

Created VM with defaults, only increased core count to 2.

pulpe, 2025-03-26:
> specific to proxmox
That you would need to ask @MichaIng , because i honestly dont know.

MichaIng, 2025-04-19:
The block device type/controller should make the difference. Our VM initramfs implies a specific set of controller drivers known to be used by the VM engines, e.g. the virtio-scsi for Proxmox. However, there are different options which may not all work with that initramfs.

For VMs, we use `tiny-initramfs` as minimal implementation, which does not provide `update-initramfs` but `update-tirfs` instead, reducing the size of the initramfs image massively. However, it adds only the needed controller drivers depending on what it detects on generation + those explicitely defined. When we generate the images, it cannot know which driver is needed for the target system.

There are some other drawbacks of `tiny-initramfs`, including the fact that it is not developed anymore since years, only still included in the Debian APT repos. So I think we should switch to `initramfs-tools` which includes all commonly used controller drivers, is hence widely conpatible and should solve the Clonezilla error about missing tool as well. An alternative would be a wrapper script to redirect `update-initramfs` calls, but overall it does not seem to be worth it, including the fact that we'd need to maintain the list of controller drivers needed for the common options of all VM platforms we want to support.

MichaIng, 2025-05-01:
Future DietPi VM images will ship with `initramfs-tools` instead of `tiny-initramfs`: https://github.com/MichaIng/DietPi/commit/f8002eb

MichaIng, 2025-05-22:
@TheLinuxGuy since even with the complete initramfs that includes all Debian kernel SCSI/SATA/etc controller drivers, it fails in your case, we need to clone your Proxmox environment exactly to replicate it. Which Proxmox version do you use?

TheLinuxGuy, 2025-05-22:
Sorry I should have been clearer, I use the proxmox.iso but I install this on my VPS provider https://sparkedhost.com/

I could grant you access for a day or two if you wish to debug, we can correspond privately and I will share some credentials for you if you're interesting in that.

MichaIng, 2025-05-22:
This means the latest ISO v8.4-1 from here? https://www.proxmox.com/en/downloads

MichaIng, 2025-05-22:
Ah, so you don't know exactly which virtualiser/version that hosting provider uses? Does a regular Debian installation (via their ISOs) work? I mean Clonezilla is Debian as well, though Debian Trixie, and the ISO is inserted into a different virtual port, likely with different controller. But from Clonezilla Live shell, it would be possible to find out which drivers are actually used. Send me an email in case, I can send you an SSH pubkey as well, in case of regular Debian with SSH access running.

TheLinuxGuy, 2025-05-25:
> Ah, so you don't know exactly which virtualiser/version that hosting provider uses?

The VPS provider finally shared they use https://virtfusion.com/

@MichaIng If you share with me your SSH public key I can add you to the VM so you can login, I just installed debian 12 bookworm from the hosting provider OS options list.

TheLinuxGuy, 2025-05-29:
Hi, I confirm v9.13 .iso image now works to install on my VPS provider which uses Virtualizor. We could mark this fixed.

Thanks @MichaIng