MichaIng/DietPi #7469: Rockchip NPU support not available for Radxa Rock 3A with EK3568 CPU

easy-and-simple, 2025-03-23:
The DietPi image for the Radxa Rock 3A single-board computer (SBC) based on the RK3568 processor does not include drivers for the Rockchip Neural Processing Unit (NPU). This means that the NPU cannot be used.

To support the use of the NPU, please add the Rockchip NPU driver to the DietPi image for the Radxa Rock 3A SBC.

MichaIng, 2025-04-19:
Reason is that we provide those images with mainline kernel. For the NPU driver, the vendor kernel is required.

We can add it to our repo, but it might additionally require to chamge the bootloader and/or device tree setting. Do you have a spare SD card to test with, just in case something goes wrong with the kernel switch?

Deoptim, 2025-04-20:
Hello.
Here is a link to a kernel with NPU support: https://github.com/MichaIng/DietPi/discussions/5168#discussioncomment-11863032