MichaIng/DietPi #7470: Update of dhcpcd5 is held back

Andaloup666, 2025-03-24:
For some time (weeks) the update of dhcpcd5 has been displayed as necessary, but then not installed.

_sudo apt upgrade
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Paketaktualisierung (Upgrade) wird berechnet… Fertig
Die folgenden Pakete sind zurückgehalten worden:
  dhcpcd5_

Why is this the case? Is this a bug?

MichaIng, 2025-04-19:
Since Bookworm, the package is called `dhcpcd` without "5". So in case you really want to use it:
```sh
sudo apt install dhcpcd
```
But if yiu did not explicitly install it to replace the `isc-dhcp-client` or if it was really only installed by the Pi-hole installer some years ago, just purge it:
```sh
sudo apt purge dhcpcd5
```

MichaIng, 2025-06-02:
Closing this issue. Feel free to reopen if still relevant.