MichaIng/DietPi #7471: qBittorrent's user is not in the dietpi-group -> no r/w permissions

raphipod, 2025-03-25:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.11.2
- Distro version | bookworm
- Kernel version | Linux DietPi 6.1.0-32-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.129-1 (2025-03-06) 
- SBC model | Native PC (x86_64)
- Power supply used | not relevant
- SD card used | not relevant

#### Additional Information (if applicable)
- Software title | qBittorrent 4.5.2 (Web UI)
- Was the software title installed freshly or updated/migrated? Yes, freshly installed.
- Can this issue be replicated on a fresh installation of DietPi? Yes.

- Bug report ID | No bug report.

#### Steps to reproduce

1. Fresh install of DietPi Native x86_64
2. Mount an internal HDD (ext4) 
3. Install qBittorrent with dietpi-software and try to download a torrent to the internal HDD with the web UI.

#### Expected behaviour

- It should start downloading immediately to the internal drive

#### Actual behaviour

- The torrent errors out and doesn't tell you why.

#### Extra details

- After hours of tinkering I found that the "qbittorrent" user is not in the "dietpi"-group and therefore can not access & download the torrent on an internally mounted drive.

MichaIng, 2025-04-19:
Hmm that is weird. We even make `dietpi` the primary group for the `qbittorrent` user. Can you check this:
```sh
systemctl cat qbittorrent
getent passwd qbittorrent
```

raphipod, 2025-04-19:
I am very sorry to report that I am not running DietPi anymore on this exact device, so I can't check/replicate it directly. I could test it in a x64 VM though.

raphipod, 2025-04-19:
Reopened for

raphipod, 2025-04-19:
Unfortunately I don't run DietPi on this device anymore to debug this issue further. I could spin up a quick VM with DietPi and approximately set it up to the device's specs.

raphipod, 2025-04-19:
So, I've now created a VM on my Desktop-PC in VMware Workstation and installed DietPi on it's "internal" 16 GB HDD. Then, I added a second 5 GB HDD with the mountpoint "/mnt/ExternalHDD" as described above in the issue with dietpi-drive_manager.
(Sorry for the big pictures!)

[image]

**The 5GB HDD is the drive the torrent should be downloaded to.**

Next, I installed **qbittorrent** with dietpi-software and then, I opened the VM's qbittorrent WebUI in my browser to confirm it works.
Then, I changed the default download location to the aforementioned "/mnt/ExternalHDD" and applied it.

[image]

Now, I chose a random Ubuntu LTS-Torrent file that works and uploaded it to the WebUI.

[image]

After that, the status of the torrent quickly changed to "Errored" in the WebUI.

[image]

And last, but not least - here are the two commands @MichaIng asked me to run (this is after the erroring-out of the torrent)

[image]

Joulinar, 2025-04-19:
can you share

`ls -la /mnt/ExternalHDD`

raphipod, 2025-04-19:
> can you share
> 
> `ls -la /mnt/ExternalHDD`

There is nothing written into the disk.

```
total 24
drwxr-xr-x 3 root root  4096 Apr 19 18:32 .
drwxr-xr-x 7 root root  4096 Apr 19 18:32 ..
drwx------ 2 root root 16384 Apr 19 18:32 lost+found

```

Joulinar, 2025-04-20:
As you can see, only the root user is allowed to write to they directory xxx. No other user is authorised to do so. It is therefore quite normal that Qbit cannot write to the directory and displays an error. You should at least assign write permissions for the dietpi group.

raphipod, 2025-04-20:
@Joulinar Ah, I've overlooked that. It makes more sense now. I already wrote the "fix" in the [Extra details](https://github.com/MichaIng/DietPi/issues/7471#issue-2946064876) at the start of this issue, though: We might have to add the "qbittorrent"-user to the dietpi-group at install time in the Software installer (if that's possible) to give qbittorrent the r/w permissions to write to the drive.

MichaIng, 2025-04-21:
The `dietpi` is made the primary group for the `qbittorrent` user and umask is adjusted so that it creates files/dirs as `qbittorrent:dietpi` with `664`/`775`. This makes it easier for other software like Servarr stuff and media players to access the downloads.

But when changing the download directory, that one needs to be made accessible to `qbittorrent` user or `dietpi` group once:
```sh
sudo chown 'qbittorrent:dietpi' /mnt/ExternalHDD
sudo chmod -R 'g+w' /mnt/ExternalHDD
```

raphipod, 2025-04-21:
Confirmed working with the last two commands. After restarting the qbittorrent-service with systemctl, the torrent started downloading right away! Thanks for the quick resolution. Closing this issue now.

closed 2025-04-21 by raphipod