MichaIng/DietPi #7475: Backup restore doesn't work

FPSUsername, 2025-03-30:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version` | current / fresh installation
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN` | Bookworm Ordroid N2+
- Kernel version | `uname -a` | current / fresh installation
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3) | Ordroid N2+
- Power supply used | (EG: 5V 1A RAVpower) | Ordroid N2+ official spply
- SD card used | (EG: SanDisk ultra) Samsung class 6

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated? | Fresh installation
- Can this issue be replicated on a fresh installation of DietPi? | Yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Have a backup of your dietpi (most recent, all up to date)
 I have it on an external hdd mounted as /mnt/hdd
2. Install a fresh dietpi installation
3. Run dietpi-backup and restore the backup
#### Expected behaviour

1. It should restore the backup

#### Actual behaviour

1. The files on the SD card of the system are deleted, e.g. `deleting boot/dietpi/.version`
2. It shows issues of not being able to delete non empty directories. E.g. `cannot delete non-empty directory: boot`
3. After deleting sbin, lib aand bin, it shows the following:
```

sent 54 bytes received 18 bytes 144.00 bytes/sec
total size is 0 speedup is 0.00
rsync error: some files/attrs were not transferred (see previous errors) (code 23)  at main.c(1338) [sender=3.2.7]
/boot/dietpi/func/dietpi-globals: line 265: tput: command not found
/boot/dietpi/func/dietpi-globals: line 266: ( 91 + 5 ) / : syntax error: operand expected (error token is "/ ")
/boot/dietpi/func/dietpi-globals: line 148: /usr/bin/rm: no such file or directory
[FAILED] DietPi-Backup | Failed to remove scripts working directory: /tmp/DietPi-Backup```

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
c15f262dd6824e6a7b7eacfd0b2f67ddb23bf26670d260d7cb10b5f943c3441b  DietPi_OdroidN2-ARMv8-Bookworm.img.xz

Joulinar, 2025-03-30:
> After deleting sbin, lib aand bin

You deleted these folders yourself?

FPSUsername, 2025-03-30:
No, the dietpi backup restore program does it

Joulinar, 2025-03-30:
Are you trying to restore a backup that was previously created on an OdroidN2? You can only restore backups that were created on the same hardware or the same system type. Theoretically, you can take a look at the backup to see if these folders are there.

FPSUsername, 2025-03-30:
Yes. Some days ago I made a full backup as my SD card started to give input/output errors (pretty much end of life).

The files are all readable and intact (I checked files it threw the i/o error on).

I got another SD card,flashed the latest dietpi on it and got it all up to date (first setup). Then mount my hdd and start the restore process.

It does warn about different UUID, but that can always be fixed later on, as long as all services, installed programs, etc are restored.

FPSUsername, 2025-04-01:
I made a fresh installation and created a backup. That restored without issues. However, when I restore my previous backup, this happens:

[image]

I'm not sure if that happens because of UUID differences or something else that it tries to wipe the partition and copy it all over or if there's something wrong with my original backup. Note that with my previous system image, something happened and rtorrent flooded its log directory with nearly 1000000 log files. I did delete them from the backup, not sure if that would affect the restore process.

Joulinar, 2025-04-01:
Can you check if these files are existing on the backup you are trying to restore? We use standard `rsync` commands to create and restore backups. `rsync` will not go to delete files if they exist on both sides, the backup and system. Actually it looks like these files are missing inside the backup folder.

FPSUsername, 2025-04-02:
The files do exist in the backup. Oddly enough, the backup had created two folders, "data" (empty) and "data_2".
The backup of a fresh install only populated the "data" folder.

So I renamed the "data_2" folder to "data" and now it seems to be restoring.

So far I see a couple of errors (the backup drive is BTRFS formatted):
```
87,615,056 0% 34.63MB/s 0:00:02 (xfr#214, to-chk=555946/556499)[  375.433814] BTRFS error (device sda1): incorrect extent count for 162165424128; counted 2, expected 3
```

Either way, the dietpi-backup doesn't seem to care if the "data" directory is empty and it will blindly restore. I assume that it's possible to do an "is empty" check or at least get it to check if DietPi OS files are present.