MichaIng/DietPi #7476: Lastest apt upgrade on my RPi-zero-w goes wrong

maartenlangeveld, 2025-04-01:
Lastest apt upgrade on my RPi-zero-w goes wrong:

```
Unpacking linux-image-rpi-v6 (1:6.12.20-1+rpt1~bpo12+1) over (1:6.6.74-1+rpt1) ...
Preparing to unpack .../12-linux-libc-dev_1%3a6.12.20-1+rpt1~bpo12+1_all.deb ...
Unpacking linux-libc-dev (1:6.12.20-1+rpt1~bpo12+1) over (1:6.6.74-1+rpt1) ...
Setting up linux-libc-dev (1:6.12.20-1+rpt1~bpo12+1) ...
Setting up libavutil57:armhf (8:5.1.6-0+deb12u1+rpt3) ...
Setting up libswresample4:armhf (8:5.1.6-0+deb12u1+rpt3) ...
Setting up raspi-firmware (1:1.20250326-1) ...
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi-firmware (--configure):
 installed raspi-firmware package post-installation script subprocess returned error exit status 1
Setting up libpostproc56:armhf (8:5.1.6-0+deb12u1+rpt3) ...
Setting up linux-image-6.12.20+rpt-rpi-v6 (1:6.12.20-1+rpt1~bpo12+1) ...
I: /vmlinuz is now a symlink to boot/vmlinuz-6.12.20+rpt-rpi-v6
I: /initrd.img is now a symlink to boot/initrd.img-6.12.20+rpt-rpi-v6
/etc/kernel/postinst.d/tiny-initramfs:
tiny-initramfs: Generating /boot/initrd.img-6.12.20+rpt-rpi-v6
/etc/kernel/postinst.d/z50-raspi-firmware:
raspi-firmware: missing /boot/firmware, did you forget to mount it?
run-parts: /etc/kernel/postinst.d/z50-raspi-firmware exited with return code 1
dpkg: error processing package linux-image-6.12.20+rpt-rpi-v6 (--configure):
 installed linux-image-6.12.20+rpt-rpi-v6 package post-installation script subprocess returned error exit status 1
Setting up libavcodec59:armhf (8:5.1.6-0+deb12u1+rpt3) ...
Setting up libswscale6:armhf (8:5.1.6-0+deb12u1+rpt3) ...
Setting up libavformat59:armhf (8:5.1.6-0+deb12u1+rpt3) ...
Setting up libavfilter8:armhf (8:5.1.6-0+deb12u1+rpt3) ...
Setting up libavdevice59:armhf (8:5.1.6-0+deb12u1+rpt3) ...
dpkg: dependency problems prevent configuration of linux-image-rpi-v6:
 linux-image-rpi-v6 depends on linux-image-6.12.20+rpt-rpi-v6 (= 1:6.12.20-1+rpt1~bpo12+1); however:
  Package linux-image-6.12.20+rpt-rpi-v6 is not configured yet.

dpkg: error processing package linux-image-rpi-v6 (--configure):
 dependency problems - leaving unconfigured
Setting up ffmpeg (8:5.1.6-0+deb12u1+rpt3) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u9) ...
Processing triggers for tiny-initramfs (0.1-5) ...
Errors were encountered while processing:
 raspi-firmware
 linux-image-6.12.20+rpt-rpi-v6
 linux-image-rpi-v6
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2025-04-01:
can you share following

```
df -h
uname -a
```

maartenlangeveld, 2025-04-01:
> can you share following
> 
> ```
> df -h
> uname -a
> ```

```
2025-04-01 15:05:57 root@M-MUSIC-PLAYER:~# uname -a
Linux M-MUSIC-PLAYER 6.6.74+rpt-rpi-v6 #1 Raspbian 1:6.6.74-1+rpt1 (2025-01-27) armv6l GNU/Linux
```

```
2025-04-01 15:05:55 root@M-MUSIC-PLAYER:~# df
Filesystem     1K-blocks      Used Available Use% Mounted on
/dev/root       14907064  11769916   2506288  83% /
devtmpfs          210628         0    210628   0% /dev
tmpfs             243624         0    243624   0% /dev/shm
tmpfs              97452      3124     94328   4% /run
tmpfs               5120         8      5112   1% /run/lock
tmpfs            1048576       200   1048376   1% /tmp
tmpfs              51200        12     51188   1% /var/log
/dev/mmcblk0p3 362425272 155039900 188901968  46% /mnt/storage
```

Joulinar, 2025-04-01:
@MichaIng do you know if RPi1/Zero W uses the same file system structure `/boot/firmware` for the new RPI kernel as other RPI SBC?

maartenlangeveld, 2025-04-01:
Maybe this helps locate the issue. There is a /boot/firmware directory

```
2025-04-01 17:34:44 root@M-MUSIC-PLAYER:~# ls /boot/firmware
total 51300
drwxr-xr-x 4 root root    4096 Feb 10 14:46  .
drwxr-xr-x 6 root root    4096 Mar 27 23:31  ..
-rwxr-xr-x 1 root root    1594 Mar  7 22:39  LICENCE.broadcom
drwxr-xr-x 2 root root    4096 Sep 24  2023 'System Volume Information'
-rwxr-xr-x 1 root root   30696 Jan 27 18:19  bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 root root   30055 Jan 27 18:19  bcm2708-rpi-b-rev1.dtb
-rwxr-xr-x 1 root root   30393 Jan 27 18:19  bcm2708-rpi-b.dtb
-rwxr-xr-x 1 root root   30012 Jan 27 18:19  bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 root root   31873 Jan 27 18:19  bcm2708-rpi-zero-w.dtb
-rwxr-xr-x 1 root root   30006 Jan 27 18:19  bcm2708-rpi-zero.dtb
-rwxr-xr-x 1 root root   32390 Jan 27 18:19  bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root   32181 Jan 27 18:19  bcm2709-rpi-cm2.dtb
-rwxr-xr-x 1 root root   32519 Jan 27 18:19  bcm2710-rpi-2-b.dtb
-rwxr-xr-x 1 root root   35346 Jan 27 18:19  bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root   34711 Jan 27 18:19  bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root   32298 Jan 27 18:19  bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root   33688 Jan 27 18:19  bcm2710-rpi-zero-2-w.dtb
-rwxr-xr-x 1 root root   33688 Jan 27 18:19  bcm2710-rpi-zero-2.dtb
-rwxr-xr-x 1 root root   56108 Jan 27 18:19  bcm2711-rpi-4-b.dtb
-rwxr-xr-x 1 root root   56112 Jan 27 18:19  bcm2711-rpi-400.dtb
-rwxr-xr-x 1 root root   38357 Jan 27 18:19  bcm2711-rpi-cm4-io.dtb
-rwxr-xr-x 1 root root   56654 Jan 27 18:19  bcm2711-rpi-cm4.dtb
-rwxr-xr-x 1 root root   53395 Jan 27 18:19  bcm2711-rpi-cm4s.dtb
-rwxr-xr-x 1 root root   52476 Mar  7 22:39  bootcode.bin
-rwxr-xr-x 1 root root      90 May 14  2024  cmdline.txt
-rwxr-xr-x 1 root root    3012 Jul  7  2024  config.txt
-rwxr-xr-x 1 root root    7328 Mar  7 22:39  fixup.dat
-rwxr-xr-x 1 root root    5459 Mar  7 22:39  fixup4.dat
-rwxr-xr-x 1 root root    3231 Mar  7 22:39  fixup4cd.dat
-rwxr-xr-x 1 root root    8455 Mar  7 22:39  fixup4db.dat
-rwxr-xr-x 1 root root    8453 Mar  7 22:39  fixup4x.dat
-rwxr-xr-x 1 root root    3231 Mar  7 22:39  fixup_cd.dat
-rwxr-xr-x 1 root root   10298 Mar  7 22:39  fixup_db.dat
-rwxr-xr-x 1 root root   10296 Mar  7 22:39  fixup_x.dat
-rwxr-xr-x 1 root root 7118288 Mar  7 22:39  kernel.img
-rwxr-xr-x 1 root root 6810592 Sep 11  2023  kernel7.img
-rwxr-xr-x 1 root root 7225232 Sep 11  2023  kernel7l.img
-rwxr-xr-x 1 root root 8219600 Sep 11  2023  kernel8.img
drwxr-xr-x 2 root root   20480 Feb 10 14:49  overlays
-rwxr-xr-x 1 root root 2987424 Mar  7 22:39  start.elf
-rwxr-xr-x 1 root root 2263328 Mar  7 22:39  start4.elf
-rwxr-xr-x 1 root root  813596 Mar  7 22:39  start4cd.elf
-rwxr-xr-x 1 root root 3761768 Mar  7 22:39  start4db.elf
-rwxr-xr-x 1 root root 3010984 Mar  7 22:39  start4x.elf
-rwxr-xr-x 1 root root  813596 Mar  7 22:39  start_cd.elf
-rwxr-xr-x 1 root root 4833736 Mar  7 22:39  start_db.elf
-rwxr-xr-x 1 root root 3734632 Mar  7 22:39  start_x.elf
```

Joulinar, 2025-04-01:
Should be an own mount point. Can you share following in addition 

```
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
```

maartenlangeveld, 2025-04-01:
> Should be an own mount point. Can you share following in addition
> 
> ```
> lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
> ```

```
2025-04-01 18:28:51 root@M-MUSIC-PLAYER:~# lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
NAME        FSTYPE LABEL   SIZE RO TYPE MOUNTPOINT   PARTUUID                             UUID
mmcblk0                  366.8G  0 disk
├─mmcblk0p1 vfat           128M  0 part              eb3ec0bf-01                          BA52-9502
├─mmcblk0p2 ext4          14.5G  0 part /            eb3ec0bf-02                          18cdc530-6cf7-4332-853c-b159b8d3638a
└─mmcblk0p3 ext4         352.2G  0 part /mnt/storage eb3ec0bf-03                          4f846044-7911-4435-a80b-bd6cc6c45d14
```

Joulinar, 2025-04-01:
Since RPi kernel 6.6, `vfat` partition is not mounted to `/boot` anymore. It should be `/boot/firmware/` nowadays. Can you check your `/etc/fstab` configuration?

maartenlangeveld, 2025-04-01:
> Since RPi kernel 6.6, `vfat` partition is not mounted to `/boot` anymore. It should be `/boot/firmware/` nowadays. Can you check your `/etc/fstab` configuration?

Mounted vfat partition to /boot/firmware (manually since dietpi-drive_manager could not be found by the shell, somehow path is not defined..). Rebooted, error message:

```
-bash: /boot/dietpi/func/dietpi-globals: No such file or directory
[FAILED] DietPi-Login | Failed to load DietPi-Globals. Skipping DietPi login scripts...
```

Manually re mounted vfat partition to /boot/firmware, but

```
2025-04-01 20:00:26 root@M-MUSIC-PLAYER:~# dietpi-drive_manager
-bash: dietpi-drive_manager: command not found
```

How can I fix this?

maartenlangeveld, 2025-04-01:
Replaced /boot/dietpi with /boot/firmware/dietpi in /boot/firmware/dietpi/dietpi-drive_manager.

Did run dietpi-drive_manager to mount vfat partition to /boot/firmware, get this error message:

```

          ┌───────────────────────────────────────────────┤ DietPi-Drive_Manager ├───────────────────────────────────────────────┐
          │                                                                                                                      │
          │ Invalid mount target location:                                                                                       │
          │  - /boot/firmware                                                                                                    │
          │                                                                                                                      │
          │ The drive will now be mounted to:                                                                                    │
          │  - /mnt/BA52-9502                                                                                                    │
          │                                                                                                                      │
          │                                                        <Ok>                                                          │
          │                                                                                                                      │
          └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

Joulinar, 2025-04-01:
Can you stop trying to fix this? We would need to clarify the actual status before doing any changes.

Can you share following 

```
cat /etc/fstab
```

maartenlangeveld, 2025-04-01:
> Can you stop trying to fix this? We would need to clarify the actual status before doing any changes.
> 
> Can you share following
> 
> ```
> cat /etc/fstab
> ```
It's all fucked up, I do not understand.

Joulinar, 2025-04-01:
you are not able to share any content from `/etc/fstab` ?

maartenlangeveld, 2025-04-01:
I do not know what went wrong. Update went wrong now everything seems to be messed up. Will re-insttall and close this thread.
Many thanks for your help, but this seems only good solution.

closed 2025-04-01 by maartenlangeveld

MichaIng, 2025-04-03:
Yeah on all RPi models with new firmware stack the FAT partition is intended to be mounted to `/boot/firmware`. If it is not, kernel and firmware package upgrades fail to configure, which is what you saw.
```
raspi-firmware: missing /boot/firmware, did you forget to mount it?
```

No idea why the mount was missing, maybe it failed at boot for some reason, or the `/etc/fstab` entry was missing for some reason. In such case check fstab, and if the entry is there, try:
```sh
mount /boot/firmware
```
and in case check why it was not present in the first place:
```sh
journalctl -u boot-firmware.mount
```

maartenlangeveld, 2025-04-03:
Hi MichaIng,

Thanks, I maybe accidently messed it up… An upgrade stalled, I rebooted and to be sure I restored from a weekly backup to another device (cron job). My weekly backup rsync command did not include --delete so the backup was contaminated with previous versions, that's why my /boot dirs had become messy.

Re-installed, learned a lesson and now everything works fine again.

Kind regards, Maarten