MichaIng/DietPi #7477: Failed install of Docker and Docker Compose via DietPi-Software

Logani06, 2025-04-03:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Details:
- Date           | 2025-04-03 10:21:36
- Bug report     | e406c57a-add8-4d6f-bfa8-821c3662cf7a
- Program name   | DietPi-Software
- Command        | `sed --follow-symlinks -i /root=/s/[[:blank:]]*$/ cgroup_enable=memory/ /boot/cmdline.txt`
- Exit code      | 4
- DietPi version | v9.11.2 (MichaIng/master)
- Distro version | bookworm (ID=7,RASPBIAN=0)
- Kernel version | `Linux DietPi-Arbeit 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux`
- Architecture   | `arm64`
- Hardware model | RPi 4 Model B (aarch64) (ID=4)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

1. Try to install Docker and Docker Compose via DietPi-Software
#### Expected behaviour:

- install Docker and Docker Compose
#### Actual behaviour:

- Error and no installation 
#### Additional logs:
```
sed: couldn't open temporary file /boot/firmware/seddKi2My: Read-only file system
```

Joulinar, 2025-04-03:
Can you check for kernel error messages? Looks like a faulty SD card.

```
dmesg -l 0,1,2,3
```

Logani06, 2025-04-03:
I think that could be posible. I have installed docker with the normal cli command and after a 10 mins or so the Pi froze and didn't came back. After reformating the SD Card it works again. Well see how long. 
Thx for the help :) 👍

closed 2025-04-03 by Logani06