MichaIng/DietPi #7478: dietpi-drive_manager stucks when NFS mount not available

maartenlangeveld, 2025-04-03:
When a device exporting a NFS share fails `dietpi-drive_manager` hangs (irresponsive) on the receiving end, i.e. the machine that has mounted the export. Also cannot list the directory containing the mount points. It is really annoying.

Is there a solution?

MichaIng, 2025-04-03:
Yes this is a known annoyance with the `x-systemd.automount` feature: systemd tries to mount the share once the mountpoint is accessed. But it tries that for at least 90 seconds, also if e.g. a hard drive is not plugged.

Hard to solve on a generic level. We could switch to `auto` mount, to just attempt the mount once at boot, but if it fails at that time, it is never tried again. And since the `network-online.target` has still not enough meaning, it is quite common that network drives fail to mount of the boot stage (if WPA takes longer on WiFi, or a DHCP takes longer), even when using the `netdev` option to assure their mount is attempted after that target.

Best solution we could find is to reduce the `DefaultTimeoutStartSec=` in `/etc/systemd/systemd.conf` to e.g. 15 seconds, which should make systemd abort the attempt after 15 seconds. Not great, also affects all other units (which do not override the default).

Great would be if systemd had an option to retry mounts which failed at boot after a defined time and defined attempts. Then we wouldn't need to use the `x-systemd.automount`. But currently I am not aware of such feature, other than defining `StartLimitBurst=` and `StartLimitIntervalSec=` explicitly for each mount individually (it if even works for mounts).

maartenlangeveld, 2025-04-03:
Hi @MichaIng,

Thanks for your quick reply!! I understand the difficulties. I have to live/deal with it.
Will close the issue.

Kind regards, Maarten

closed 2025-04-07 by maartenlangeveld

MichaIng, 2025-04-19:
Hmm, the 600 seconds might be an NFS specific default 🤔. Good to know, and seems unreasonably long.

Mid-term kinda solution would be to allow switching between `auto` and `x-systemd.automount` for drives in general.