MichaIng/DietPi #7481: Docker images do not restart after backup

FPSUsername, 2025-04-05:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version` | 9.11.2
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN` | ARMBIAN 25.2.3
- Kernel version | `uname -a` | 6.12.21
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3) | Ordroid N2+
- Power supply used | (EG: 5V 1A RAVpower) | official supply
- SD card used | (EG: SanDisk ultra) | samsung pro endurance 128gb

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) | Docker/Home Assistant Supervisor
- Was the software title installed freshly or updated/migrated? | Upgraded
- Can this issue be replicated on a fresh installation of DietPi? | Not tested

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Have home assistant supervisor running. You can check statusses with `ha core status` and `ha su status`.
2. Create a system backup with `dietpi-backup`
3. See that docker is stopped and restarted

#### Expected behaviour

1. Docker should start up and start all previously running containers
2. You can check that Home Assistant is running by executing the commands mentioned above. In addition you can check docker for running containers.

#### Actual behaviour

1. Docker starts up, but no containers are running

#### Extra details

- ...

Joulinar, 2025-04-05:
As you installed the container yourselves, you would need to check your container settings. I doubt there is anything we can do from DietPi side.

FPSUsername, 2025-04-05:
I think I found a solution, as you said I installed the containers (and services) myself, I had to add them to `dietpi-services` so dietpi can stop and restart them once the backup is completed.

closed 2025-04-05 by Joulinar

MichaIng, 2025-04-19:
Out of interest, do the containers start on boot, so they have own systemd services? If so makes sense. So they are not started with the Docker service, but require it (of course). So when Docker is stopped, those services forcefully stop as well, but do not start with Docker automatically.

There is a way to bind one systemd service to another, so that (re)starting/stopping one starts/stops the other in the same turn. Seems to be a fitting feature for such cases. But adding them to `dietpi-services` of course is a good solution for this particular case/environment as well.

https://www.freedesktop.org/software/systemd/man/latest/systemd.unit.html#BindsTo=
And the `PartOf=` setting, too.

FPSUsername, 2025-04-21:
Yes, on a system boot, HA supervisor starts up like normal, but I can't figure out how it starts up. There are a couple of systemd files that start up some processes, but I couldn't find the file which starts up the home assistant core (`ha core start`). As a quick resolution I created a systemd file that can start and stop the home assistant core (I can share it here later).