MichaIng/DietPi #7486: Bash shell breaks after giving command

kx1t, 2025-04-08:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.11 RC2 master/MichaIng`
- Distro version | `bookworm 0`
- Kernel version | `Linux coriovallum 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64) 4Gb`
- Power supply used |
- SD card used | `Samsung Evo Endurance 128 Gb`

#### Additional Information (if applicable)
- I replicated this issue on 2 other machines with DietPi 9.11RC2 (1x RPi4B/4Gb, 1x Intel x86 machine)

#### Steps to reproduce

1. Open a bash shell into a non-root account
2. Check that all works fine by giving some random commands
3. Provide this command: `if [[ ]]; then echo true; else echo false; fi`
4. Observer error messages printed after each subsequent command (see below)
5. Log out, log in, repeat steps 2-4 to replicate the issue
6. To stop seeing the issue within the same shell (without logging out), type any text and then interrupt with <CTRL-C> before pressing the <Return> key

#### Expected behaviour

- Command should return within the expected behavior of bash without any subsequent issues

#### Actual behaviour

- After each subsequent command, the error message (see below) is printed
- After waiting for some time, the error message is no longer printed after a command

#### Extra details

See this sequence:

```bash
pi@coriovallum:~$ echo hello
hello
pi@coriovallum:~$ echo this is a test
this is a test
pi@coriovallum:~$ if [[ ]]; then echo true; else echo false; fi
-bash: PROMPT_COMMAND: line 37: syntax error near unexpected token `-w'
-bash: PROMPT_COMMAND: line 37: `[[ -w '/tmp/dietpi-process.pid' ]] && rm -f /tmp/dietpi-process.pid &> /dev/null && echo -ne '\r\e[J'; '
pi@coriovallum:~$ echo hello
hello
-bash: PROMPT_COMMAND: line 38: syntax error near unexpected token `-w'
-bash: PROMPT_COMMAND: line 38: `[[ -w '/tmp/dietpi-process.pid' ]] && rm -f /tmp/dietpi-process.pid &> /dev/null && echo -ne '\r\e[J'; '
pi@coriovallum:~$ echo this is a test
this is a test
-bash: PROMPT_COMMAND: line 39: syntax error near unexpected token `-w'
-bash: PROMPT_COMMAND: line 39: `[[ -w '/tmp/dietpi-process.pid' ]] && rm -f /tmp/dietpi-process.pid &> /dev/null && echo -ne '\r\e[J'; '
pi@coriovallum:~$ some-random-text^C
pi@coriovallum:~$ echo hello this is a test
hello this is a test
pi@coriovallum:~$
```

Joulinar, 2025-04-08:
ok and where is this an issue with DietPi? Because DietPi themselves is not an own os, it's a set of bash script on top of Debian.

kx1t, 2025-04-08:
It only happens in DietPi and the error message clearly refers to something that DietPi must have added:
```
-bash: PROMPT_COMMAND: line 39: `[[ -w '/tmp/dietpi-process.pid' ]] && rm -f /tmp/dietpi-process.pid [...]
```

MichaIng, 2025-04-17:
Yeah, this is added to abort the `G_DIETPI-NOTIFY -2` processing animation in case the calling script is SIGKILLed or otherwise terminated a way where it cannot finish its EXIT trap anymore. Not elegant, but the best way I could find to assure there will never be an endlessly running processing animation left. I need to review this at some point.

However, the issue you face is due to `[[ ]]` not being a valid bash command. I leaves the `[[` bash conditional open, `]]` closes it only if there is any argument in between. This renders the `PROMPT_COMMAND` invalid. Interesting behavior. Can be fixed by either additionally close the conditional with `]]` or finish the (invalid) command with `;` or abort it with CTRL+C as you did. A `;` in front of the `PROMPT_COMMAND` should fix it DietPi-wise. Let me test .... hmm no a leading `;` is no valid syntax in general 🤔.

We need to find a better way to handle the processing animation, not using `PROMPT_COMMAND` as last resort. But a non-closing bash conditional in console is at least a rare case.