MichaIng/DietPi #7488: orange pi zero 3 freeze under load

milkywade, 2025-04-10:
### Creating a bug report/issue
- [ YES] I have searched the existing open and closed issues

#### Required Information
- DietPi version

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=11
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

bookworm

- Kernel version | `uname -a`

6.6.44-current-sunxi64 #1 SMP Sat Aug  3 06:54:42 UTC 2024 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

Orange Pi Zero 3 (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)

5V 1A

- SD card used | (EG: SanDisk ultra)

Sandisk

#### Additional Information (if applicable)
- Bug report ID | `echo $G_HW_UUID`

Reference code: 3112da94-1c6e-4470-b47d-cbf50150cf15

#### Steps to reproduce
1. tar + bzip a 21 mb text file
2. see yourself kicked out of ssh, all services disconnect.

#### Expected behaviour
- it should finish the command

#### Actual behaviour
- it requires a power cycle to work again

#### Extra details
- I wonder if the system is not having an issue with thermal throttling. It is on a passive sink (alu).
- using hostap. journalctl regularly sends "protocol 0000 is buggy, dev wlan0" error messages.

juckendesAuge, 2025-04-10:
Hello,
Test it with a 5V 3A Power Supply.
You have written your Power Supply is 5V 1A.

juckendesAuge, 2025-04-10:
> That might just be it thank you.
> 
> Please allow me a few days before closing.

If you have a power bank, you can use it for testing, but make sure that the USB port can supply at least 3 amps.

milkywade, 2025-04-13:
Hey - I just checked the power supply. It's a 3A. I updated the issue description.

Now it's a cheap charger model and it could be defective so next time it freezes I'll try with a laptop charger (that delivers at minimum 3A) .

Juliano-rb, 2025-04-15:
I faced the same issue with my Orange Pi Zero 3.
It tends to freeze during high processing tasks, such as:

- Installing large packages via apt (some times hangs at the "Unpacking..." stage)

- Building an application using Coolify (the system froze during the build process)

This eventually caused my flash card to become corrupted.

My hardware setup is the same as the provided by [milkywade](https://github.com/milkywade).
I couldn't check all the software at the moment.

juckendesAuge, 2025-04-15:
Format the microSD card, it must be empty for testing. 
Then download the tool "H2testw 1.4" from heise.de . You can use it to test the microSD card and all other USB sticks. 
For Linux there is the alternative "F3" but only via terminal. As far as I know there is no GUI.

Juliano-rb, 2025-04-15:
Hello @juckendesAuge,

> Use you a 5V, 3A Power Supply?

Yes.

> Format the microSD card, it must be empty for testing. Then download the tool "H2testw 1.4" from heise.de. You can use it to test the microSD card and all other USB sticks. For Linux there is the alternative "F3", but only via terminal. As far as I know, there is no GUI.

I followed your suggestion and tested my SD card. Turns out it's really poor quality — I had recently replaced it, and I’m not sure if the issue occurred before the change.

After running H2testw, the tool reported some problems with the card.

Thanks a lot for pointing this out!

@milkywade — maybe this could be the cause in your case as well?

juckendesAuge, 2025-04-16:
I only use Sandisk extreme pro microSD cards. And have some extreme pro SD for digital camera and 3D printer. None of them have broken yet. The 128gb version costs about 16 euros in Germany. I can only recommend it.

milkywade, 2025-04-18:
thank you both - putting the cards to the test at the moment

sp2ong, 2025-04-20:
I don't know if this isn't a problem characteristic of when the system starts up and the memory size is incorrectly recognized.

For example, if you have a 1 Gb RAM version, the system sometimes says you have 2 Gb and then when compiling the installation of large packages when memory is needed, the system starts using memory addresses that are not physically there and freezes the memory
It seems that with kernel 6.12.23 this phenomenon does not occur

Check after turning on the computer with the command:
free -m
whether the size shown matches the physical RAM that you have

milkywade, 2025-04-20:
free -m shows total: 918 which checks given it's a 1GB board.

I have not had the issue in a while even though the system is now solicited at regular intervals.
I'm thus inclined to put the issue on a bad sector of the SD card that got finally marked (or something similar).

Taking out the SD, formatting it and testing it as per above is going to take me a few weeks whilst I build a new system so I won't know for sure until then. Good news is that the new system will be using cards that are now tested :-)

juckendesAuge, 2025-04-20:
@milkywade You should also not forget that you have used a 5 volt 1 amp power supply. An Orange Pi Zero 3 is specified with 5 volts 3 amps. This means that 2 amps are missing. This must also have caused the instability.
Your problems with the SD card could also be the result of the power shortage of the Orange Pi Zero 3.

milkywade, 2025-04-20:
> [@milkywade](https://github.com/milkywade) You should also not forget that you have used a 5 volt 1 amp power supply. An Orange Pi Zero 3 is specified with 5 volts 3 amps. This means that 2 amps are missing. This must also have caused the instability. Your problems with the SD card could also be the result of the power shortage of the Orange Pi Zero 3.
> 
> Edit: I'm surprised that the Orange Pi booted at all with only 1 ampere.

I checked and I had a 3 am -> https://github.com/MichaIng/DietPi/issues/7488#issuecomment-2800093004

juckendesAuge, 2025-04-20:
That came after two other answers. i didn't realize that. why didn't you check that right away?
I could have used my time for something else.

milkywade, 2025-04-20:
because I did not had physical access to the machine at the moment of opening the ticket

juckendesAuge, 2025-04-20:
oh well ;)

MichaIng, 2025-05-06:
I pushed a new kernel and bootlloader to out APT repository could you upgrade and see whether it enhances things?
```sh
sudo apt update
sudo apt update
sudo /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
sudo reboot
```

juckendesAuge, 2025-05-06:
> I pushed a new kernel and bootlloader to out APT repository could you upgrade and see whether it enhances things?
> 
> sudo apt update
> sudo apt update
> sudo /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
> sudo reboot

why sudo?
unfortunately there is a root account with dietpi. i would also prefer to do everything as a user with sudo only and leave the root account deactivated, but i have read that this will only come with a new dietpie version.

i have an orange pi zero 3 with 4gb ram. should i also test it? but it always ran stable for me.

MichaIng, 2025-05-06:
Many do create an own login user or use the `dietpi` user, hence I started to add `sudo` for instructions which would require it. Also as `root` user, it does not hurt to use `sudo`.

> Borrowed from Armbian :)

Sure, for most SBCs with use an Armbian build system fork to generate kernel and U-Boot packages: https://github.com/MichaIng/build

MichaIng, 2025-05-06:
`dietpi-set_hardware flash-uboot-mmc` flashes the U-Boot binary from the `linux-u-boot-*` package to disk. As there were  bunch of other bootloader related issues, like that RAM size on 1 GB and 2 GB models were sometimes detected as 2 GB/4 GB, I did a rebuild for both.

juckendesAuge, 2025-05-06:
I did it. This time I got to ssh immediately after the reboot. Before I always had to wait 30 or more seconds.

juckendesAuge, 2025-05-06:
after updating the kernel and rebooting, it took just as long. only after the u-boot flashing was it faster, but that could have been a coincidence.

MichaIng, 2025-05-06:
Great. Let's hope it enhances stability as well.

juckendesAuge, 2025-05-06:
> Great. Let's hope it enhances stability as well.

I had no problems with stability before the new kernel. Probably because I have the 4GB version.

MichaIng, 2025-05-06:
Ah right, so maybe @milkywade finds time to give it a try:
```sh
sudo apt update
sudo apt update
sudo /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
sudo reboot
```

milkywade, 2025-05-06:
> I pushed a new kernel and bootlloader to out APT repository could you upgrade and see whether it enhances things?
> 
> sudo apt update
> sudo apt update
> sudo /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
> sudo reboot

Thanks! Just installed on my side. Will thrash it tomorrow... ;-)

MichaIng, 2025-05-06:
> P.S. The second line above is upgrade iso update

Ops 😅 fixed

milkywade, 2025-05-06:
> > I pushed a new kernel and bootlloader to out APT repository could you upgrade and see whether it enhances things?
> > sudo apt update
> > sudo apt update
> > sudo /boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
> > sudo reboot
> 
> Thanks! Just installed on my side. Will thrash it tomorrow... ;-)
> 
> P.S. The second line above is upgrade iso update

OK, I could not avoid thrashing it a bit already.

At first sight the system feels overall a bit more responsive, I seem to be able to go up to 32Mb/s on the TX side w/o major hiccups (using glances to monitor) and able to play streaming games on a tablet w/o issue.

System log was not all rosy though, I saw a big chunk of new red lines I had never seen before (the last two, the first two were there already w the last version):

```
kernel: net_ratelimit: 32 callbacks suppressed
kernel: protocol 0000 is buggy, dev wlan0
kernel: sprdwl:tx timeout drop num:7, l_num:378, maxnum:20000
kernel: sprdwl:TID=BE, timeout!

```

Will check again tomorrow, more thoroughly.

MichaIng, 2025-05-06:
This driver will never be error free. It is out of tree, badly maintained. Armbian does their best to keep it somewhat compatible with recent kernel versions. So as long as WiFi (and optionally Bluetooth) works, this is all we can expect: https://github.com/armbian/build/tree/main/patch/misc/wireless-uwe5622

This will only ever work flawlessly when added to actual mainline Linux code, but the vendor is not caring about this.

juckendesAuge, 2025-05-06:
> > P.S. The second line above is upgrade iso update
> 
> Ops 😅 fixed

oh,
I didn't notice that because I had already thought of the second line without reading it.
I use nala though. You only need a "sudo nala upgrade".
nala is not a standard tool, you have to install it once with apt.

MichaIng, 2025-05-06:
You should test again `apt` with Debian Trixie. They tweaked output format a lot, with colors and more structured indentation. No further frontend needed IMO 🙂. Only thing I do to quickly to an APT upgrade without DietPi update:
```sh
alias APT='apt clean && apt update && apt list --upgradeable && apt full-upgrade && apt autopurge'
```
`APT` then is my default command to bring any test system on latest stage, before doing anything else.

juckendesAuge, 2025-05-06:
> Hey - I just checked the power supply. It's a 3A. I updated the issue description.
> 
> Now it's a cheap charger model and it could be defective so next time it freezes I'll try with a laptop charger (that delivers at minimum 3A) .

Can you please tell us exactly which power supply you are using? I have a hunch.

milkywade, 2025-05-07:
Understood about the wifi driver. I wanted to note the error persisted in case it had anything to do with anything.

I had to hard cycle the machine today evening.

It was not possible for me to log in, after ssh'ing and entering the password there was an error message I did not capture where bash complained about needing a version. I had the same problem whether logging in via the eth cable or wifi. Also, checking the machine's "glances" through an already running browser tab showed a red "disconnected" message on the web page.

I am uploading here the full result of dmesg for your reference:

[dmesg.txt](https://github.com/user-attachments/files/20093763/dmesg.txt)

journalctl entries begin after the machine has rebooted, in the middle of an @reboot crontab command that waits 2 minutes to launch after reboot then takes 4 to execute. I see in dmesg a line with `systemd-journald[213]: Received client request to flush runtime journal`. /etc/systemd/journald.conf is all commented out.

P.S. I also have the above same output error as in the previous post after rebooting:
`sudo dmesg |grep -i error
[    1.488638] dw-apb-uart 5000000.serial: Error applying setting, reverse things back
[    1.488903] sun6i-spi 5010000.spi: Error applying setting, reverse things back
[    1.494651] sunxi-mmc 4020000.mmc: Error applying setting, reverse things back
[    1.495224] sunxi-mmc 4021000.mmc: Error applying setting, reverse things back
[    1.495412] dw-apb-uart 5000000.serial: Error applying setting, reverse things back
[    1.495673] sun6i-spi 5010000.spi: Error applying setting, reverse things back`

P.S.P.S. dmesg has also a very visual warning:
`[    7.791117] **********************************************************
[    7.791131] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
[    7.791143] **                                                      **
[    7.791156] ** trace_printk() being used. Allocating extra memory.  **
[    7.791169] **                                                      **
[    7.791181] ** This means that this is a DEBUG kernel and it is     **
[    7.791193] ** unsafe for production use.                           **
[    7.791206] **                                                      **
[    7.791218] ** If you see this message and you are not debugging    **
[    7.791231] ** the kernel, report this immediately to your vendor!  **
[    7.791243] **                                                      **
[    7.791255] **   NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE NOTICE   **
[    7.791268] **********************************************************`

MichaIng, 2025-05-08:
That "invalid protocol" every second is not great, spamming the logs. I'll check whether I can find it in the source code and possibly a patch.

I saw thr big debug warning as well. What is weird is that the same kernel does not throw this warning on other SBCs, at least on my NanoPi R4S it does not show up. So seems that sone driver loading triggers it. Was reported at Armbian as well a longer time ago, and I guess these dynamic debug flags are related: https://forum.armbian.com/topic/42488-kernel-debug-active-in-production/#comment-196824
Most likely they are needed for something if triggered, but we could just try to disable them and see what changes, with a test build.

Okay so overall the new kernel seems an improvement, better performance, but causes some concerning and spamming kernel log entries. And it does not solve your stability issues. Others reported that overclocking (manipulating the device tree) causes instabilities as well. A possibility is that you got a bad board based on chip variability, which does not run stable on the default max frequency either. If new PSU does not help, you could try to limit the max frequency to one p-state lower via `dietpi-config` performance options.

milkywade, 2025-05-08:
Hi - thanks for the review. I ordered a raspi psu, should get it nx week. If that does not change anything, I have a couple of spare opi03s to test, will post updates here.

juckendesAuge, 2025-05-20:
> Hi - thanks for the review. I ordered a raspi psu, should get it nx week. If that does not change anything, I have a couple of spare opi03s to test, will post updates here.

moin,
psu arrived?
:)

MichaIng, 2025-05-23:
@milkywade 
How do you use the WiFi adapter? As access point/hotspot? Because I do not see the "invalid protocol" errors, but there are reports at Armbian when using it with hostapd: https://forum.armbian.com/topic/36428-orange-pi-zero-3-wifi-ap-mode-protocol-0000-is-buggy-dev-wlan0/
And at Xunlong, also when using it as hotspot:
- https://github.com/orangepi-xunlong/linux-orangepi/issues/57
- https://github.com/orangepi-xunlong/linux-orangepi/issues/50

I btw most likely solved the `trace_printk` warning: https://github.com/MichaIng/build/commit/65dbdc6
The call was patched out for Rockchip kernel builds already, but there is no reason to keep it for any kernel build, unless one wants to actually debug something.

MichaIng, 2025-05-23:
Okay thanks. Hence not related to IPv6 in this case. I guess there are endless possibilities from where there faulty call could have been triggered.

Let's leave it for now. There are reports about this error on Orange Pi Zero 3 all around and I subscribed to most of them, just in case some more information is coming up. At least it does not seem to cause functional issues as the kernel handles the "buggy" protocol properly.

I'll mark the issue as closed then, starting to debug the more problematic false RAM size detection issue that bugs this SBC from day 0.

closed 2025-05-23 by MichaIng

milkywade, 2025-05-24:
thanks @MichaIng 

FYI I got a kernel oops when upgrading to the last kernel version today: https://github.com/MichaIng/DietPi/issues/7545
I had done a reboot just before to bring back ipv6.