MichaIng/DietPi #7489: Chromium won't run in headless mode on NanoPi Neo3 (headless SBC)

maartenlangeveld, 2025-04-10:
Installed Chromium via Dietpi-software. I need to run Chromium on my headless system but it won't start, it gets stuck:

System: 
1. NanoPi Neo3 (headless)
2. Linux M-CONTROL 6.6.56-current-rockchip64 #1 SMP PREEMPT Thu Oct 10 10:50:06 UTC 2024 aarch64 GNU/Linux
3. DietPi v9.11.2 (Bookworm)

```
2025-04-10 22:25:14 root@M-CONTROL:/home# chromium --headless --no-sandbox
[6501:6530:0410/222518.538734:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6501:6530:0410/222518.539273:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6501:6530:0410/222518.540569:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6501:6530:0410/222518.540960:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6544:6544:0410/222518.569920:ERROR:gl_factory.cc(103)] Requested GL implementation (gl=egl-gles2,angle=none) not found in allowed implementations: [(gl=egl-angle,angle=default)].
[6544:6544:0410/222518.579987:ERROR:viz_main_impl.cc(183)] Exiting GPU process due to errors during initialization
[6501:6530:0410/222518.603339:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6501:6530:0410/222518.685759:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6501:6530:0410/222518.759982:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6501:6501:0410/222519.071266:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[6501:6501:0410/222519.090662:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[6501:6530:0410/222519.091221:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6501:6530:0410/222519.091757:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[6501:6501:0410/222519.094475:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[6501:6501:0410/222519.104020:ERROR:global_accelerator_listener_linux.cc(350)] Failed to connect to signal: org.freedesktop.portal.GlobalShortcuts.Activated
[6581:6581:0410/222519.100131:ERROR:gl_factory.cc(103)] Requested GL implementation (gl=egl-gles2,angle=none) not found in allowed implementations: [(gl=egl-angle,angle=default)].
[6581:6581:0410/222519.129697:ERROR:viz_main_impl.cc(183)] Exiting GPU process due to errors during initialization
[6501:6589:0410/222519.157013:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[6501:6589:0410/222519.158762:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[6501:6589:0410/222519.160344:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[6501:6501:0410/222519.183568:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[6501:6501:0410/222519.185573:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[6590:6590:0410/222519.241562:ERROR:gl_factory.cc(103)] Requested GL implementation (gl=egl-gles2,angle=none) not found in allowed implementations: [(gl=egl-angle,angle=default)].
[6590:6590:0410/222519.277495:ERROR:viz_main_impl.cc(183)] Exiting GPU process due to errors during initialization
[6574:6582:0410/222519.641878:ERROR:command_buffer_proxy_impl.cc(127)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
```

maartenlangeveld, 2025-04-18:
Hi @MichaIng,

Thanks. It does not help,

```
2025-04-18 20:13:59 root@M-CONTROL:~# chromium
[819:819:0418/201537.321404:ERROR:ozone_platform_x11.cc(249)] Missing X server or $DISPLAY
[819:819:0418/201537.321661:ERROR:env.cc(257)] The platform failed to initialize.  Exiting.
```
```
2025-04-18 20:16:40 root@M-CONTROL:~# chromium --headless
[863:892:0418/201646.954778:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[863:892:0418/201646.955246:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[863:892:0418/201646.955429:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[863:892:0418/201646.955578:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[863:892:0418/201647.193353:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[863:892:0418/201647.453236:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[863:892:0418/201647.667755:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[863:863:0418/201648.416506:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[863:863:0418/201648.436502:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[863:892:0418/201648.436885:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[863:892:0418/201648.437060:ERROR:bus.cc(408)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[863:863:0418/201648.439588:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[863:863:0418/201648.452590:ERROR:global_accelerator_listener_linux.cc(350)] Failed to connect to signal: org.freedesktop.portal.GlobalShortcuts.Activated
[863:943:0418/201648.521809:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[863:943:0418/201648.523450:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[863:943:0418/201648.524693:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[863:863:0418/201648.539134:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[863:863:0418/201648.540140:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[936:944:0418/201649.265970:ERROR:command_buffer_proxy_impl.cc(127)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.

```

MichaIng, 2025-04-18:
Oh, it seems `dbus` is not installed. I guess you installed the package manually without X11, right?
```sh
sudo apt install dbus
```
Furthermore, possibly systemd PAM support might be needed:
```sh
systemctl unmask systemd-logind
systemctl start systemd-logind
apt install libpam-systemd
```
This is all done automatically when installing Chromium via `dietpi-software`, but it installs X11 as well, which I understand does not make sense for headless mode on a headless machine.

maartenlangeveld, 2025-04-18:
I did not install manually but using dietpi-software.

```

          ┌─────────────────────────────────────────────────┤ DietPi-Software ├──────────────────────────────────────────────────┐
          │ Use the spacebar to select the software you would like to remove:                                                    │
          │                                                                                                                      │
          │  [ ] 5    ALSA: Advanced Linux Sound Architecture                                                                    │
          │  [ ] 6    X.Org X Server: aka X11 - X Window System implementation                                                   │
          │  [ ] 17   Git: Clone and manage Git repositories locally                                                             │
          │  [ ] 84   Lighttpd: Extremely lightweight webserver                                                                  │
          │  [ ] 88   MariaDB: Persistent cached file-per-table database server                                                  │
          │  [ ] 89   PHP: Hypertext Preprocessor for dynamic web content                                                        │
          │  [ ] 96   Samba Server: Feature-rich SMB/CIFS server                                                                 │
          │  [ ] 103  DietPi-RAMlog: Makes /var/log a RAM disk, preserves file structure on reboot                               │
          │  [ ] 104  Dropbear: Lightweight SSH server                                                                           │
          │  [ ] 109  NFS Server: Network File System server                                                                     │
          │  [ ] 110  NFS Client: network file system client                                                                     │
          │  [ ] 113  Chromium: web browser for desktop or autostart                                                             │
          │  [ ] 129  O!MPD: Feature-rich, web interface audio player for MPD                                                    │
          │  [ ] 130  Python 3: Runtime system, pip package installer and development headers                                    │
          │  [ ] 170  UnRAR: unarchiver for .rar files                                                                           │
          │  [ ] 195  youtube-dl: Download videos from YouTube and other sites (using yt-dlp fork)                               │
          │  [ ] 196  Java JRE: OpenJDK Runtime Environment                                                                      │
          │                                                                                                                      │
          │                                  <Confirm>                                 <Back>                                    │
          │                                                                                                                      │
          └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

maartenlangeveld, 2025-04-18:
Ho can i best install chromium wihtout installing X11?

MichaIng, 2025-04-18:
Ah right, otherwise `/etc/chromium.d/dietpi` wouldn't be there. No point to uninstall Chromium. Calling it with `--headless` already fixes the "missing X11" error, so that works all as intended. The "Requested GL implementation (gl=egl-gles2,angle=none) not found in allowed implementations" error is gone as well. Now there are still errors related to failing DBus connection.

Can you check whether DBus is running as expected?
```sh
systemctl status dbus
```

maartenlangeveld, 2025-04-18:
> Oh, it seems `dbus` is not installed. I guess you installed the package manually without X11, right?
> 
> sudo apt install dbus
> Furthermore, possibly systemd PAM support might be needed:
> 
> systemctl unmask systemd-logind
> systemctl start systemd-logind
> apt install libpam-systemd
> This is all done automatically when installing Chromium via `dietpi-software`, but it installs X11 as well, which I understand does not make sense for headless mode on a headless machine.

```
2025-04-18 20:57:35 root@M-CONTROL:~# apt install libpam-systemd
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libpam-systemd is already the newest version (252.36-1~deb12u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

maartenlangeveld, 2025-04-18:
> Ah right, otherwise `/etc/chromium.d/dietpi` wouldn't be there. No point to uninstall Chromium. Calling it with `--headless` already fixes the "missing X11" error, so that works all as intended. The "Requested GL implementation (gl=egl-gles2,angle=none) not found in allowed implementations" error is gone as well. Now there are still errors related to failing DBus connection.
> 
> Can you check whether DBus is running as expected?
> 
> systemctl status dbus

```
2025-04-18 20:57:59 root@M-CONTROL:~# systemctl status dbus
● dbus.service - D-Bus System Message Bus
     Loaded: loaded (/lib/systemd/system/dbus.service; static)
     Active: active (running) since Fri 2025-04-18 20:13:45 CEST; 47min ago
TriggeredBy: ● dbus.socket
       Docs: man:dbus-daemon(1)
   Main PID: 293 (dbus-daemon)
      Tasks: 1 (limit: 2201)
     Memory: 1.9M
        CPU: 2.610s
     CGroup: /system.slice/dbus.service
             └─293 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only

Apr 18 20:13:43 M-CONTROL systemd[1]: Starting dbus.service - D-Bus System Message Bus...
Apr 18 20:13:45 M-CONTROL systemd[1]: Started dbus.service - D-Bus System Message Bus.
Apr 18 20:13:45 M-CONTROL dbus-daemon[293]: [system] Successfully activated service 'org.freedesktop.systemd1'
```

MichaIng, 2025-04-18:
Hmm, there might be some variables missing when skipping the X11 session. Can you try these steps: https://github.com/microsoft/WSL/issues/7915#issuecomment-2466607845
```sh
export XDG_RUNTIME_DIR=/run/user/$(id -u)
sudo mkdir "$XDG_RUNTIME_DIR"
sudo chmod 700 "$XDG_RUNTIME_DIR"
sudo chown "$(id -un):$(id -gn)" "$XDG_RUNTIME_DIR"
export DBUS_SESSION_BUS_ADDRESS="unix:path=$XDG_RUNTIME_DIR/bus"
```
If this alone does not work, try to start DBus with this address:
```sh
dbus-daemon --session --address="$DBUS_SESSION_BUS_ADDRESS" --nofork --nopidfile --syslog-only &
```

maartenlangeveld, 2025-04-18:
I'm sorry, it still doesn't work.

```
2025-04-18 22:40:29 root@M-CONTROL:~# chromium
[3682:3682:0418/224038.824560:ERROR:zygote_host_impl_linux.cc(105)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.
```
```
2025-04-18 22:40:38 root@M-CONTROL:~# chromium --no-sandbox
[3703:3703:0418/224050.472779:ERROR:ozone_platform_x11.cc(249)] Missing X server or $DISPLAY
[3703:3703:0418/224050.473016:ERROR:env.cc(257)] The platform failed to initialize.  Exiting.
```
```
2025-04-18 22:40:50 root@M-CONTROL:~# chromium --no-sandbox --headless
[3737:3766:0418/224058.545440:ERROR:bus.cc(408)] Failed to connect to the bus: Failed to connect to socket /run/user/0/bus: No such file or directory
[3737:3766:0418/224058.548430:ERROR:bus.cc(408)] Failed to connect to the bus: Failed to connect to socket /run/user/0/bus: No such file or directory
[3737:3766:0418/224058.548838:ERROR:bus.cc(408)] Failed to connect to the bus: Failed to connect to socket /run/user/0/bus: No such file or directory
[3737:3766:0418/224058.549531:ERROR:bus.cc(408)] Failed to connect to the bus: Failed to connect to socket /run/user/0/bus: No such file or directory
[3737:3766:0418/224058.611477:ERROR:bus.cc(408)] Failed to connect to the bus: Failed to connect to socket /run/user/0/bus: No such file or directory
[3737:3766:0418/224058.701782:ERROR:bus.cc(408)] Failed to connect to the bus: Failed to connect to socket /run/user/0/bus: No such file or directory
[3737:3766:0418/224058.769398:ERROR:bus.cc(408)] Failed to connect to the bus: Failed to connect to socket /run/user/0/bus: No such file or directory
[3737:3737:0418/224059.125129:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[3737:3737:0418/224059.139686:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[3737:3766:0418/224059.140527:ERROR:bus.cc(408)] Failed to connect to the bus: Failed to connect to socket /run/user/0/bus: No such file or directory
[3737:3766:0418/224059.141419:ERROR:bus.cc(408)] Failed to connect to the bus: Failed to connect to socket /run/user/0/bus: No such file or directory
[3737:3737:0418/224059.143546:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[3737:3737:0418/224059.158666:ERROR:global_accelerator_listener_linux.cc(350)] Failed to connect to signal: org.freedesktop.portal.GlobalShortcuts.Activated
[3737:3835:0418/224059.228409:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[3737:3835:0418/224059.230716:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[3737:3835:0418/224059.232667:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[3737:3737:0418/224059.236705:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
[3737:3737:0418/224059.237880:ERROR:object_proxy.cc(589)] Failed to call method: org.freedesktop.DBus.NameHasOwner: object_path= /org/freedesktop/DBus: unknown error type:
```

MichaIng, 2025-04-18:
But we are one step further now. The address is now correctly used by Chromium. Instead of "Could not parse server address", it does not try to connect to DBus via `/run/user/0/bus` as intended. Just the socket does not exist. Did you run the dedicated `dbus-daemon ...` command I provided?

Checking further, this seems to be the sane method to have the needed environment variable set and dbus launched:
```sh
dbus-launch chromium --no-sandbox --headless
```

maartenlangeveld, 2025-04-18:
```
2025-04-18 23:29:59 root@M-CONTROL:~# dbus-daemon --session --address="$DBUS_SESSION_BUS_ADDRESS" --nofork --nopidfile --syslog-only &
[1] 4525
```
```
2025-04-18 23:30:15 root@M-CONTROL:~# dbus-launch chromium --no-sandbox --headless
-bash: dbus-launch: command not found
```

MichaIng, 2025-04-18:
Ah, `dbus-launch` is part of the `dbus-x11` package. However, the output of Chromium indicates it the `dbus-daemon` works.

I am not sure what Chromium headless mode is supposed to do or which output is expected. It still does not work? The remaining errors are about missing upower, but that shouldn't prevent Chromium from starting:
```sh
apt install upower
systemctl start upower
```

maartenlangeveld, 2025-04-18:
Now chromium is running when using --headless switch! Thanks!

maartenlangeveld, 2025-04-18:
> Hmm, there might be some variables missing when skipping the X11 session. Can you try these steps: [microsoft/WSL#7915 (comment)](https://github.com/microsoft/WSL/issues/7915#issuecomment-2466607845)
> 
> export XDG_RUNTIME_DIR=/run/user/$(id -u)
> sudo mkdir "$XDG_RUNTIME_DIR"
> sudo chmod 700 "$XDG_RUNTIME_DIR"
> sudo chown "$(id -un):$(id -gn)" "$XDG_RUNTIME_DIR"
> export DBUS_SESSION_BUS_ADDRESS="unix:path=$XDG_RUNTIME_DIR/bus"
> If this alone does not work, try to start DBus with this address:
> 
> dbus-daemon --session --address="$DBUS_SESSION_BUS_ADDRESS" --nofork --nopidfile --syslog-only &

Hi @MichaIng,

After reboot I had to repeat following commands in order to start chromium again:

- export XDG_RUNTIME_DIR=/run/user/$(id -u)
- sudo mkdir "$XDG_RUNTIME_DIR"
- sudo chmod 700 "$XDG_RUNTIME_DIR"
- sudo chown "$(id -un):$(id -gn)" "$XDG_RUNTIME_DIR"
- export DBUS_SESSION_BUS_ADDRESS="unix:path=$XDG_RUNTIME_DIR/bus"
- dbus-daemon --session --address="$DBUS_SESSION_BUS_ADDRESS" --nofork --nopidfile --syslog-only &

How can I make these changes permanent?
And what should I do to start chromium at at startup (after boot)?

Thanks!
Maarten

MichaIng, 2025-04-19:
Interesting, so Chromium really requires upower (in headless mode) 🤔. I wonder whether this is true for kiosk mode as well. Most desktops come with upower, so it might have been missed.

About autostarting this: First of all, after reboot can you try this (for simplicity):
```sh
apt install dbus-x11
dbus-launch chromium --no-sandbox --headless
```
Then you could e.g. use the `dietpi-autostart` custom script option.

maartenlangeveld, 2025-04-21:
Hi @MichaIng,

Almost there! Thank you so much for your great help and support!!
There's still a non-fatal error message:

```
2025-04-21 14:35:12 root@M-CONTROL:~# cat /var/lib/dietpi/dietpi-autostart/custom.sh
#!/bin/bash
# DietPi-AutoStart custom script
# Location: /var/lib/dietpi/dietpi-autostart/custom.sh

dbus-launch chromium --no-sandbox --headless --disable-gpu
exit 0
```
```
2025-04-21 14:35:53 root@M-CONTROL:~# journalctl -u dietpi-autostart_custom
Apr 21 14:30:42 M-CONTROL systemd[1]: Started dietpi-autostart_custom.service - DietPi-AutoStart custom script.
Apr 21 14:33:55 M-CONTROL custom.sh[1620]: Warning: vkCreateInstance: Found no drivers!
Apr 21 14:33:55 M-CONTROL custom.sh[1620]: Warning: vkCreateInstance failed with VK_ERROR_INCOMPATIBLE_DRIVER
Apr 21 14:33:55 M-CONTROL custom.sh[1620]:     at CheckVkSuccessImpl (../../third_party/dawn/src/dawn/native/vulkan/VulkanError.cpp:106)
Apr 21 14:33:55 M-CONTROL custom.sh[1620]: [1620:1620:0421/143355.168555:ERROR:gpu_blocklist.cc(119)] Unable to get gpu adapter
Apr 21 14:33:55 M-CONTROL custom.sh[629]: [629:629:0421/143355.174178:ERROR:service_client.cc(36)] Unexpected on_device_model service disconnect: The device's GPU is not supported.
```

MichaIng, 2025-04-21:
Seems like it tried to use Vulkan. In case it still appears, the driver lib might help. But generally Chromium should work well without it/OpenGLES instead:
```sh
apt install mesa-vulkan-drivers
```

DerWeller, 2025-05-08:
Hello, I read the conversation with excitement because I have the same problem. But despite all the tips, the problem is still there for me. Chromium starts, the page hangs, and then the error messages appear - contacts to x server lost

MichaIng, 2025-05-20:
Did you install `upower`?

DerWeller, 2025-05-20:
No, I have an RPI 3 running Firefox in the meantime. Even on the desktop, Chrome is crashed.

MichaIng, 2025-05-20:
Hmm, all this stuff is actually not required. Just tested on a clean NanoPi NEO3:
```sh
apt install chromium
dbus-run-session chromium --headless --no-sandbox --dump-dom https://dietpi.com/
```
Throws the `upower` warnings, so those could be muted, but otherwise works. `dbus-run-session` is the counterpart to `dbus-launch` from the `dbus-user-session` package, which is installed by default (compared to `dbux-x11`).

I did not install X11 or anything else, all not needed for headless run, of course.

Even better, there is a headless version of Chromium, which requires nothing of all this stuff, no dbus either. It throws warnings regarding dbus, but works without it.
```sh
apt install chromium-headless-shell
chromium-headless-shell --no-sandbox --dump-dom https://dietpi.com/
```
Better of course to not run it as root, but e,g, as `dietpi` user, and skip `--no-sandbox`.

DerWeller, 2025-05-21:
Thanks for your work. I will test it when I get the chance.

MichaIng, 2025-09-21:
Closing this for now. Feel free to reopen if the issue persists.

closed 2025-09-21 by MichaIng

wangjia184, 2025-09-30:
will try