MichaIng/DietPi #7490: Orange Pi 3B v1.1 Ethernet adapter not working

iivailo, 2025-04-11:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `v9.11.2 (MichaIng/master)`
- Distro version | `bookworm (ID=7)`
- Kernel version | `Linux DietPi 6.12.7-current-rockchip64 #1 SMP PREEMPT Fri Dec 27 13:02:20 UTC 2024 aarch64 GNU/Linux`
- SBC model | `Orange Pi 3B (aarch64) (ID=87)`
- Power supply used | `5V 3A RAVpower`
- SD card used | `SanDisk Extreme 64GB`

#### Steps to reproduce
Simply turn on the SBC.

#### Expected behaviour
The SBC receives an IP address from my DHCP server.

#### Actual behaviour
The SBC isn't receiving an IP address. The Diet Pi network configuration tool reports that the network cable is disconnected (the cable is connected, and the network works on Armbian with Linux kernel 6.1).

#### Extra details
dmesg: https://pastebin.com/8nAxLrqW

MichaIng, 2025-04-18:
Many thanks for your report. Can you check whether the correct device tree has been applied, i.e. does the following show v1.1 revision?
```sh
cat /proc/device-tree/model
```

MichaIng, 2025-04-18:
Now I remember that I was working on this already 😅: https://dietpi.com/forum/t/after-upgrade-emmc-died-and-the-new-installation-via-sd-card-does-not-connect-to-the-network/22129/42?u=michaing

https://github.com/MichaIng/build/commit/83a6e89

I will try to remember the next step/attempt I wanted to do. Good to have an issue on GitHub about this now.

MichaIng, 2025-04-18:
New build on the way: https://github.com/MichaIng/DietPi/actions/runs/14541707500
Once this is done, could you repeat above steps to install it? I adjusted `reset-assert-us` and `reset-deassert-us` to match the `snps,reset-delays-us` from the vendor kernel.

iivailo, 2025-04-19:
I've installed the new kernel version. The network isn't working yet, but there is some progress. With the stock kernel, the LED of the Ethernet port is off as is the LED of the switch. With the new version, the green LED on the SBC's Ethernet port indicates activity, and the LED on the switch is also on. The SBC isn't receiving an IPv4 address from my DHCP server. I've set a static IP address, but I can't ping any IP address on my network. The ping indicates a complete packet loss. Here are the new dmesg: https://pastebin.com/wCE2KLuu

MichaIng, 2025-04-19:
Okay, I mixed a bunch of different fixes/methods in this device tree. I'll try to clean it up, first of all using thr very same method used in the old vendor kernel device tree.

Vyacheslav-S, 2025-04-21:
Hello everyone!
I have solved the problem with the Internet adapter.
I have the first version Orange Pi 3B.
**DietPi:**  - v9.12.1 
**Kernel version**  - Linux opi3b 6.12.7-current-rockchip64 #1 SMP PREEMPT Fri Dec 27 13:02:20 UTC 2024 aarch64 GNU/Linux.

 I used dtb from orange-xunlong and the network now work without any problems _(rk3566-orangepi-3b.dtb)_.

```
dietpi@opi3b ~> cat /boot/dietpiEnv.txt
rootdev=UUID=dc5713dd-2e43-420c-8f91-3384feaf8672
rootfstype=ext4
consoleargs=console=ttyS2,1500000 console=tty1
fdtfile=rockchip/rk3566-orangepi-3b.dtb
usbstoragequirks=
extraargs=net.ifnames=0
docker_optimizations=off
overlay_path=rockchip
overlay_prefix=rockchip
overlays=
user_overlays=
```
```
dietpi@opi3b ~> dmesg |grep eth0
[   11.787423] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   11.790003] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-0:00] driver [YT8531 Gigabit Ethernet] (irq=POLL)
[   11.790050] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[   11.790077] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   11.790397] rk_gmac-dwmac fe010000.ethernet eth0: registered PTP clock
[   11.795128] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii link mode
[   14.854111] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 6368.714461] rk_gmac-dwmac fe010000.ethernet eth0: Link is Down
[ 6368.865837] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 6368.868706] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-0:00] driver [YT8531 Gigabit Ethernet] (irq=POLL)
[ 6368.877672] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[ 6368.877706] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[ 6368.878101] rk_gmac-dwmac fe010000.ethernet eth0: registered PTP clock
[ 6368.878136] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii link mode
[ 6371.941707] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 6555.228508] rk_gmac-dwmac fe010000.ethernet eth0: Link is Down
[ 6555.358595] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 6555.361329] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-0:00] driver [YT8531 Gigabit Ethernet] (irq=POLL)
[ 6555.369627] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[ 6555.369668] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[ 6555.370154] rk_gmac-dwmac fe010000.ethernet eth0: registered PTP clock
[ 6555.370196] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii link mode
[ 6558.437657] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx

```

[rk3566-orangepi-3b.zip](https://github.com/user-attachments/files/19835703/rk3566-orangepi-3b.zip)

I also wrote a small script to control the cooler from 2 pin PWM:
https://github.com/Vyacheslav-S/opi3b-pwm

MichaIng, 2025-04-22:
Just to rule it out: I created another build with only the reset delays adjusted (the modern way), reverting all other changes. This is to be sure that modern way to set reset delays in PHY does not just conflict with `rgmii` PHY mode, compared to `rgmii-id`:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-rockchip64.deb
sudo dpkg -i linux-{image,dtb}-current-rockchip64.deb
sudo reboot
```

Vyacheslav-S, 2025-04-23:
I installed it, and the system booted up. 
The Ethernet adapter is working.
```
~$ uname -a
Linux opi3b 6.12.24-current-rockchip64 #1 SMP PREEMPT Sun Apr 20 08:16:02 UTC 2025 aarch64 GNU/Linux
```
```
~$ dmesg |grep eth0
[   12.360467] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   12.363294] rk_gmac-dwmac fe010000.ethernet eth0: PHY [stmmac-0:01] driver [YT8531 Gigabit Ethernet] (irq=POLL)
[   12.363351] rk_gmac-dwmac fe010000.ethernet eth0: No Safety Features support found
[   12.363378] rk_gmac-dwmac fe010000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
[   12.363776] rk_gmac-dwmac fe010000.ethernet eth0: registered PTP clock
[   12.368360] rk_gmac-dwmac fe010000.ethernet eth0: configuring for phy/rgmii-id link mode
[   15.436727] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
dietpi@opi3b:~$ 

```
But there is a small problem - cooler on the 2-pin does not work.
In rk3566-orangepi-3b-v1.1.dtb, I did not find any mention of a: 

```
pwm-fan {
......
......
};
```
  I decompiled rk3566-orangepi-3b-v1.1.dtb, made changes, assembled, installed, and now the cooler control is working through the [service](https://github.com/Vyacheslav-S/opi3b-pwm), depending on the CPU temperature.
rk3566-orangepi-3b-v1.1.dtb and rk3566-orangepi-3b-v1.1.dts are archived.

[rk3566-orangepi-3b-v1.zip](https://github.com/user-attachments/files/19861758/rk3566-orangepi-3b-v1.zip)

iivailo, 2025-04-23:
I installed the last version of the packages, and with them, I have the same behavior as with the stock kernel. The LEDs of the Ethernet port are off, the LED of the switch is also off and the network adapter is not working. Here are the errors related to the Ethernet adapter:

> [    0.000000] psci: probing for conduit method from DT.
[    4.751586] rk_gmac-dwmac fe010000.ethernet: IRQ eth_lpi not found
[    4.752278] rk_gmac-dwmac fe010000.ethernet: IRQ sfty not found
[    4.768273] rk_gmac-dwmac fe010000.ethernet: clock input or output? (input).
[    4.768969] rk_gmac-dwmac fe010000.ethernet: Can not read property: tx_delay.
[    4.769628] rk_gmac-dwmac fe010000.ethernet: set tx_delay to 0x30
[    4.770197] rk_gmac-dwmac fe010000.ethernet: Can not read property: rx_delay.
[    4.770849] rk_gmac-dwmac fe010000.ethernet: set rx_delay to 0x10
[    4.771432] rk_gmac-dwmac fe010000.ethernet: integrated PHY? (no).
[    4.773312] rk_gmac-dwmac fe010000.ethernet: clock input from PHY
[    4.779119] rk_gmac-dwmac fe010000.ethernet: init for RGMII_ID
[    4.780304] rk_gmac-dwmac fe010000.ethernet: User ID: 0x30, Synopsys ID: 0x51
[    4.781020] rk_gmac-dwmac fe010000.ethernet: 	DWMAC4/5
[    4.781529] rk_gmac-dwmac fe010000.ethernet: DMA HW capability register supported
[    4.782224] rk_gmac-dwmac fe010000.ethernet: RX Checksum Offload Engine supported
[    4.782908] rk_gmac-dwmac fe010000.ethernet: TX Checksum insertion supported
[    4.783554] rk_gmac-dwmac fe010000.ethernet: Wake-Up On Lan supported
[    4.784428] rk_gmac-dwmac fe010000.ethernet: TSO supported
[    4.784966] rk_gmac-dwmac fe010000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    4.785703] rk_gmac-dwmac fe010000.ethernet: Enabled RFS Flow TC (entries=10)
[    4.786360] rk_gmac-dwmac fe010000.ethernet: TSO feature enabled
[    4.786917] rk_gmac-dwmac fe010000.ethernet: Using 32/32 bits DMA host/device width
[   11.063493] rk_gmac-dwmac fe010000.ethernet eth0: Register MEM_TYPE_PAGE_POOL RxQ-0
[   11.064811] rk_gmac-dwmac fe010000.ethernet eth0: __stmmac_open: Cannot attach to PHY (error: -19)

Vyacheslav-S, 2025-04-23:
@iivailo, show the output:
`cat /boot/dietpiEnv.txt`

iivailo, 2025-04-23:
@Vyacheslav-S under Armbian the Ethernet adapter is working fine.

Vyacheslav-S, 2025-04-23:
@iivailo,
`[   11.064811] rk_gmac-dwmac fe010000.ethernet eth0: __stmmac_open: Cannot attach to PHY (error: -19) `
It seems that the error is that the PHY ID is set incorrectly in the device tree.
Try the following:
Download **rk3566-orangepi-3b.dtb**
https://github.com/MichaIng/DietPi/issues/7490#issuecomment-2818695722
Copy to _/boot/dtb/rockchip._
Edit it dietpiEnv.txt:
```
rootdev=UUID=dc5713dd-2e43-420c-8f91-3384feaf8672
rootfstype=ext4
# The init system logs to the console defined last.
consoleargs=console=ttyS2,1500000 console=tty1
fdtfile=rockchip/rk3566-orangepi-3b.dtb
usbstoragequirks=
extraargs=net.ifnames=0
docker_optimizations=off
overlay_path=rockchip
overlay_prefix=rockchip
overlays=
user_overlays=
```
And reboot.

iivailo, 2025-04-23:
@Vyacheslav-S with your dtb file and the last packages test packages the network is working on my Orange Pi 3B board.

iivailo, 2025-04-23:
I created a new SD card with DietPi 9.12 and only changed the device tree file and the Ethernet adapter works. I used the file provided by Vyacheslav-S.

MichaIng, 2025-04-23:
@Vyacheslav-S 
Also for you it only works with the vendor device tree, right? The device tree is all I am changing, so makes sense, and hence the modern way to set those reset delays does not work, for whatever reasons.

Regarding the fan, how does your entry look like? This is the one from the vendor kernel device tree:
```
	fan: pwm-fan {
		compatible = "pwm-fan";
		#cooling-cells = <2>;
		pwms = <&pwm7 0 50000 0>;
		cooling-levels = <0 50 100 150 200 255>;
		rockchip,temp-trips = <
			50000   1
			55000   2
			60000   3
			65000   4
			70000   5
		>;
	};
```

iivailo, 2025-04-23:
I created a new SD card with DietPi 9.12 and after installing the latest version(from today) of the device tree and kernel, the Ethernet adapter works.

MichaIng, 2025-04-23:
Many thanks for testing, and great that we found the solution. I will push the build to our APT server and create a new Orange Pi 3B image as well based on this.

Vyacheslav-S, 2025-04-24:
@MichaIng, I'm sorry for the slight delay in responding, but at that time I had a night and I was sleeping))

```
--- /home/dietpi/tmp/dtb/bk/rk3566-orangepi-3b-v1.1.dts
+++ /home/dietpi/tmp/dtb/rk3566-orangepi-3b-v1.1.dts
@@ -2122,7 +2122,7 @@
 		pinctrl-0 = <0xa5>;
 		pinctrl-names = "default";
 		#pwm-cells = <0x03>;
-		status = "disabled";
+		status = "okay";
 		phandle = <0x121>;
 	};
 
@@ -4778,6 +4778,14 @@
 	chosen {
 		stdout-path = "serial2:1500000n8";
 	};
+        pwm-fan {
+	compatible = "pwm-fan";
+	#cooling-cells = <0x02>;
+	pwms = <0x121 0x00 0xc350 0x00>;
+	cooling-levels = <0x00 0x32 0x64 0x96 0xc8 0xff>;
+	rockchip,temp-trips = <0xc350 0x01 0xd6d8 0x02 0xea60 0x03 0xfde8 0x04 0x11170 0x05>;
+	phandle = <0x276>;
+        };
 
 	hdmi-con {
 		compatible = "hdmi-connector";

```
I made these changes in a previous comment -  https://github.com/MichaIng/DietPi/issues/7490#issuecomment-2823174630
They were in _rk3566-orangepi-3b-v1.zip._ 
These changes are taken from the original RK3566-orangepi-3b.db from orangepi-xunlong.

_PS; I'm not as well versed in the device tree as you are yet. I'm just looking for something, reading, doing experiments._

MichaIng, 2025-04-24:
@Vyacheslav-S 
Thanks. The node which you switched from disabled to okay, is it the `pwm7` one, resp. `pwm@fe6e0030`?

Vyacheslav-S, 2025-04-25:
@MichaIng,  
Yes, this one:
```
	pwm@fe6e0030 {
		compatible = "rockchip,rk3568-pwm\0rockchip,rk3328-pwm";
		reg = <0x00 0xfe6e0030 0x00 0x10>;
		clocks = <0x0f 0x15a 0x0f 0x159>;
		clock-names = "pwm\0pclk";
		pinctrl-0 = <0xa5>;
		pinctrl-names = "default";
		#pwm-cells = <0x03>;
		status = "okay";
		phandle = <0x121>;
	};
```

MichaIng, 2025-04-28:
Okay, I added support: https://github.com/MichaIng/build/commit/6da6cbd
Build running: https://github.com/MichaIng/DietPi/actions/runs/14714807872

MichaIng, 2025-09-09:
Got it. Not sure whether it makes a difference compared to manual `ip link set up` + `dhclient` call, but to experiment:
```sh
echo -e 'allow-hotplug eth1\niface eth1 inet dhcp' | sudo tee /etc/network/interfaces.d/eth1.conf
ifup eth1
```
Or to test with a static IP without gateway:
```sh
ifdown eth1
echo -e 'allow-hotplug eth1\niface eth1 inet static\naddress 192.168.1.157/24' | sudo tee /etc/network/interfaces.d/eth1.conf
ifup eth1
```
The `allow-hotplug eth1` could be also skipped, but triggers the `ifup@eth1.service` at boot, which basically calls `ifup eth1` once the device is registered.

Also check for kernel errors:
```sh
dmesg -l 0,1,2,3
```

> To resolve ethernet issue there was a shell code innerpart of which I don't understand but it worked!!! And I basically wrote a daemon to execute this script with every boot.

Hmm, I don't know the `io` command either. Do you still have/find the link to that Orange Pi post? Seems then your board has an issue that is not related to v2.1 vs v1.1 revisions. Maybe we can apply the same fix/workaround on a DietPi system, but need to find out where this `io` command is coming from.

... okay found a related issue: https://github.com/orangepi-xunlong/orangepi-build/issues/146
Looks like we need to check Xunlong's Ubuntu image regarding the `io` command.

MichaIng, 2026-01-01:
@AdLf0208 
Sorry for the late reply. First of all, after reading through related conversations, this is a hardware damage that seems to appear on revision 1.1 after a while. Probably one of the reasons why revision 2.1 uses an entirely different Ethernet chip. So there is no "fix" we can apply our end. The `io` commands seem to work around the issue, but do not fix the underlying hardware damage.

The `io` command is a blob vom Xunlong, no sources, no info what it is and does exactly, but just added for Orange Pi 3B and CM4: https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/packages/bsp/rk356x/usr/local/bin/io

You can download it manually to apply your workaround on a DietPi system:
```sh
sudo wget 'https://raw.githubusercontent.com/orangepi-xunlong/orangepi-build/refs/heads/next/external/packages/bsp/rk356x/usr/local/bin/io' -O /usr/local/bin/io
sudo chmod +x /usr/local/bin/io
```
But without knowing what the underlying hardware damage is, what this command does exactly, and why it helps, I won't add this to our images.

AdLf0208, 2026-01-01:
> @AdLf0208
> Sorry for the late reply. First of all, after reading through related conversations, this is a hardware damage that seems to appear on revision 1.1 after a while. Probably one of the [...]

This is exactly why I bought Realtek adapter,connected it via usb 3.0 and gave up on that idea (btw, I did the exact commands via io binary which helped on the vendor kernel but no success on dietpi kernel). So,thanks for going through the issue!!! You are truly a dedicated person)))

MichaIng, 2026-01-01:
> I did the exact commands via io binary which helped on the vendor kernel but no success on dietpi kernel

Ah okay. Maybe because Xunlong images use own kernel sources which might be closer to the Rockchip vendor kernel sources than mainline Linux: https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-6.6-rk35xx

MichaIng, 2026-01-01:
Btw, related to #7728 for which we needed to remove the forced `sprdwl_ng` from Orange Pi 3B images to fix boot on revision 2.1: The v1.1 device tree contains a patch which might load this driver automatically, while on all other Orange Pi SBCs with this WiFi chip, it needs to be loaded manually: https://github.com/armbian/build/blob/main/patch/kernel/archive/rockchip64-6.12/board-orangepi3b-add-uwe5622-wifi-bt-nodes.patch

@AdLf0208 If you have WiFi disabled, when you `sudo rm /etc/modprobe.d/dietpi-disable_wifi.conf`, does the module load automatically for you on boot, i.e. does the `wlan0` interface show up in `ip l`?

And similarly, does `sprdbt_tty` load automatically if you `sudo rm /etc/modprobe.d/dietpi-disable_bluetooth.conf`?