MichaIng/DietPi #7491: Why there is no /dev/video0 on RPi3B+ ?

danergo, 2025-04-16:
I don't have any "video" devices.

I wish to use v4l2m2m with ffmpeg to decode and encode h264 and hevc from rtps stream to local disk.

v4l2m2m needs a video interface, which I don't have.

Kernel didn't contained v4l2 module, but I could add:

> modprobe bcm2835-v4l2

But still I don't have any video device.

Is it intended? What is the current way of using video capabilities of RPi3B+? I've read both mmal and omx are deprecated, and v4l2m2m is the way now. But for me it's not working.

My kernel is: 6.12.1-v8+
My OS: Debian 12 (DietPi)

What would you suggest?

MichaIng, 2025-04-18:
Just to be sure: you do not want to use an RPi camera module, and no actual camera device, but just convert a network camera stream via FFmpeg, right?

I wonder whether a `/dev/video*` device appears at all if no actual local camera is attached. Generally, to enable the modern RPi camera module API, KMS/DRM needs to be enabled:
```sh
G_CONFIG_INJECT 'dtoverlay=vc4-kms-v3d' 'dtoverlay=vc4-kms-v3d' /boot/dietpi.txt
```

And yes, MMAL, OpenMAX and DispmanX APIs have been deprecated and related libraries removed with recent RPi firmware packages.

danergo, 2025-04-18:
Thanks for confirming! 

I don't have any local camera attached, and yes, I wish only to convert network camera stream (to test performance of my custom ffmpeg-build).

In this case I will just have to add your recommendation into my config.txt?

Thank you!

MichaIng, 2025-04-18:
Tbh I am not sure. The KMS/DRM API is required for the modern libcamera API to work, but not sure whether the same is true for v4l2m2m and it does not add a `/dev/video*` device if there is no actual camera attached.

danergo, 2025-04-18:
> G_CONFIG_INJECT 'dtoverlay=vc4-kms-v3d' 'dtoverlay=vc4-kms-v3d' /boot/dietpi.txt

Is it intended to have 'dtoverlay=vc4-kms-v3d' twice?

How can I verify this config has been applied (before I try ffmpeg)?

danergo, 2025-04-18:
Unfortunately I don't have neither drm nor dri in the folders mentioned.

danergo, 2025-04-18:
Okay, I have found it also. Now I have `/dev/dri` and `/sys/class/drm`.

However, ffmpeg still runs onto error with this:

> ./ffmpeg -c:v h264_v4l2m2m -rtsp_transport tcp -i 'rtsp://admin:admin@camip:camport' -c:v h264_v4l2m2m -pix_fmt yuv420p -b:v 128k -an test.mp4

> [h264_v4l2m2m @ 0x246f4260] Could not find a valid device
> [h264_v4l2m2m @ 0x246f4260] can't configure decoder

MichaIng, 2025-04-18:
Reading a bit through it, `h264_v4l2m2m` really seems to meant for V4L2 hardware video devices only. Have you tried the regular `h264` codec via `-c:v h264`?

Joulinar, 2025-04-18:
Could that be a similar case? https://dietpi.com/forum/t/raspberry-pi-3-b-camera-issues-after-upgrade/23280

danergo, 2025-04-18:
> Have you tried the regular `h264` codec via `-c:v h264`?

No, not yet because I thought I need to request hw encode/decode to apply it correctly to streams.

How can I verify (in case I use plain h264 codec) whether it is sw or hw decoding/encoding?

> Could that be a similar case? https://dietpi.com/forum/t/raspberry-pi-3-b-camera-issues-after-upgrade/23280

No I don't think so. In my case this is an rtsp network camera (not a local one).

Thank you!

MichaIng, 2025-04-18:
I found the info here btw: https://github.com/arthenica/ffmpeg-kit/issues/287#issuecomment-1002539894

danergo, 2025-04-20:
Tried now with h264 for both decoder and encoder:

Log:
```
[graph -1 input from stream 0:0 @ 0x7f70001ae0] w:1920 h:1080 pixfmt:yuv420p tb:1/90000 fr:25/1 sar:0/1 csp:unknown range:unknown
[format @ 0x7f70002060] Setting 'color_ranges' to value 'tv'
[AVFilterGraph @ 0x7f70000f70] query_formats: 4 queried, 9 merged, 0 already done, 0 delayed
[graph -1 input from stream 0:0 @ 0x7f70001ae0] video frame properties congruent with link at pts_time: 2.640733
[h264_v4l2m2m @ 0x16ee26f0] Could not find a valid device
[h264_v4l2m2m @ 0x16ee26f0] can't configure encoder
[vost#0:0/h264_v4l2m2m @ 0x16fa3660] [enc:h264_v4l2m2m @ 0x16ee2670] Error while opening encoder - maybe incorrect parameters such as bit_rate, rate, width or height.
[vf#0:0 @ 0x16f8f610] Error sending frames to consumers: Invalid argument
[h264 @ 0x16f30810] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[vf#0:0 @ 0x16f8f610] Task finished with error code: -22 (Invalid argument)
[h264 @ 0x16f41ad0] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[vf#0:0 @ 0x16f8f610] Terminating thread with return code -22 (Invalid argument)
[vost#0:0/h264_v4l2m2m @ 0x16fa3660] [enc:h264_v4l2m2m @ 0x16ee2670] Encoder thread received EOF
[vost#0:0/h264_v4l2m2m @ 0x16fa3660] [enc:h264_v4l2m2m @ 0x16ee2670] Could not open encoder before EOF
[vost#0:0/h264_v4l2m2m @ 0x16fa3660] Task finished with error code: -22 (Invalid argument)
[vost#0:0/h264_v4l2m2m @ 0x16fa3660] Terminating thread with return code -22 (Invalid argument)
[h264 @ 0x16ee18c0] nal_unit_type: 1(Coded slice of a non-IDR picture), nal_ref_idc: 2
[vist#0:0/h264 @ 0x16fba920] [dec:h264 @ 0x16f8fc70] Decoder returned EOF, finishing
[vist#0:0/h264 @ 0x16fba920] [dec:h264 @ 0x16f8fc70] Terminating thread with return code 0 (success)
[vist#0:0/h264 @ 0x16fba920] All consumers of this stream are done
[in#0/rtsp @ 0x16eb3690] All consumers are done
[in#0/rtsp @ 0x16eb3690] Terminating thread with return code 0 (success)
[out#0/mp4 @ 0x16fd0d20] Nothing was written into output file, because at least one of its streams received no packets.
frame=    0 fps=0.0 q=0.0 Lsize=       0KiB time=N/A bitrate=N/A speed=N/A    
[AVIOContext @ 0x16f53040] Statistics: 0 bytes written, 0 seeks, 0 writeouts
[in#0/rtsp @ 0x16eb3690] Input file #0 (rtsp://admin:dmann87!!!@192.168.87.205:8554):
[in#0/rtsp @ 0x16eb3690]   Input stream #0:0 (video): 140 packets read (439526 bytes); 6 frames decoded; 0 decode errors; 
[in#0/rtsp @ 0x16eb3690]   Total: 140 packets (439526 bytes) demuxed
Conversion failed!
```

Command:

> ffmpeg -c:v h264 -rtsp_transport tcp -i 'rtsp://camera' -c:v h264 -b:v 1024k -an test.mp4 -loglevel debug

MichaIng, 2025-04-21:
Hmm it still tries to use `h264_v4l2m2m`. Can you test with `-c:v libx264`?

Might be needed to install `apt install libx264-164`.

danergo, 2025-04-21:
Well, `libx264` is providing software decoding/encoding, isn't it?

My goal is to utilize hardware decoding/encoding. Do you think libx264 still worth trying?

MichaIng, 2025-04-21:
Ehh yeah you are right. Does this help?
```sh
/boot/dietpi/func/dietpi-set_hardware gpumemsplit 76
```

danergo, 2025-04-21:
No, unfortunately.

I tried with below configs (one-by-one of course):

```
#-------GPU memory splits-------
gpu_mem_256=16
gpu_mem_512=16
gpu_mem_1024=16

```
```
#-------GPU memory splits-------
gpu_mem_256=128
gpu_mem_512=128
gpu_mem_1024=128
```

```
#-------GPU memory splits-------
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
```

MichaIng, 2025-04-22:
More than 76 (the default) should never be required. 16 MiB disables some GPU features automatically. But okay, if 128 MiB does not help, then this is not the issue, or not the only issue.

If it was 16 MiB originally (DietPi default), did the error message change, or is it still "Could not find a valid device"?

danergo, 2025-04-22:
Hi!

Changing GPU split memory didn't cause any changes in error messages. Full log is the same unfortunately.

MichaIng, 2025-04-28:
Does this change something?
```sh
/boot/dietpi/func/dietpi-set_hardware rpi-codec enable
```

danergo, 2025-04-29:
Hi!

Can't tell you right now.

But this is an unofficial answer from StackExchange (by AI):

> In summary, the Pi 3 B+ doesn’t support full hardware transcoding with h264_v4l2m2m on modern kernels and drivers. If you must keep using the Pi 3, try reverting to older distributions that still support MMAL/OMX or accept partial hardware support combined with software decode. Otherwise, upgrade to a Pi 4 for reliable hardware-accelerated H.264 transcoding through V4L2 M2M.

It seems pretty much true. What do you think?

MichaIng, 2025-04-30:
I am not convinced. There are many examples of V4L2 M2M being used on RPi 3 as well, e.g.: https://forums.raspberrypi.com/viewtopic.php?t=353958

And indeed there is supposed to be a special decode device, `/dev/video11` there, here is another site stating it be `/dev/video10`: [/dev/video11](https://faint.day/blog/raspberry-pi-3-hardware-accelerated-video-playback/)

At least for H264 decoding this should be possible.

danergo, 2025-04-30:
Hm. I must have some misconfiguration.

I dont even have any video device in /dev at all.

MichaIng, 2025-04-30:
Did you run above command already? https://github.com/MichaIng/DietPi/issues/7491#issuecomment-2836542780
Then run `modprobe bcm2835_codec` to enable it directly. This driver is definitely needed.

danergo, 2025-05-05:
Hi!

```
# /boot/dietpi/func/dietpi-set_hardware rpi-codec enable

 DietPi-Set_hardware
─────────────────────────────────────────────────────
 Mode: rpi-codec (enable)

[  OK  ] rpi-codec enable | Completed

#
# ls /dev/ | grep -i *vid*
#
# modprobe bcm2835_codec
# lsmod
Module                  Size  Used by
bcm2835_v4l2           45056  0
videobuf2_vmalloc      12288  1 bcm2835_v4l2
bcm2835_codec          45056  0
bcm2835_mmal_vchiq     36864  2 bcm2835_codec,bcm2835_v4l2
vc_sm_cma              28672  1 bcm2835_mmal_vchiq
v4l2_mem2mem           45056  1 bcm2835_codec
videobuf2_dma_contig    20480  1 bcm2835_codec
videobuf2_memops       12288  2 videobuf2_vmalloc,videobuf2_dma_contig
videobuf2_v4l2         32768  3 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videodev              303104  4 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       73728  7 bcm2835_codec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_v4l2,v4l2_mem2mem,videobuf2_memops
mc                     61440  5 videodev,bcm2835_codec,videobuf2_v4l2,videobuf2_common,v4l2_mem2mem
cmac                   12288  3
algif_hash             12288  1
aes_arm64              12288  3
aes_generic            32768  1 aes_arm64
algif_skcipher         12288  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   24576  2
hci_uart               49152  1
btbcm                  24576  1 hci_uart
bluetooth             610304  28 hci_uart,btbcm,bnep
ecdh_generic           16384  2 bluetooth
ecc                    36864  1 ecdh_generic
rfkill                 32768  3 bluetooth
libaes                 12288  3 aes_arm64,bluetooth,aes_generic
sc16is7xx_spi          12288  0
regmap_spi             12288  1 sc16is7xx_spi
sc16is7xx              24576  1 sc16is7xx_spi
binfmt_misc            16384  1
ax88179_178a           28672  0
snd_bcm2835            24576  0
snd_pcm               139264  1 snd_bcm2835
spi_bcm2835            16384  0
raspberrypi_hwmon      12288  0
snd_timer              36864  1 snd_pcm
snd                   110592  3 snd_bcm2835,snd_timer,snd_pcm
raspberrypi_gpiomem    12288  0
uio_pdrv_genirq        12288  0
uio                    20480  1 uio_pdrv_genirq
nfsd                  700416  5
dm_mod                143360  0
fuse                  163840  1
ip_tables              32768  0
x_tables               49152  1 ip_tables
ipv6                  569344  58
#
# ls /dev/ | grep -i *vid*
#

```

I tried it now. But `video` devices are simply not getting populated into `/dev` in spite all my efforts unfortunately.

MichaIng, 2025-05-06:
I just tried on RPi 5. Even with `bcm2835_codec` blacklisted (codec disabled), there is a `v4l2_mem2mem` kernel module loaded, and a lot of video devices are present:
```sh
root@DietPi:~# ls -l /dev/video*
crw-rw---- 1 root video 81, 16 May  6 14:58 /dev/video19
crw-rw---- 1 root video 81,  0 May  6 14:58 /dev/video20
crw-rw---- 1 root video 81,  1 May  6 14:58 /dev/video21
crw-rw---- 1 root video 81,  2 May  6 14:58 /dev/video22
crw-rw---- 1 root video 81,  3 May  6 14:58 /dev/video23
crw-rw---- 1 root video 81,  4 May  6 14:58 /dev/video24
crw-rw---- 1 root video 81,  5 May  6 14:58 /dev/video25
crw-rw---- 1 root video 81,  6 May  6 14:58 /dev/video26
crw-rw---- 1 root video 81,  7 May  6 14:58 /dev/video27
crw-rw---- 1 root video 81,  8 May  6 14:58 /dev/video28
crw-rw---- 1 root video 81,  9 May  6 14:58 /dev/video29
crw-rw---- 1 root video 81, 10 May  6 14:58 /dev/video30
crw-rw---- 1 root video 81, 11 May  6 14:58 /dev/video31
crw-rw---- 1 root video 81, 12 May  6 14:58 /dev/video32
crw-rw---- 1 root video 81, 13 May  6 14:58 /dev/video33
crw-rw---- 1 root video 81, 14 May  6 14:58 /dev/video34
crw-rw---- 1 root video 81, 15 May  6 14:58 /dev/video35
```
Oh I see now this is loaded in your case as well, pulled in by `bcm2835_v4l2`, in case of RPi 5 by `v4l2_mem2mem`. Your command to check for video devices is wrong. You'd need to quote `'*vid*'`, or use the command I used above to work with globbing correctly.

danergo, 2025-05-06:
```
# ls -all /dev
total 4
drwxr-xr-x 14 root root        3700 Apr 18 20:18 .
drwxr-xr-x 19 root root        4096 Nov 26 18:52 ..
crw-r--r--  1 root root     10, 235 Apr 18 20:18 autofs
drwxr-xr-x  2 root root         620 Apr 18 20:18 block
crw-------  1 root root     10, 234 Apr 18 20:18 btrfs-control
drwxr-xr-x  3 root root          60 Jan  1  1970 bus
drwxr-xr-x  2 root root        2660 Apr 18 20:18 char
crw--w----  1 root tty       5,   1 Apr 18 20:18 console
crw-------  1 root root     10, 124 Apr 18 20:18 cpu_dma_latency
crw-------  1 root root     10, 203 Apr 18 20:18 cuse
drwxr-xr-x  7 root root         140 Apr 18 20:18 disk
drwxr-xr-x  2 root root          80 Jan  1  1970 dma_heap
crw-rw----  1 root video    29,   0 Apr 18 20:18 fb0
lrwxrwxrwx  1 root root          13 Oct 10  2024 fd -> /proc/self/fd
crw-rw-rw-  1 root root      1,   7 Apr 18 20:18 full
crw-rw-rw-  1 root root     10, 229 Apr 18 20:18 fuse
crw-rw----  1 root gpio    254,   0 Apr 18 20:18 gpiochip0
crw-rw----  1 root gpio    254,   1 Apr 18 20:18 gpiochip1
crw-rw----  1 root gpio    254,   2 Apr 18 20:18 gpiochip2
crw-rw----  1 root gpio    254,   3 Apr 18 20:18 gpiochip3
crw-rw----  1 root gpio    254,   4 Apr 18 20:18 gpiochip4
crw-rw----  1 root gpio    239,   0 Apr 18 20:18 gpiomem
crw-------  1 root root     10, 183 Apr 18 20:18 hwrng
lrwxrwxrwx  1 root root          12 Oct 10  2024 initctl -> /run/initctl
drwxr-xr-x  2 root root          60 Jan  1  1970 input
crw-r--r--  1 root root      1,  11 Apr 18 20:18 kmsg
crw-rw----  1 root kvm      10, 232 Apr 18 20:18 kvm
lrwxrwxrwx  1 root root          28 Oct 10  2024 log -> /run/systemd/journal/dev-log
crw-rw----  1 root disk     10, 237 Apr 18 20:18 loop-control
brw-rw----  1 root disk      7,   0 Apr 18 20:18 loop0
brw-rw----  1 root disk      7,   1 Apr 18 20:18 loop1
brw-rw----  1 root disk      7,   2 Apr 18 20:18 loop2
brw-rw----  1 root disk      7,   3 Apr 18 20:18 loop3
brw-rw----  1 root disk      7,   4 Apr 18 20:18 loop4
brw-rw----  1 root disk      7,   5 Apr 18 20:18 loop5
brw-rw----  1 root disk      7,   6 Apr 18 20:18 loop6
brw-rw----  1 root disk      7,   7 Apr 18 20:18 loop7
drwxr-xr-x  2 root root          60 Apr 18 20:18 mapper
crw-r-----  1 root kmem      1,   1 Apr 18 20:18 mem
brw-rw----  1 root disk    179,   0 Apr 18 20:18 mmcblk0
brw-rw----  1 root disk    179,  32 Apr 18 20:18 mmcblk0boot0
brw-rw----  1 root disk    179,  64 Apr 18 20:18 mmcblk0boot1
brw-rw----  1 root disk    179,   1 Apr 18 20:18 mmcblk0p1
brw-rw----  1 root disk    179,   2 Apr 18 20:18 mmcblk0p2
crw-------  1 root root    243,   0 Apr 18 20:18 mmcblk0rpmb
drwxrwxrwt  2 root root          40 Jan  1  1970 mqueue
drwxr-xr-x  2 root root          60 Apr 18 20:18 net
crw-rw-rw-  1 root root      1,   3 Apr 18 20:18 null
crw-r-----  1 root kmem      1,   4 Apr 18 20:18 port
crw-------  1 root root    108,   0 Apr 18 20:18 ppp
crw-rw-rw-  1 root tty       5,   2 May  6 15:05 ptmx
drwxr-xr-x  2 root root           0 Oct 10  2024 pts
brw-rw----  1 root disk      1,   0 Apr 18 20:18 ram0
brw-rw----  1 root disk      1,   1 Apr 18 20:18 ram1
brw-rw----  1 root disk      1,  10 Apr 18 20:18 ram10
brw-rw----  1 root disk      1,  11 Apr 18 20:18 ram11
brw-rw----  1 root disk      1,  12 Apr 18 20:18 ram12
brw-rw----  1 root disk      1,  13 Apr 18 20:18 ram13
brw-rw----  1 root disk      1,  14 Apr 18 20:18 ram14
brw-rw----  1 root disk      1,  15 Apr 18 20:18 ram15
brw-rw----  1 root disk      1,   2 Apr 18 20:18 ram2
brw-rw----  1 root disk      1,   3 Apr 18 20:18 ram3
brw-rw----  1 root disk      1,   4 Apr 18 20:18 ram4
brw-rw----  1 root disk      1,   5 Apr 18 20:18 ram5
brw-rw----  1 root disk      1,   6 Apr 18 20:18 ram6
brw-rw----  1 root disk      1,   7 Apr 18 20:18 ram7
brw-rw----  1 root disk      1,   8 Apr 18 20:18 ram8
[... 90 lines cut ...]
crw-rw----  1 root tty       7,   4 Apr 18 20:18 vcs4
crw-rw----  1 root tty       7,   5 Apr 18 20:18 vcs5
crw-rw----  1 root tty       7,   6 Apr 18 20:18 vcs6
crw-rw----  1 root tty       7, 128 Apr 18 20:18 vcsa
crw-rw----  1 root tty       7, 129 Apr 18 20:18 vcsa1
crw-rw----  1 root tty       7, 130 Apr 18 20:18 vcsa2
crw-rw----  1 root tty       7, 131 Apr 18 20:18 vcsa3
crw-rw----  1 root tty       7, 132 Apr 18 20:18 vcsa4
crw-rw----  1 root tty       7, 133 Apr 18 20:18 vcsa5
crw-rw----  1 root tty       7, 134 Apr 18 20:18 vcsa6
crw-rw----  1 root tty       7,  64 Apr 18 20:18 vcsu
crw-rw----  1 root tty       7,  65 Apr 18 20:18 vcsu1
crw-rw----  1 root tty       7,  66 Apr 18 20:18 vcsu2
crw-rw----  1 root tty       7,  67 Apr 18 20:18 vcsu3
crw-rw----  1 root tty       7,  68 Apr 18 20:18 vcsu4
crw-rw----  1 root tty       7,  69 Apr 18 20:18 vcsu5
crw-rw----  1 root tty       7,  70 Apr 18 20:18 vcsu6
crw-------  1 root root     10, 127 Apr 18 20:18 vga_arbiter
crw-------  1 root root     10, 137 Apr 18 20:18 vhci
crw-rw----  1 root kvm      10, 238 Apr 18 20:18 vhost-net
crw-rw----  1 root kvm      10, 241 Apr 18 20:18 vhost-vsock
crw-------  1 root root     10, 130 Apr 18 20:18 watchdog
crw-------  1 root root    248,   0 Apr 18 20:18 watchdog0
crw-rw-rw-  1 root root      1,   5 Apr 18 20:18 zero
```

MichaIng, 2025-05-06:
Just tested on RPi Zero W, where all that was needed is enabling the codec and raising GPU memory to 32 MiB:
```console
[   17.645000] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   17.960656] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   18.168706] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   18.168810] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   18.262898] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   18.262977] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   18.312559] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   18.312637] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   18.333730] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[   18.333813] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[   18.382707] bcm2835-codec bcm2835-codec: Device registered as /dev/video31
[   18.382805] bcm2835-codec bcm2835-codec: Loaded V4L2 encode_image
```
Can you show your `dmesg` output?

danergo, 2025-05-10:
Hi, sure!

Currently, my GPU/System split is at 16Mb:

```
# dmesg | grep bcm2835
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_hdmi=0 8250.nr_uarts=1 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:6C:35:F3 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  root=PARTUUID=0140a7f0-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=tty1
[    0.088024] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.109577] bcm2835-dma 3f007000.dma-controller: DMA legacy API manager, dmachans=0x1
[    1.077872] pinctrl-bcm2835 3f200000.gpio: GPIO_OUT persistence: yes
[    1.089322] bcm2835-rng 3f104000.rng: hwrng registered
[    2.040565] bcm2835_vchiq 3f00b840.mailbox: there is not valid maps for state default
[    2.079967] bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer
[    2.080528] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[    2.156172] sdhost-bcm2835 3f202000.mmc: loaded - DMA enabled (>1)
[    5.510894] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[1429575.592061] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[1429575.600794] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[1429575.615580] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[1429575.618735] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-22)
[1429621.960354] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[1429621.963233] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-22)
```

I have configured now it to 32Mb and reboot:

```
[ SUB1 ] DietPi-Set_hardware > gpumemsplit (32)
[  OK  ] DietPi-Set_hardware | Setting in /boot/config.txt adjusted: gpu_mem_1024=32
[  OK  ] DietPi-Set_hardware | Setting in /boot/config.txt adjusted: gpu_mem_512=32
[  OK  ] DietPi-Set_hardware | Setting in /boot/config.txt adjusted: gpu_mem_256=32
[  OK  ] gpumemsplit 32 | Completed
```

Now my dmesg is this:

```
# dmesg | grep bcm2835
[    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_headphones=0 cgroup_disable=memory snd_bcm2835.enable_hdmi=0 8250.nr_uarts=1 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:6C:35:F3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=PARTUUID=0140a7f0-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=tty1
[    0.092115] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.113573] bcm2835-dma 3f007000.dma-controller: DMA legacy API manager, dmachans=0x1
[    1.082154] pinctrl-bcm2835 3f200000.gpio: GPIO_OUT persistence: yes
[    1.096195] bcm2835-rng 3f104000.rng: hwrng registered
[    2.047567] bcm2835_vchiq 3f00b840.mailbox: there is not valid maps for state default
[    2.087553] bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer
[    2.088122] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[    2.164212] sdhost-bcm2835 3f202000.mmc: loaded - DMA enabled (>1)
[    5.536241] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.

# ls /dev/v*
/dev/vc-mem  /dev/vcs	/dev/vcs3  /dev/vcs6   /dev/vcsa2  /dev/vcsa5  /dev/vcsu1  /dev/vcsu4  /dev/vga_arbiter  /dev/vhost-vsock
/dev/vchiq   /dev/vcs1	/dev/vcs4  /dev/vcsa   /dev/vcsa3  /dev/vcsa6  /dev/vcsu2  /dev/vcsu5  /dev/vhci
/dev/vcio    /dev/vcs2	/dev/vcs5  /dev/vcsa1  /dev/vcsa4  /dev/vcsu   /dev/vcsu3  /dev/vcsu6  /dev/vhost-net
```

danergo, 2025-05-29:
I believe codecs are enabled:

[image]

Other than that:

```
# ls /etc/modprobe.d/dietpi-disable_rpi_codec.conf
ls: cannot access '/etc/modprobe.d/dietpi-disable_rpi_codec.conf': No such file or directory
```

After doing `modprobe bcm2835_codec`, I have finally got back these video devices! Thank you!

```
# dmesg | grep bcm2835 | tail -n15
[    2.157644] sdhost-bcm2835 3f202000.mmc: loaded - DMA enabled (>1)
[    5.725831] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[52461.699028] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[52461.710393] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[52461.726400] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[52461.732678] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[52461.732743] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[52461.735731] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[52461.735807] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[52461.740761] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[52461.741129] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[52461.753002] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[52461.753074] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[52461.756462] bcm2835-codec bcm2835-codec: Device registered as /dev/video31
[52461.756542] bcm2835-codec bcm2835-codec: Loaded V4L2 encode_image

```

What could be the reason `bcm2835_codec` is not auto-loaded by default?

danergo, 2025-05-30:
Ah got it now! RPi camera is disabled and it disables also `bcm2835_codec` and `bcm2835_v4l2`:

```
# for i in /etc/modprobe.d/*; do echo "#### $i"; cat "$i"; done
#### /etc/modprobe.d/dietpi-disable_rpi_camera.conf
blacklist bcm2835_codec
blacklist bcm2835_v4l2
blacklist bcm2835_isp
#### /etc/modprobe.d/dietpi-disable_wifi.conf
blacklist cfg80211
```

Thanks a lot @MichaIng! One more addition: which is the safest option for me?

- remove dietpi-disable_rpi_camera.conf entirely, OR
- remove "blacklist bcm2835_codec" and "blacklist bcm2835_v4l2" from dietpi-disable_rpi_camera.conf OR
- enable rpi-camera OR
- any other suggestion?

Thank you!

MichaIng, 2025-05-30:
Fixed with: https://github.com/MichaIng/DietPi/commit/6a8d6bc

danergo, 2025-05-30:
Thank you! Shall we also remove `bcm2835_v4l2` from this file?

danergo, 2025-05-30:
I went back testing ffmpeg, and get this massive crash: https://github.com/raspberrypi/firmware/issues/1972

Do you maybe have any idea where shall I post this / how can I solve this?

MichaIng, 2025-05-30:
> Thank you! Shall we also remove `bcm2835_v4l2` from this file?

It was not needed on my RPi Zero W.

> Do you maybe have any idea where shall I post this / how can I solve this?

Try to set the GPU memory split to 64 MiB.

danergo, 2025-05-30:
Thank you very much! Now its running! :)

closed 2025-05-30 by danergo

MichaIng, 2025-05-30:
Great. I adjusted our script to set 64 MiB minimum, when enabling RPi codecs. Even if less is theoretically sufficient to have the functionality up, looks like actual transcoding still requires more. And the kernel error is not really self-explaining about it.