MichaIng/DietPi #7496: update to 9.12.1 fails because of missing python3-rpi-lgpio

kampfschlaefer, 2025-04-19:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
should be in the package with reference code c1db3178-0c66-4a8b-b2c7-30d11741881e ?

Its a RPi 3 Model B.

#### Additional Information (if applicable)
- Bug report ID : c1db3178-0c66-4a8b-b2c7-30d11741881e

#### Steps to reproduce
1. run sudo dietpi-update

#### Expected behaviour
- should successfully update?

#### Actual behaviour

snip from the fail
```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying incremental patches

[ INFO ] DietPi-Update | Current version : v9.11.2
[ INFO ] DietPi-Update | Latest version  : v9.12.1
[ INFO ] DietPi-Patch | Patching to DietPi v9.12...
[ INFO ] DietPi-Patch | Migrating from legacy RPi.GPIO to modern compatible rpi-lgpio Python 3 library ...
[ INFO ] DietPi-Patch | APT install python3-rpi-lgpio, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python3-rpi-lgpio
[FAILED] DietPi-Patch | APT install python3-rpi-lgpio
 - Command: apt-get -y --allow-change-held-packages install python3-rpi-lgpio
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: c1db3178-0c66-4a8b-b2c7-30d11741881e

Please file a related bug report at GitHub or our forum:
- https://github.com/MichaIng/DietPi/issues
- https://dietpi.com/forum/c/troubleshooting/10

The uploaded file will be removed automatically after 48 hours or when your issue has been solved. Additionally you can remove it by running:
- dietpi-bugreport -1

Press any key to continue...
```
#### Extra details
- I had python3-rpi-gpio (i think) installed, removed this after the first failed run. still no progress...

MichaIng, 2025-04-19:
This is a Bullseye system, right?

In other context I recognised that this package, a shim for lgpio providing legacy RPi.GPIO API, is available from Bookworm only.

When running automated install tests but forgetting to toggle RPi emulation for RPi-only software ... I'll do a patch later.

As a quick workaround, find `aSOFTWARE_INSTALL_STATE[69]=2' in `/boot/dietpi/.installed` and change it to `=0`.

kampfschlaefer, 2025-04-19:
> This is a Bullseye system, right?

Yes, Bullseye.

> As a quick workaround, find `aSOFTWARE_INSTALL_STATE[69]=2' in `/boot/dietpi/.installed`and change it to`=0`.

This helped and let the update run through. What did I disable with this change?

MichaIng, 2025-04-21:
The change flags RPi.GPIO as not installed for `dietpi-software`. As/if you uninstalled the `python3-rpi.gpio` package, that it true anyway.

devifast, 2025-05-19:
I'm trying to install Node-Red but I can't install python. I've done this installation dozens of times and I've never had a problem before.
``#### Details:
- Date           | 2025-05-19 11:45:23
- Program name   | DietPi-Software
- Command        | `apt-get -y --allow-change-held-packages install python3-rpi-lgpio`
- Exit code      | 100
- DietPi version | v9.12.1 (MichaIng/master)
- Distro version | bullseye (ID=6,RASPBIAN=1)
- Kernel version | `Linux mario32 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`
- Architecture   | `armhf`
- Hardware model | RPi Zero 2 W (armv7l) (ID=3)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package python3-rpi-lgpio
```

MichaIng, 2025-05-19:
As said, fixed for upcoming DietPi version, not released yet. From the error handler menu, please select "Change command", and replace the whole line with "true", to finish the update.