MichaIng/DietPi #7498: Please include Technitium DNS on repository and script

User20230904, 2025-04-21:
### Creating a feature request

Hello,

I'm here to ask you to gently add https://technitium.com/dns/ to the repository.

This tool is FAR NICER than Adguard and unbound and it works differently, you can simply make a copy of root zone and use everything local it's FANTASTIC.

It's also very lightweight and i think it would be a shame to not see it listed as software to install on dietpi launcher for the next update.

They deserve a chance, please give it. Thanks.

MichaIng, 2025-04-21:
Interesting as it can provide downstream DoT/DoH/DoQ IIUC.

User20230904, 2025-04-21:
YEah the possibilities are many, it's a very nice project, thanks for looking into it :)

andiohn, 2025-04-24:
Super cool!

lutfor-diu, 2025-07-04:
when add?

MichaIng, 2025-07-05:
No ETA for such, unless someone else does a start.