MichaIng/DietPi #7500: Uninstall script Fish gives error - after that no SSH-login

echtfrank, 2025-04-22:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `NO ACCESS TO DEVICE: Dietpi 9.12`
- Distro version | `NO ACCESS TO DEVICE`
- Kernel version | `NO ACCESS TO DEVICE`
- SBC model | `NO ACCESS TO DEVICE: Odroid C4 (ODROID HC4)'
- Power supply used | (5V 3A ORIGINAL ODROID)
- SD card used | SanDisk ultra 32GB

#### Additional Information (if applicable)
- Software title | Fish
- Was the software title installed freshly or updated/migrated? Installed Freshly
- Can this issue be replicated on a fresh installation of DietPi? UNKNOWN

- Bug report ID | `NO ACCESS TO DEVICE`

#### Steps to reproduce

first try:
1. log into device as user dietpi, using Windows Terminal
2. uninstall Fish by giving command `sudo dietpi-software uninstall 4` (error)
 
second try
3. manually entering bash using command `bash`
4. uninstall Fish by giving command `sudo dietpi-software uninstall 4` 

#### Expected behaviour

first try:
- I am correctly logged in into the device and I see the Fish-shell
- Fish is removed from my system 

second try:
- Logged into bash (within Fish)
- Fish is removed from my system

#### Actual behaviour

first try: 
- I am correctly logged in into the device and I see the Fish-shell
- Fish-shell does not understand the command and gives an error, probably it cannot handle the SUDO-command???

second try:
- uninstall-script for removing Fish starts... but stops with an error that a RM-command cannot remove file, because it is actually a fiolder; sorry I can't remember the exact folder name; it was something like /etc/.../<name.d>

After this, I cannot access the device via SSH anymore: it gives an error `permission denied` if I log in using user dietpi as well as user root... (and I have not changed my password)

#### Extra details

- After update DietPi to v9.12 last week, I wanted to try Fish and I installed Fish using bash-command 'sudo dietpi-software install 4'; this all went well. After minimal use of the shell (Fish in this case), I concluded that I wanted to uninstall Fish, because of problems using sudo (I am not sure).
- After afore mentioned steps I loggd out of the system, but now I cannot login anymore and fear that I have locked myself out... The applications are working fine also after disconnect and reconnect power: I can login to the web-UI's of Nextcloud, Jellyfin and Adguard, but I cannot access the device using any shell (Windows Terminal on Windows, Termius on Android). I have no time today to find and connect a monitor and keyboard directly to the device; may be tonight I have some time.
-
- Any help would be greatly appreciated. Is there something I can do to alter on the SD-card wich the system is running from? I have a laptop running MX Linux so I can put the SD-card in that laptop and investigate.

MichaIng, 2025-04-22:
Thanks for reporting. Indeed we need to revert login shells from fish to bash on uninstall. Fixed with: https://github.com/MichaIng/DietPi/commit/bbd4725

MichaIng, 2025-04-22:
The other issue about the `sudo` command: In bash, `sudo` needs to be an alias itself to be able to use it to call aliases. Hence we do this weird `alias sudo='sudo '`. Probably in fish this is different. Can you try this:
```sh
unalias sudo
sudo dietpi-launcher
```

echtfrank, 2025-04-22:
Many thanks back to you @MichaIng and the whole DietPi-team.
I will try your `unalias`-suggestion when I have managed access to the system again. Then I'll let you know.

MichaIng, 2025-04-22:
SSH clients can send a custom remote command, e.g. in PuTTY here:

[image]

echtfrank, 2025-04-23:
I have tried PuTTY and WinSCP to remotely login to the device, using a remote command. WinSCP has a similar settings screen like PuTTY. I have also tried SSH -t on the Windows-commandline (Powershell). I also tried to connect to open a remote Powershell-session (I have actually installed Powershell on my Odroid HC4...). Unfortunately al these tries result in "connection refused". This gave me the idea that for some reason my password was somehow altered.

Finally I connected the device to keyboard and tv and booted it. I have to login and this also does not work... Very shortly I can see this message: 

[image]

The bad thing is that I still can not login, but the good thing is that I can now see that logging in using definitely wrong passwords does not show the above message. My conclusion is that my password has not been changed (phew!).

It becomes clear that Fish is partly removed (namely from /usr/bin/fish), but I suspect there is some configuration setting that defines a default shell. Can I alter this manually? I can power off the device and put the SD-card in my Linux laptop?

MichaIng, 2025-04-23:
An alternative would be to start a `fish` shell from the original `bash` shell via `~/.bashrc` or even globally via `/etc/bashrc.d/` script. However, the argument that many scripts are not compatible with `fish` is nonsense: they are executed, not sourced, hence `fish` invokes `bash` or `dash` (depending on shebang or file extension) to execute those. The really only problem here is that the default shell of a user must exist to permit a login, it seems.

Do you have another Linux system you can attack this system drive to? You could create a symlink from `/usr/bin/fish` to `/bin/bash` there, to work around the missing executable.

echtfrank, 2025-04-23:
Yes, I have. I can put the SD-card ( the system drive) in a linux laptop. Do I understand correctly that a symlink is 'something' that is created on the drive - a feature of the filesystem ?. I will try that. Is this correct command?
`sudo ln -s /usr/bin/fish /bin/bash`

Otherwise I was thinking about obtaining the correct files within /usr/bin/fish  by installing a fresh dietpi-instance from scrathch on another SD-card and copy that back on my current SD. Is that worth a try when the symlink does not help? Or do I miss way to many other files and settings?

MichaIng, 2025-04-23:
When mounting the drive to `/mnt/odroidc4`, this is the command to create the correct symlink:
```sh
sudo -s /bin/bash /mnt/odroidc4/usr/bin/fish
```
The first argument is where the symlink shall be pointing to, and you want it to call bash. The second argument is where the symlink shall be created, and you want it at `/usr/bin/fish` in the Odroid's rootfs.

echtfrank, 2025-04-23:
Great, I can connect to a shell again. Thanks.

compdealer, 2025-04-28:
Hit the same error after fish uninstall recently, ended with clean dietpi install. 
[From fish documentation](https://fishshell.com/docs/current/faq.html#faq-uninstalling):

Uninstalling fish
If you want to uninstall fish, first make sure fish is not set as your shell. Run chsh -s /bin/bash if you are not sure.

closed 2025-05-25 by MichaIng

MichaIng, 2025-08-22:
While I could not find a good way to execute aliases via `sudo` in `fish`, only very complex constructs which are all not failsafe/robust, I added now automatic `sudo` invocation to our scripts, making it unnecessary to call them with sudo explicitly: #7686

Moving our scripts into proper `$PATH` locations is another step, but a larger change which I have no time for in foreseeable future. But above should be convenient anyway, _and for bash shells it also works for our `G_*` functions, like `G_AGI` for unattended error-handled APT installs etc._