MichaIng/DietPi #7501: Orange Pi 5 max support in Linux kernel 6.12

enx1998, 2025-04-22:
I would like to know if the Orange Pi 5 max is suported in Linux kernel 6.12 actually present in the repos.  
I need to run the orange pi as a server and I need the openzfs module which in the latest version (2.3) needs a kernel version higher than the rockchivp vendor one.

Thanks, ENx

MichaIng, 2025-04-22:
Linux 6.14 has support for Orange Pi 5 Max: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3588-orangepi-5-max.dts?h=linux-6.14.y

Armbian added support for "current" kernel as well: https://github.com/armbian/build/blob/main/patch/kernel/archive/rockchip64-6.12/dt/rk3588-orangepi-5-max.dts

Jep that can be done. I just need to assure the WiFi driver is added to the "current" build as well.

enx1998, 2025-05-06:
Hi @MichaIng ,
thanks for the reply. If I understand correctly, I have to wait to the 9.13 release of DietPi to be able to install the armbian current kernel, right? Also if I don't need the WiFi?

Thanks, ENx

MichaIng, 2025-05-06:
We can create an `edge` kernel image based on Linux 6.14 already now. But I am currently working on something else, so give me some time.

CertainBot, 2025-05-18:
Sorry to hijack the thread. Just a quick question, would there be support for the OrangePi 5 Ultra. I tried the Max version on it and it works but still interested.

Thanks

MichaIng, 2025-05-20:
Yes, I already started to implement Orange Pi 5 Ultra, CM5 and 4A support. But images will be generated after next release.

MichaIng, 2025-06-02:
Some info: This will take a little longer. Problem is the mainline Linux does not have the bcmdhd WiFi driver, hence onboard WiFi does not work. Same problem with Orange Pi 5 Ultra I am currently implementing.

Armbian maintains sources to work with more recent kernel versions: https://github.com/armbian/bcmdhd-dkms/tree/main/src
But this is for DKMS, which I would like to avoid as implicit need for such a basic feature. I started to work in implementing another WiFi driver directly into the kernel, which has the very same issue: https://github.com/MichaIng/build/commit/9559f31
In this case, for Radxa ZERO 3, but that chip is used by other boards as well, like official VisionFive 2 WiFi module. But it is not that trivial to implement it properly, so it currently has lower priority to me.

But the edge kernel (Linux 6.15) is present in our APT repo for the Orange Pi 5 Max, so you can install it:
```sh
apt install linux-{dtb,image}-edge-rockchip64
```
Works well on Orange Pi 5 Ultra, have no tested on Max yet, but those boards are very similar. Onboard WiFi will be gone, as well as several other board features, like NPU, and support for certain GPIO-attached hardware, DSI displays and such.

CertainBot, 2025-06-05:
Thanks for the update @MichaIng! 

One problem, probably not related, I had while trying the Max version on the Ultra is that I was unable to burn the image properly on the nvme disc. I only got a single partition so it was not bootable. It is the same image I used on the SD card where I can see two partitions. That usualy works for me using dd with all other images eg. Raspberry Lite OS (on a Raspberry 5) or Orange Pi OS on OPi 5 Ultra but for some reason was not done correctly using DietPi.
Any clues on what am I doing wrong?

MichaIng, 2025-06-05:
The image is expected to have a single partition only and should work well like that. I mean, precisely it has 2 partitions at first, a 1 MiB config file FAT partition at the end, which is however removed on first boot.

Does the original SPI bootloader expect a dedicated boot partition? If so, boot from SD card and flash our bootloader to SPI via `dietpi-config` Advanced Options.

But do not flash the Max image. We have an Orange Pi 5 Ultra image now: https://dietpi.com/downloads/images/
It is not at the download page yet, since I added support just a few days ago. But works fine on my board.

enx1998, 2025-06-06:
Hi @MichaIng I'm using the edge kernel on orange pi 5 max, and it is working ok form my needs (headless server), but I need to install zfs module and the latest version (2.3.1) needs at most the 6.13 version of the kernel, so I cannot use the edge version of the kernel. I need to wait you to finish the work on 6.12 stable kernel?

Thanks, ENx

MichaIng, 2025-06-06:
For the Orange Pi 5 Max, the additional problem is that mainline Linux 6.12 does not natively support it. Armbian patches a device tree inside, but the one for the 5 Ultra (very similar board) works pretty badly, much worse than the upstream one. So not sure it will ever work well, as I won't invest time to fix that device tree.

Any issus with the vendor kernel and zfs?

MichaIng, 2025-06-06:
Ah @CertainBot Orange Pi Ultra support has been finished, and it is working quite well with vendor Linux 6.1. You can find the images here: https://dietpi.com/downloads/images/

Will be added to our website after next DietPi release.

CertainBot, 2025-06-06:
Thank you very much for the effort @MichaIng! Will try it ASAP.

MichaIng, 2025-09-19:
See here: https://github.com/MichaIng/DietPi/issues/7337#issuecomment-3282766590
But note that those have feature limits, mostly around VPU and NPU. The fully featured vendor kernel will stay at 6.1, until mainline Linux support is good enough to switch defaults.