MichaIng/DietPi #7502: Official touch display broken with v1.1 PCB Pi 2 on bookworm

SixFive7, 2025-04-24:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
Cant boot that version anymore so these fields may not be in the required format
- DietPi version | `New install from 2025-04-24` 
- Distro version | `DietPi_RPi2-ARMv7-Bookworm.img.xz`
- Kernel version | `Unknown`
- SBC model | `Raspberry Pi 2 PCB v1.1 (ARMv7)`
- Power supply used | `5V 2.5A`
- SD card used | `Samsung EVO 64GB`

#### Additional Information (if applicable)
- Software title | `Not relevant`
- Was the software title installed freshly or updated/migrated? | `Everything fresh. Tried twice.`
- Can this issue be replicated on a fresh installation of DietPi? | `Yes`
- Bug report ID | `Unavailable`

#### Steps to reproduce
1. Get a Raspberry Pi 2 (PCB v1.1).
2. Hook up the official [Raspberry Pi Touch Display (v1)](https://www.raspberrypi.com/products/raspberry-pi-touch-display/).
3. Boot.
4. Observe no display output.
5. Fiddle with /boot/config.txt until the end of time.
6. Get only as far as some corrupted vertical lines.
7. Do exactly the same with `DietPi_RPi-ARMv7-Bullseye.img.xz` instead of `DietPi_RPi2-ARMv7-Bookworm.img.xz`.
8. Observe how on the first boot (without any custom preconfig whatsoever) the display works right away.

#### Expected behaviour
- The display should work out of the box on bookworm like it did on bullseye.

#### Actual behaviour
- It doesn't. Messing with different overlays does not help. Something far beyond my understanding is broken in the overlay/firmware/driver system.

#### Extra details
Although it is an old Pi I'm [not the first](https://dietpi.com/forum/t/raspberry-pi-touchscreen-black-screen-after-update/23226) to run into this.

MichaIng, 2025-04-28:
Can you please try the following:
```sh
sudo /boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d
G_SUDO G_CONFIG_INJECT 'display_auto_detect=' 'display_auto_detect=1' /boot/config.txt
sudo reboot
```
You should then get output, and you can rotate the screen via `dietpi-display`. To rotate touch input accordingly, check out: https://www.raspberrypi.com/documentation/accessories/display.html#rotate-touch-input

SixFive7, 2025-04-28:
Thank you for your response. Once I regain access to the device in approximately two weeks, I will attempt to document your suggestion.

jangrewe, 2025-06-10:
@MichaIng I have the (nearly) same problem, and you commands didn't fix it for me.

But for me, i do get the kernel log until chromium-kiosk starts, then it just blanks the display until shutdown.

I've added this to `config.txt` to get at least the boot output:
```
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-7inch
```

jangrewe, 2025-06-12:
> Did you need to add dtoverlay=vc4-kms-dsi-7inch? Because display_auto_detect=1 is supposed to detect and select the right overlay automatically.
You're right, using `display_auto_detect=1` instead of `dtoverlay=vc4-kms-dsi-7inch` also works - but i do need to add `dtoverlay=vc4-kms-v3d` nonetheless, otherwise the display doesn't even enable the backlight.

Chrome does start, as i can connect via VNC (RealVNC running, with shared desktop) and see Chrome with the configured URL opened in kiosk mode.
Removing `--use-gl=egl` from `/etc/chromium.d/dietpi` doesn't seem to make any difference.
So after the console output, X does get started (and the backlight stays on), it's just a black screen...

jangrewe, 2025-06-12:
When running anything with `startx`, it starts whatever i ran, but the display still stays blanked.

I can launch xterm with `DISPLAY=:0` and it starts in the (already running, used by chrome) X server - but as with chrome, i can only see this via VNC.

So everything runs as expected, it's just that the X server doesn't output to the display.

MichaIng, 2025-06-12:
Enforcing display 1 does not change something does it?
```sh
startx /usr/bin/chromium -- :1
```
Based on the output, while it finds the HDMI connector, it seems to use the DSI display, also with the intended KMS driver and module. It also detects the touch input. It selects HDMI audio, but that does not matter for video output.

Not sure at this point. Please try to remove or comment some legacy, possibly conflicting `/boot/config.txt` entries: `start_x`, `framebuffer_width`/`framebuffer_height`, `lcd_rotate`
Also assure to have 64 MiB GPU memory at least. But it should be 76 MiB already (board default) after Chromium installation.

Does it work OOTB with Raspberry Pi OS? I somehow cannot find their desktop repository at GitHub anymore, but can check the image for X11 configs which might make a differnce.

jangrewe, 2025-06-30:
Unfortunately, none of that made any difference... i was using `bullseye` before and only updated because my SD card got corrupted (unrelated to regular usage), so i'll go back to `bullseye` now, to check if that works "again".

MichaIng, 2025-06-30:
If it is the same with Raspberry Pi OS Bookworm, maybe the RPi community can help at their forum. Though when using the image with preinstalled RPi desktop, it uses Wayland, not X11.

Reading through a thread about why an X11 config is needed on RPi: https://forums.raspberrypi.com/viewtopic.php?t=358853#p2181567
If you still have a Bookworm SD card or time to test, could you check which driver is used by the DSI display:
```sh
l /sys/class/drm/*/device/driver
```