MichaIng/DietPi #7506: DietPi-Update

BeamBam, 2025-04-25:
#### Details:
- Date           | Fri Apr 25 09:42:23 EEST 2025
- DietPi version | v8.25.3 (MichaIng/8)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | `Linux DietPi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `ping -4nc 1 -W 10 IP`
- Exit code      | 127
- Software title | DietPi-Update
#### Steps to reproduce:

1. Run dietpi-update
2. ...
#### Expected behaviour:
 DietPi-Update was complete successfully
- ...
#### Actual behaviour:
DietPi-Update not complete successfully, "ping: error while loading shared libraries: libnettle.so.6: cannot open shared object file: No such file or directory"
- ...
#### Extra details:

- ...
#### Additional logs:
```
ping: error while loading shared libraries: libnettle.so.6: cannot open shared object file: No such file or directory

#### Extra details
<!-- Please post any extra details that might help solve the issue -->
- ...

Joulinar, 2025-04-25:
This looks more like a fundamental problem if a simple `ping` does not work. You could try reinstalling the corresponding package.

```
apt install --reinstall libnettle6
ping 8.8.8.8

```

BeamBam, 2025-04-25:
root@DietPi:~# sudo apt install --reinstall libnettle6
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libnettle6

MichaIng, 2025-04-25:
On Bullseye it is `libnettle8`. So the `ping` seems to be an incompatible left over from an upgraded Buster instance. Can you check the package which provides it?
```sh
which ping
dpkg -S "$(which ping)"
```
The intended package is `iputils-ping`, which should have been upgraded as part of the Buster => Bookworm distro upgrade. But probably it is another ping executable or a package which does not exist on Bullseye.

BeamBam, 2025-04-28:
root@DietPi:~# which ping
/usr/bin/ping
root@DietPi:~# dpkg -S "$(which ping)"
dpkg-query: no path found matching pattern /usr/bin/ping
root@DietPi:~#

BeamBam, 2025-04-28:
After reinstall package iputils-ping, update was completed successfully.
Thank you, MichaIng.

closed 2025-04-28 by BeamBam

MichaIng, 2025-04-28:
Great. Weird that there was a `/usr/bin/ping` was present but without any associated package installed 😄.