MichaIng/DietPi #7507: Add option chsh to fish as default shell upon fish installation

ytqxyd, 2025-04-26:
`dietpi-software` has already a few pre and post installation scripts upon software installation. 

This is a feature request to add an option to allow user to choose whether they want to set fish as default shell upon installation. So user doesn't have to configure `chsh` manually post installation.

MichaIng, 2025-04-27:
This is done already on installation for the `root` and `dietpi` users. You mean that option for other users? In case `sudo` was used to call `dietpi-software`, at least we could apply it for `$SUDO_USER` as well, but otherwise the script cannot know which user it might be wanted for. So then a real user management script would be needed instead. I thought about such already, but uff ... way more ideas than time 😅.

ytqxyd, 2025-04-27:
> This is done already on installation for the root and dietpi users. 

I am not aware that it is already the case. I did a chsh manually on diet-pi and root. 

Instead of having user management script, maybe have a few options to allow user to choose which user to change default shell:
* No change
* root only
* current user + root
* all users 

If user need more granular control, they can do it themselves.

MichaIng, 2025-04-27:
This is the commands that run on fish installation:
```sh
G_EXEC chsh root -s /usr/bin/fish
getent passwd dietpi &> /dev/null && G_EXEC chsh dietpi -s /usr/bin/fish
```
Your suggestion is already difficult as of the question what the "current user" is, as `dietpi-software` is executed as root. Of course we could implement a more or less complex menu and logic, but as you say: more granular control needs to be done anyway manually, and for the two known users things are applied already.

The user management tool would not be solely for selecting a shell, but to add/remove users, change passwords, lock them, add SSH keys, lock SSH password login, manage groups/permissions etc etc. More short-term I want to replace the `dietpi` user with a prompt to optionally create a custom new login user. That might consequently come along with the script then, at least a first minimal implementation of it, to be extended with more features later on.

ytqxyd, 2025-04-28:
I see the problem. Definitely much harder than I imagined. 

In theory, we can easily grab the current user with `last | awk 'NR==1{print $1}'` (add check for login/logout time for better reliability). 

But it seems that wtmp log got wiped due to log rotation.

MichaIng, 2025-04-28:
> But it seems that wtmp log got wiped due to log rotation.

Right. And on Debian Trixie, `last` `lastb` and `lastlog` got removed from `util-linux` and moved into their own packages, sadly not well documented in the changelogs. As of their are about to be deprecated 🤔. There is https://packages.debian.org/trixie/lastlog2 but no idea whether it is even the same (reads from `/var/log/lastlog`).

... here is `last` at least: https://packages.debian.org/trixie/wtmpdb ... 2038 safe, hmm, that might be the reason. Imports traditional logs 🤔.

Both sound like the old logs are deprecated and both get replaced with own new databases and new tools to read them.

But apart of this: whether the last logged in user is the one you aim to enable fish for is a rough guess as well. Perfectly possible that other users log in and our in the meantime. I personally use `screen` and have a 24/7 session.

But I mean, it is possible to add a checkbox menu with all login/non-system users to select those. On uninstall, from next DietPi version on, we already loop through all existing users, and replace `fish` with `bash` in all cases, to be failsafe.

mk12, 2025-05-04:
> This is done already on installation for the root and dietpi users.

You might want to reconsider that, or ensure it gets reverted when uninstalled fish. I tried installing fish and then changed my mind and uninstalled it, and was unable to login. I had to remove the SD card and edit it from another machine to add " S" to the end of /boot/cmdline.txt. Then I was able to login as root and edit /etc/passwd.

MichaIng, 2025-05-05:
Already done with: https://github.com/MichaIng/DietPi/commit/bbd4725