MichaIng/DietPi #7509: Errore in update vpn server list "protonVPN"

black00019, 2025-04-27:
Hi
When i update the vpn server list i get an error:

[image]

same at terminal :
root@DietPi:~# curl -sSfLO https://api.protonvpn.ch/vpn/logicals
curl: (22) The requested URL returned error: 401
root@DietPi:~#

Joulinar, 2025-04-27:
I guess this could be related https://www.reddit.com/r/ProtonVPN/comments/1k3lrl5/great_the_httpsapiprotonvpnchvpnlogicals_api_has/

MichaIng, 2025-04-27:
Yeah or this deprecation notice: https://github.com/Rafficer/linux-cli-community?tab=readme-ov-file#deprecation-notice

black00019, 2025-04-27:
I don't have a token (??? whre is??)
But alternative i set manually ip in config files:
/etc/openvpn/protonvpn/serverlist.txt

black00019, 2025-04-27:
I logged in web site protonvpn
and when open the link api i recieved:
"Invalid Token"

Joulinar, 2025-04-28:
I guess they have disabled the API completely

MichaIng, 2025-04-28:
Yeah, or at least we'd first need to find out how to add a token to the request. At least, if this works, the same token can then be used for VPN connection, and no username/password required anymore.

dandycyclone, 2025-09-20:
I can confirm, that https://api.protonvpn.ch/vpn/logicals API still works. However Proton does not suggest authentication tokens (aka Bearer) for personal accounts (I did not find) - only for business and only in the scope of identities management: https://protonvpn.com/support/scim-provisioning-api

For personal accounts you have to use your username+password + optional 2FA for login. After successful login you will get auth [cookies](https://developer.mozilla.org/en-US/docs/Web/HTTP/Guides/Cookies#what_cookies_are_used_for), that could be reused for subsequent API calls with any client of your choice (WEB browser, curl, wget, postman, etc). When you log out, cookies are invalidated immediately, the same happens after timeout.

For that reason, I can suggest three options here:
 - reverse engineering of the official [ProtonVPN client for Linux](https://github.com/ProtonVPN/proton-vpn-gtk-app) - I use it at my home Linux desktop every day and it works good. But it also uses the same username+password+2FA authentication scheme. In my opinion, this option does not worth the efforts - we will be always behind Proton continuously trying to catch up with never-ending changes;
 - rewrite _dietpi-vpn_ script to pass previously mentioned user cookies inside _curl https://api.protonvpn.ch/vpn/logicals_ command - here is an [example](https://github.com/yt-dlp/yt-dlp/wiki/FAQ#how-do-i-pass-cookies-to-yt-dlp), how another project solves similar kind of problem. Sounds as an option, but very sophisticated;
 - rewrite _dietpi-vpn_ script to follow [standard and supported way of using ProtonVPN](https://protonvpn.com/support/linux-openvpn) on headless machines. Yes, it is not scalable and not very elegant, however it is a working setup representing more or less stable interface. In my opinion, this option is the best one - easy to implement and fits very well into existing script. It also should preserve already available functionality the script provides.

MichaIng, 2026-08-28:
So ProtonVPN has no OpenVPN client config available anywhere anymore, right? Then we can just remove it from `dietpi-vpn` now.

Dynamic5912, 2026-08-28:
> So ProtonVPN has no OpenVPN client config available anywhere anymore, right? Then we can just remove it from `dietpi-vpn` now.

You can still get OVPN configurations from their website once signed in.

I believe they just deprecated the API that was being used previously.

MichaIng, 2026-08-28:
Is there a way to do that via `curl`, like passing an authentication token as header or so?

Btw interesting that it works with vanilla WireGuard. This means you'll have a static VPN IP, the big privacy issue of WireGuard when using it with public VPN providers, which is why they all use their own apps with IP rotation on top of the plain WireGuard protocol.

Dynamic5912, 2026-08-28:
I used the Wireguard app (per my posts above) but I kept getting IP/DNS leaks (my UniFi was alerting me to some "interesting" behaviour being blocked...🏴‍☠️).

Now I've switched to using an old macOS device with the native ProtonVPN GUI App and no more leaks.

Dynamic5912, 2026-08-28:
Also @MichaIng this looks interesting..

https://protonvpn.com/blog/protun-technical

MichaIng, 2026-08-28:
Are you using this WireGuard instance to forward traffic from other LAN/VPN clients through ProtonVPN? If so, it is essential that IPv6 is enabled (and forwarded) on that WireGuard host. Otherwise, any IPv6 request from connected clients will bypass the VPN.

> https://protonvpn.com/blog/protun-technical

This is what I mean. But interesting that they do not mention the static IP issue at all. But it is probably part of the project. Or dynamic IPs is in their apps wrapped around WireGuard already, but just not necessary to connect to their servers.

Can someone check whether the VPN-internal client IP remains the same throughout reconnects when using their app instead of a vanilla WireGuard client?

_____________

Regarding the original issue: Since connecting with plain WireGuard is usually not possible with other VPN providers, as it does imply the mentioned static IP issues, `dietpi-vpn` cannot implement WireGuard support. OpenVPN might be still fine or even needed in some cases, which is the reason we keep it. But for this to work, we need to be able to download OVPN configs via `curl`. I do not have a ProtonVPN account. If someone can tell me whether/how to download a config via `curl`, using whichever authentication method (basic, token header, ...), we can fix this in `dietpi-vpn`, otherwise we need to remove ProtonVPN support from the script.

Dynamic5912, 2026-08-28:
I don't use IPv6, only v4

MichaIng, 2026-08-28:
If you use this VPN connection from the DietPi host itself only, this is fine, you just need to remove the IPv6 masks from `AllowedIPs` in the config. `AllowedIPs = 0.0.0.0/0` should assure that everything is passed through the tunnel, hence no leakage should be seen.

But if you open this VPN connection to other remote clients, this would be a reason for IPv6 leakage, whenever those clients are connected to a network with IPv6 available.

MichaIng, 2026-10-26:
I found something interesting: https://github.com/tn3w/ProtonVPN-IPs/blob/master/protonvpn_logicals.json
Implemented, along with showing server name and city in the menu as well. Let me know if there is other info valuable to see in the server selection menu.

Dynamic5912, 2026-10-26:
> I found something interesting: https://github.com/tn3w/ProtonVPN-IPs/blob/master/protonvpn_logicals.json
> Implemented, along with showing server name and city in the menu as well. Let me know if there is other info valuable to see in the server selection menu.
> 
> But we'll see how long this works: https://github.com/ProtonVPN/python-proton-core/issues/11#issuecomment-5527036637

Looks promising.

I am not sure if it can show the different "protocols" the server supports - ie p2p, streaming, tor etc.

MichaIng, 2026-10-26:
Not directly, as this info is not in the logicals list. But the "Tier" should allow to derive that info.
- Tier 0 seems to be the "secure core servers" from here: https://protonvpn.com/vpn-servers#secure-core
  No extra services. These servers all have a name starting with "FREE", which the now deleted user was asking about here: #6030
  Most likely these are the ones usable with a free plan: https://protonvpn.com/free-vpn
- Tier 2 servers seem to be the "plus" servers from here: https://protonvpn.com/vpn-servers#plus
  Almost all with higher bandwidth, streaming support, and P2P. But there are exceptions, and I see no way to get that info.

Dynamic5912, 2026-10-26:
OK no worries.

The p2p servers are needed for certain.. erm.. activities so would be good to know which ones they are - but if it can't be done then it can't be done :)

MichaIng, 2026-10-26:
It is always possible to check back with the list linked above. The features are per city, and the city is shown in the list. So selecting e.g. any server from Germany Berlin will have P2P, and any from Frankfurt will have TOR support, too.

Dynamic5912, 2026-10-26:
Not all servers have tor - in Frankfurt (using your example) only select ones do - likewise for IPv6.

[image]

[image]

Admittedly most seem to support p2p as standard.

MichaIng, 2026-10-26:
Oh how annoying, so on to the public list the symbols just mean "at last one server in this city supports it".

closed 2026-10-27 by MichaIng