MichaIng/DietPi #7510: Jellyfin DietPi Software Installation utterly Obsolete & Broken

theAkito, 2025-04-28:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=12
G_DIETPI_VERSION_RC=1
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bookworm
```
- Kernel version | `uname -a`
```
Linux DietPi 6.1.0-34-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.135-1 (2025-04-25) x86_64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Generic Mini-PC x86_64
- Power supply used | (EG: 5V 1A RAVpower)
Generic Power Supply
- SD card used | (EG: SanDisk ultra)
Generic NVMe

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Jellyfin
- Was the software title installed freshly or updated/migrated?
Yes, installed freshly.
- Can this issue be replicated on a fresh installation of DietPi?
Yes.

#### Steps to reproduce

1. Install Jellyfin the official way by using DietPi Software List.

#### Expected behaviour

Jellyfin should be installed properly.

#### Actual behaviour

Jellyfin is installed a very wrong/obsolete way.

For example, it does not even install the packages, which are shown in the [documentation](https://dietpi.com/docs/software/media/#jellyfin-update-to-latest-version).

Then, it needs ancient fixes, like [this one](https://github.com/jellyfin/jellyfin/issues/3638#issuecomment-662457950).

#### Extra details

Most basic DietPi installation. Only installed software from official DietPi Software List, like Jellyfin, Docker and OpenSSH.

MichaIng, 2025-04-28:
> Jellyfin is installed a very wrong/obsolete way.

Can you explain in how for it is wrong/obsolete? We install the official `jellyfin` meta package.

> For example, it does not even install the packages, which are shown in the [documentation](https://dietpi.com/docs/software/media/#jellyfin-update-to-latest-version).

The docs are outdated, the `jellyfin-ffmpeg5` package is not needed anymore, even problematic as in the meantime there was `jellyfin-ffmpeg6` and now `jellyfin-ffmpeg7`. Only the meta package needs to be installed/updated.

> Then, it needs ancient fixes, like https://github.com/jellyfin/jellyfin/issues/3638#issuecomment-662457950.

Hmm, this indicates that the web interface moved from `/usr/lib` to `/usr/share` at some point, but the package maintainer script did not apply the migration to the config file. Though some examples show the config file being correct, but Jellyfin still looking at the wrong place.

... I just tested it, and indeed there is a problem/bug:
```
Apr 28 13:37:31 VM-Bookworm jellyfin[2891]: [13:37:31] [INF] Arguments: ["/usr/lib/jellyfin/bin/jellyfin.dll", "--webdir=/usr/share/jellyfin/web", "--ffmpeg=/usr/lib/jellyfin-ffmpeg/ffmpeg"]
...
Apr 28 13:37:31 VM-Bookworm jellyfin[2891]: [13:37:31] [INF] Program data path: /mnt/dietpi_userdata/jellyfin
Apr 28 13:37:31 VM-Bookworm jellyfin[2891]: [13:37:31] [INF] Log directory path: /var/log/jellyfin
Apr 28 13:37:31 VM-Bookworm jellyfin[2891]: [13:37:31] [INF] Config directory path: /etc/jellyfin
Apr 28 13:37:31 VM-Bookworm jellyfin[2891]: [13:37:31] [INF] Cache path: /mnt/dietpi_userdata/jellyfin/cache
Apr 28 13:37:31 VM-Bookworm jellyfin[2891]: [13:37:31] [INF] Temp directory path: /tmp/jellyfin
Apr 28 13:37:31 VM-Bookworm jellyfin[2891]: [13:37:31] [INF] Web resources path: /usr/share/jellyfin/web
Apr 28 13:37:31 VM-Bookworm jellyfin[2891]: [13:37:31] [INF] Application directory: /usr/lib/jellyfin/bin/
...
Apr 28 13:37:36 VM-Bookworm jellyfin[2891]: [13:37:36] [WRN] The WebRootPath was not found: /mnt/dietpi_userdata/jellyfin/wwwroot. Static files may be unavailable.
```
So the web dir is applied correctly at first to `/usr/share/jellyfin/web`, but later the app looks for `/mnt/dietpi_userdata/jellyfin/wwwroot` (`wwwroot` within the data dir). Yeah this matches other reports. Let's see what the guys come up with: https://github.com/jellyfin/jellyfin-packaging/issues/36#issuecomment-2834998268

theAkito, 2025-04-28:
> The docs are outdated, the `jellyfin-ffmpeg5` package is not needed anymore, even problematic as in the meantime there was `jellyfin-ffmpeg6` and now `jellyfin-ffmpeg7`. Only the meta package needs to be installed/updated.

When `ffmpeg` did not work, I ran a  `apt install -y ffmpeg jellyfin-ffmpeg5` at once, so yeah, maybe only `ffmpeg` is enough, but didn't wanna try twice, so I just installed both right away. Either way, it complains about `ffmpeg` not being available and crashing on first run, before installing either `ffmpeg`. `apt install` also definitely installed new packages and did not say, that those were already installed.

> Hmm, this indicates that the web interface moved from `/usr/lib` to `/usr/share` at some point, but the package maintainer script did not apply the migration to the config file. Though some examples show the config file being correct, but Jellyfin still looking at the wrong place.

This is why I said it's "wrong/obselete", because the workaround I mentioned from the Jellyfin issue is from 2020, which was one year after the Jellyfin package was added to DietPi. Hence, I assumed, the DietPi Jellyfin Installation was not improved or adjusted since 2019.

> So the web dir is applied correctly at first to `/usr/share/jellyfin/web`, but later the app looks for `/mnt/dietpi_userdata/jellyfin/wwwroot` (`wwwroot` within the data dir). Yeah this matches other reports. Let's see what the guys come up with: [jellyfin/jellyfin-packaging#36 (comment)](https://github.com/jellyfin/jellyfin-packaging/issues/36#issuecomment-2834998268)

Thanks for showing me this. I thought, this was a DietPi-specific issue.

MichaIng, 2025-04-28:
The `ffmpeg` package is not required at all. I guess Jellyfin is able to work with it as well, but the meta package provides its own `jellyfin-ffmpeg7` which is newer and assured to work. So better use this, I mean really only the meta package. So it can move to `jellyfin-ffmpeg8` once FFmpeg 8 is released and they move over.

To fix this part for you:
```sh
apt install jellyfin
```
Really only ever this is needed, same as we do on (re)install and as the docs state now (I adjusted them).

> This is why I said it's "wrong/obselete", because the workaround I mentioned from the Jellyfin issue is from 2020

There was a whole log of issues when they moved the web dir from `/usr/lib` to `/usr/share`:
- Some people had mismatching Jellyfin package, when they did not use the meta package but installed server and web separately. For them it worked to just upgrade all packages, or move to the meta package. This was the first issue you linked.
- Some had outdated `/etc/default/jellyfin` entries still pointing to `/usr/lib`. There was probably no config migration at first, maybe not even now, it was the issue at first for the new report at the packaging repo. However, does not apply to a fresh install, of course.
- And now there is this mismatch between the applied web dir and what the server is looking for. I explained it (what I think is going on) in my new comment to that issue, after investigating it a bit. So that is a new bug.

theAkito, 2025-04-28:
> The `ffmpeg` package is not required at all. I guess Jellyfin is able to work with it as well, when configured correctly, but the meta package provides its own `jellyfin-ffmpeg7` which is newer FFmpeg (on Debian Bookworm) and assured to work. So better use this, I mean really only the meta package. So it can move to `jellyfin-ffmpeg8` once FFmpeg 8 is released and they move over.
> 
> To fix this part for you:
> 
> apt install jellyfin
> 
> Really only ever this is needed, same as we do on (re)install and as the docs state now (I adjusted them).
> 
> > This is why I said it's "wrong/obselete", because the workaround I mentioned from the Jellyfin issue is from 2020
> 
> There was a whole log of issues when they moved the web dir from `/usr/lib` to `/usr/share`:
> 
>     * Some people had mismatching Jellyfin package, when they did not use the meta package but installed server and web separately. For them it worked to just upgrade all packages, or move to the meta package. This was the first issue you linked.
> 
>     * Some had outdated `/etc/default/jellyfin` entries still pointing to `/usr/lib`. There was probably no config migration at first, maybe not even now, it was the issue at first for the new report at the packaging repo. However, does not apply to a fresh install, of course.
> 
>     * And now there is this mismatch between the applied web dir and what the server is looking for. I explained it (what I think is going on) in my new comment to that issue, after investigating it a bit. So that is a new bug.

After I installed Jellyfin via DietPi software installation, which apparently only installs the `jellyfin` meta-package, I created the symlink for the workaround and after that, I tried to run Jellyfin, which complained about FFMPEG missing, which it crashed for. So, something's not right or does the DietPi installation not install the `jellyfin` meta-package? I mean, Jellyfin runs, after I manually install FFMPEG.

MichaIng, 2025-04-28:
Please do this again:
```sh
sudo apt install jellyfin # this should install jellyfin-ffmpeg7
sudo ln -s /usr/share/jellyfin /mnt/dietpi_userdata/jellyfin/wwwroot
sudo systemctl restart jellyfin
```
You might need to CTRL+F5 the web UI, and point it to `<IP>:8097` again. It should then switch to `/web` subdir, which now does exist as expected.

MichaIng, 2025-04-28:
... ah wait, nah this does not work that great either. I need to access `http://<IP>:8097/web/index.html#/wizardstart.html` in my case to actually access the setup wizard. From that point on, it however works for me. Just tested with a fresh install on DietPi.

theAkito, 2025-04-28:
This is, what it says in my configuration. Copied it right out of there.

```xml
<EncoderAppPathDisplay>/usr/lib/jellyfin-ffmpeg/ffmpeg</EncoderAppPathDisplay>
```

Seems, like this corresponds to what you showed.

MichaIng, 2025-04-28:
Hmm, then it seems the (now wrong) path is stored somewhere else in a database or what. This is the only place I could find it in config files. Just to be sure, that one exists, right?
```sh
/usr/lib/jellyfin-ffmpeg/ffmpeg -version
```
Ah wait, there is another place:
```sh
grep 'JELLYFIN_FFMPEG_OPT=' /etc/default/jellyfin

You did restart the server in between?
```sh
systemctl restart jellyfin
```

MichaIng, 2025-04-28:
You cannot start `jellyfin` like this with this command. The service started it correctly already. Check the service logs:
```sh
journalctl -u jellyfin
```

theAkito, 2025-04-28:
Now, when applying all the fixes from this thread, it seems to work. Kind of forgot how complicated everything is, when not using containers.

So, what needs to be updated in the DietPi Installation for Jellyfin? It needs to install the `jellyfin` meta-package, then apply the aforementioned symbolic links?

MichaIng, 2025-04-28:
Whether container or systemd unit or anything else, it always needs to be started/invoked the correct way. In case of Jellyfin, it autostarts via systemd unit, same as a Docker container does, so nothing else needs to be done.

The `dietpi-software` install option is 100% correct, nothing needs to be adjusted. The bug you run into is because of an internal Jellyfin bug, which needs to be resolved their end. What we did here was a workaround which works ... sort of, but not well enough to apply it to our code.

The FFmpeg issue was only because you installed the `jellyfin-ffmpeg5` package in between (as of outdated info in our docs, fixed already) and then the `ffmpeg` package. Installing the `jellyfin` meta package again fixed things, probably just the service restart was outstanding.

Joulinar, 2025-04-28:
yap I remember the day where Docker software themselves introduced a bug on their engine causing containers to fail. Some people lost their entire home lab just because of a single software bug. Docker is a great thing, but it also has one or two pitfalls.