MichaIng/DietPi #7511: Failed update for DietPi from 8.17.2 to 9.12.1 on RPi Zero W (armv6l)

pablojr, 2025-04-29:
#### Details:                                                                                                  
- Date           | Tue Apr 29 09:38:19 -03 2025                                                                
- DietPi version | v8.17.2 (MichaIng/master)                                                                   
- Image creator  |                                                                                             
- Pre-image      |                                                                                             
- Hardware       | RPi Zero W (armv6l) (ID=1)                                                                  
- Kernel version | `Linux piaware 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`                 
- Distro         | bullseye (ID=6,RASPBIAN=1)                                                                  
- Command        | `curl -sSfLO https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches`  
- Exit code      | 22                                                                                          
- Software title | DietPi-Update                                                                               
#### Steps to reproduce:                                                                                       
                                                                    
1. ...                                                                                                         
2. ...                                                                                                         
#### Expected behaviour:                                                                                       
                                                                                   
- ...                                                                                                          
#### Actual behaviour:                                                                                         
                                                                                    
- ...                                                                                                          
#### Extra details:                                                                                            
                                         
- ...                                                                                                          
#### Additional logs:                                                                                          
```                                                                                                            
curl: (22) The requested URL returned error: 429                                                               
```

Joulinar, 2025-04-29:
for me the URL is reachable. It's the global GitHub repository and should be available. HTTP error 429 means that there were too many requests to the GitHub server. Possibly a protection mechanism? Can you try this manually?

```
cd /tmp
curl -sSfLO https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/pre-patches
```

is this working?

pablojr, 2025-04-29:
Hi. I've just tried command `dietpi-update` again and everything went Ok this time.
Updated to 9.12.1.
Thanks.

closed 2025-04-29 by pablojr