MichaIng/DietPi #7513: Default libs in /lib?

danergo, 2025-04-30:
Folks, I have no idea where to look, but maybe you have some clue:

I wish to know which exact libs are installed by default on DietPi installation.

For example, do we have:

- libffi
- libglib
- libexpat
- libformw
- libgio-2
- libmenuw
- libncursesw
- libreadline

Thank you!

danergo, 2025-04-30:
Thanks! I accidentally pushed some libs to that folder and wish to know which libs are installed by default to remove my extras (without reinstalling the whole OS).

:)

Joulinar, 2025-04-30:
As already said we install packages from Debian global repository. On a fresh system, it will always be latest version available as we update all packages during initial installation. 

We don't know what the status was on your system. Not even the Debian version, as you ignored to fill the troubleshooting template.

MichaIng, 2025-04-30:
Generally, you can check whether a file belongs to an APT package this way:
```sh
dpkg -S foobar.so
```
Using a loop with `find` could be used to find all those which do not belong to any package.

There is no real list for installed libraries on DietPi. Our build scripts do not explicitly install any library, but a small set of fixed packages, some depending on the underlying Debian version, and some (kernel/firmware/bootloader and such) depending on the hardware, hence the final package list differs. These packages then pull a bunch of libraries as dependencies: https://github.com/MichaIng/DietPi/blob/master/.build/images/dietpi-installer#L908

If you did not install anything else (no other APT package) from the original image, you could also just download it again and compare the dirs.

MichaIng, 2025-04-30:
This works well:
```sh
find /usr/lib | while read -r path; do dpkg -S "$path" > /dev/null; done
```

Only Python creates a bytecode cache for scripts stored in this dir as well. Better ignore this. Wouldn't hurt to remove them, as they are all re-generated, but still false positives:
```sh
find /usr/lib | while read -r path; do [[ $path == *'/__pycache__'* ]] || dpkg -S "$path" > /dev/null; done
```

danergo, 2025-04-30:
Thanks!

Dpkg -s will confirm me in case library is belonging to something I installed via "apt" or via dietpi-software, right?

danergo, 2025-05-05:
Thank you.

This needed a little finetune, because `dpkg -S` needs library names (paths) without prefix:

```
# dpkg -S /lib/aarch64-linux-gnu/libc.so
dpkg-query: no path found matching pattern /lib/aarch64-linux-gnu/libc.so
BUT
# dpkg -S aarch64-linux-gnu/libc.so
libc6-dev:arm64: /usr/share/gdb/auto-load/lib/aarch64-linux-gnu/libc.so.6-gdb.py
libc6-dev:arm64: /usr/lib/aarch64-linux-gnu/libc.so
libc6:arm64: /lib/aarch64-linux-gnu/libc.so.6
```

So I created a bit more usable one-liner:

```
find /usr/lib -iname "*.so*" -type f | sed 's|^/usr/lib/||' | while read -r path; do dpkg -S $path > /dev/null; done
```

This will hide `dpkg`'s output of which library belongs to what package (which is unnecessary information for me as of now), and dpkg will otherwise run onto error in case a library is not found thus it will be printed.

Thank you!

closed 2025-05-05 by danergo

MichaIng, 2025-05-05:
I needs them with the correct prefix, hence not `/lib` but `/usr/lib` as latest since the usr-merge in Debian Bookworm. This is why I used the `find /usr/lib` statement. So removing that prefix in your final `find` is not needed.

danergo, 2025-05-05:
Well, I'm on Bookworm (Debian 12), but even with `/usr/lib` prefix it's not working:

```
# dpkg -S /usr/lib/aarch64-linux-gnu/libc.so.6
dpkg-query: no path found matching pattern /usr/lib/aarch64-linux-gnu/libc.so.6
```

But

```
# dpkg -S aarch64-linux-gnu/libc.so.6                       
libc6-dev:arm64: /usr/share/gdb/auto-load/lib/aarch64-linux-gnu/libc.so.6-gdb.py
libc6:arm64: /lib/aarch64-linux-gnu/libc.so.6
```

```
# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

danergo, 2025-05-05:
It might be a "mule" system, because I installed it long time ago and everytime DietPi offered an update i accepted. So its quite up2date but some libs might ended up in wrong location.

Currently, fresh install is not possible, so i recommend closing this for now, don't want to take more time from yoi :)

MichaIng, 2025-05-05:
Oh yeah some libraries still install to `/lib` instead of `/usr/lib` on Bookworm, I didn't know that: https://packages.debian.org/bookworm/amd64/libc6/filelist

Okay in that case easiest to remove the prefix. It works as well with some path elements, but it must not start with a slash, else is interpreted as complete path:
```sh
dpkg -S lib/aarch64-linux-gnu/libc.so.6
```
But this is moreless the same you did already.