MichaIng/DietPi #7515: switch to chrony for better accuracy on clock

klh, 2025-05-01:
is a drop in replacement but really helps especially with audio

MichaIng, 2025-05-01:
```sh
/boot/dietpi/func/dietpi-set_software ntpd-mode custom # or do via dietpi-config
apt install chrony
```
Other alternatives: `ntp` `ntpsec` `htpdate` or `systemd-timesyncd` ("Daemon + Drift" in `dietpi-config`).

The default is to not run any network time daemon at all, but sync it once a day instead, which is sufficient for like 95% of typical server systems. For more accuracy, `systemd-timesyncd` as 24/7 daemon is a light alternative, for high accuracy indeed Chrony is great, though its core feature is to sync between multiple Chrony instances within the same network. Still I agree it is probably the best available option if high accuracy is required.

MichaIng, 2025-06-08:
I'll mark this as closed. Feel free to reopen if any question left.

closed 2025-06-08 by MichaIng